john30/ebusd #901: unable to open /dev/ttyAMA0: ERR: send error #900

zarch1972, 2023-05-19:
### Description

unable to open /dev/ttyAMA0: ERR: send error #900 on startup of ebusd

### Actual behavior

unable to open /dev/ttyAMA0: ERR: send error #900

### Expected behavior

when starting ebusd, i'd hope to connect the /dev/ttyAMA0

### ebusd version

current source from git

### ebusd arguments

ebusd --scanconfig=full -d ens:/dev/ttyAMA0 --latency=50 --loglevel=debug

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

adapter 3.1 RPi

### Logs

- Brand new eBUS adapter for Raspberry Pi.
- Running on a Pi 4.
- Vaillant device is 5kW Arotherm Plus heat pump
- I installed ebusd 23.1.23.1 firmware via USB connection (Windows machine)
- And installed ebusd software on the Pi4 via https://github.com/john30/ebusd-debian (went for MQTT variant as want to get stuff into Home Assistant)

```
[mick@ebusd:~ $ sudo systemctl status ebusd
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
     Loaded: loaded (/lib/systemd/system/ebusd.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-05-19 15:30:57 BST; 4min 30s ago
    Process: 1657 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=0/SUCCESS)
   Main PID: 1659 (ebusd)
      Tasks: 4 (limit: 3933)
        CPU: 208ms
     CGroup: /system.slice/ebusd.service
             └─1659 /usr/bin/ebusd --scanconfig=full -d ens:/dev/ttyAMA0 --latency=50 --loglevel=debug]

May 19 15:30:57 ebusd systemd[1]: Starting ebusd, the daemon for communication with eBUS heating systems....
May 19 15:30:57 ebusd systemd[1]: Started ebusd, the daemon for communication with eBUS heating systems..
```

enable_uart=1 was already set in /boot/config.txt

```
mick@ebusd:~ $ dmesg|grep ttyAMA0
[    1.523816] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2

mick@ebusd:~ $ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 May 19 14:27 /dev/ttyAMA0
mick@ebusd:~ $ 
```
ebusd is running as root as a service, so permissions look ok?

Here is a snapshot of the logfile
```
2023-05-19 15:30:57.560 [main notice] ebusd 23.1.23.1 started with full scan on enhanced device /dev/ttyAMA0
2023-05-19 15:30:57.561 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-05-19 15:30:57.735 [main info] reading templates /
2023-05-19 15:30:57.890 [main info] read templates in /
2023-05-19 15:30:57.891 [main info] reading file memory.csv
2023-05-19 15:30:58.085 [main info] successfully read file memory.csv
2023-05-19 15:30:58.085 [main info] reading file broadcast.csv
2023-05-19 15:30:58.240 [main info] successfully read file broadcast.csv
2023-05-19 15:30:58.240 [main info] read config files, got 11 messages
2023-05-19 15:30:58.241 [bus error] unable to open /dev/ttyAMA0: ERR: send error
2023-05-19 15:30:58.242 [bus notice] bus started with own address 31/36
2023-05-19 15:30:58.242 [main info] registering data handlers
2023-05-19 15:30:58.242 [main info] registered data handlers
2023-05-19 15:31:08.243 [main debug] performing regular tasks
```

So i'm not sure what part of the setup process i've missed?
It seems like it can't open the USB AMA0 connection, but not sure why.

Any help or pointers would be appreciated.

Thanks.

john30, 2023-05-19:
maybe have a look at `dmesg` output for some kernel error messages.
if nothing is in there, you need to get that device being readable and writable to a process (not necessarily ebusd). so `cat >dev` and `cat <dev` both need to work.
maybe there is another process using the device, check with `lsof` and friends

zarch1972, 2023-05-20:
Thanks @john30 

I don't see anything the shouts out in dmesg, unless no DMA platform data?

```
mick@ebusd:~ $ dmesg | grep serial
[    1.522906] uart-pl011 fe201000.serial: there is not valid maps for state default
[    1.523648] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    1.523816] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 36, base_baud = 0) is a PL011 rev2
[    1.532179] bcm2835-aux-uart fe215040.serial: there is not valid maps for state default
[    1.533078] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 37, base_baud = 62500000) is a 16550
[    7.957109] uart-pl011 fe201000.serial: no DMA platform data
mick@ebusd:~ $ 
```

Found this in lsof, hciattach, which is bluetooth.

```
mick@ebusd:~ $ sudo lsof | grep ttyAMA0
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
hciattach  775                            root    3u      CHR             204,64      0t0        133 /dev/ttyAMA0
ebusd     1659                            root    1uW     CHR             204,64      0t0        133 /dev/ttyAMA0
ebusd     1659 1660 bushandle             root    1uW     CHR             204,64      0t0        133 /dev/ttyAMA0
ebusd     1659 1661 network               root    1uW     CHR             204,64      0t0        133 /dev/ttyAMA0
ebusd     1659 1662 mainloop              root    1uW     CHR             204,64      0t0        133 /dev/ttyAMA0
mick@ebusd:~ $ 
```

Added the following to /boot/config.txt to disable bluetooth and rebooted

```
# Disable Bluetooth
dtoverlay=disable-bt
```

Boom!

```
mick@ebusd:~ $ ebusctl i
version: ebusd 23.1.23.1
device: /dev/ttyAMA0, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 43
max symbol rate: 179
min arbitration micros: 7
max arbitration micros: 87
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 5
messages: 584
conditional: 0
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803", loaded "vaillant/15.basv.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0217;HW=8803", loaded "vaillant/ec.sol.sc.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4029;HW=5703"

mick@ebusd:~ $ 
```

Perhaps Bluetooth shares the same channel on the board as the USB/serial on the Pi4?
Maybe there is a way to enable bluetooth, but use a different /dev/tty?

Very strange.  Anyway, thanks @john30 and hope this helps someone out.

I will continue to have a play with things.

closed 2023-06-03 by john30