john30/ebusd #911: error reading config files (latest docker  john30/ebusd:devel)

Mozzarella7, 2023-05-26:
### Description

Hi, 
on latest docker developer version I got  reading config files error.
When I switch to "latest" version evrything is working fine.
Config files held locally.
See logs for both senarios:

### Actual behavior

[main error] error reading config files from /config/ebusd/config/en/: ERR: end of input reached, last error: vaillant/fc.rcc.7.csv:1: ERR: end of input reached, missing message type/name/pbsb

### Expected behavior

no error

### ebusd version

current source from git

### ebusd arguments

version: '3' services:   ebusd:     container_name: ebusd_new     image: john30/ebusd:devel     restart: unless-stopped     #tty: true     #devices:       #- '/dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_62928ffa24deea119e8ccf149a583cc7-if00-port0:/dev/ttyUSB0'       #- '/dev/ttyUSB0:/dev/ttyUSB0'     #ports:       #- 9999:9999     #configuration colume if you need to customize it (also add --configpath=/config to command param)     volumes:       - /usr/share/hassio/homeassistant:/config     command: --foreground --scanconfig --device=enh:192.168.5.110:9999 --mqtthost=192.168.5.104 --mqttport=1883 --mqttuser=mqtt-user --mqttpass=19761109 --configpath=/config/ebusd/config/en/ --latency=50 --pollinterval=1 --accesslevel=* --log=all:notice

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

devel

### Hardware interface

adapter 2

### Logs

john30/ebusd:devel

> `2023-05-26 18:51:58.987 [main notice] ebusd 23.1.23.1-15-g773a4dd started with auto scan on enhanced device 192.168.5.110:9999

2023-05-26 18:51:59.022 [main error] error reading config files from /config/ebusd/config/en/: ERR: end of input reached, last error: vaillant/fc.rcc.7.csv:1: ERR: end of input reached, missing message type/name/pbsb
2023-05-26 18:51:59.060 [bus notice] device status: resetting
2023-05-26 18:51:59.061 [bus notice] bus started with own address 31/36
2023-05-26 18:51:59.062 [bus notice] device status: reset
2023-05-26 18:51:59.065 [mqtt notice] connection established
2023-05-26 18:51:59.091 [bus notice] signal acquired
2023-05-26 18:52:08.221 [bus notice] new master 10, master count 2
2023-05-26 18:52:08.284 [bus notice] new master 03, master count 3
2023-05-26 18:52:08.284 [update notice] received unknown MS cmd: 1008b5110101 / 09605d0080ff700000ff
2023-05-26 18:52:08.555 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-05-26 18:52:08.808 [update notice] received unknown MS cmd: 1008b5110102 / 06033c72467e6e
2023-05-26 18:52:09.217 [bus notice] scan 08: ;Vaillant;BAI00;0603;9102
2023-05-26 18:52:09.217 [update notice] store 08 ident: done
2023-05-26 18:52:09.217 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0603;9102
2023-05-26 18:52:09.217 [bus notice] scan 08: ;Vaillant;BAI00;0603;9102
2023-05-26 18:52:09.221 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0603, HW9102
2023-05-26 18:52:09.222 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found
2023-05-26 18:52:09.237 [main error] error executing instructions: ERR: duplicate entry, last error: error resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not founderror resolving condition for "08.bai": ERR: element not found condition scan id: message not found, error loading "vaillant/bai.308523.inc" for "08": ERR: duplicate entry vaillant/bai.308523.inc:13: ERR: duplicate entry, duplicate ID, included "vaillant/hcmode.inc" for "08", included "vaillant/iotesthp.inc" for "08", included "vaillant/errors.inc" for "08", error including "vaillant/hcmode.inc" for "08": ERR: duplicate entry vaillant/hcmode.inc:4: ERR: duplicate entry, duplicate ID, included "vaillant/hwcmode.inc" for "23", included "vaillant/timer.inc" for "23", included "vaillant/iotest630.inc" for "23", included "vaillant/errors.inc" for "23", included "vaillant/yield3f40.inc" for "a0", included "vaillant/hwcmode.inc" for "a0", included "vaillant/timer.inc" for "a0", included "vaillant/errors.inc" for "a0", included "vaillant/iotestbsol.inc" for "a0"
2023-05-26 18:52:09.237 [main notice] found messages: 441 (142 conditional on 51 conditions, 5 poll, 55 update)
2023-05-26 18:52:09.348 [update error] unable to parse poll-read bai PartloadHcKW from 3108b509030d0704 / 00: ERR: invalid position
2023-05-26 18:52:10.239 [mqtt error] decode bai PartloadHcKW: ERR: invalid position
2023-05-26 18:52:11.129 [update error] unable to parse poll-read bai WPPostrunTime from 3108b509030df703 / 00: ERR: invalid position
2023-05-26 18:52:11.405 [bus notice] scan 15: ;Vaillant;B7V00;0422;5503
2023-05-26 18:52:11.405 [update notice] store 15 ident: done
[... 63 lines cut ...]
2023-05-26 18:49:37.482 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0
2023-05-26 18:49:37.719 [update notice] received update-read broadcast vdatetime QQ=10: 20:49:35;26.05.2023
2023-05-26 18:49:37.978 [update notice] received read bai Status02 QQ=10: auto;60;57.0;70;63.0
2023-05-26 18:49:38.207 [update notice] sent poll-read bai Gasvalve QQ=31: off
2023-05-26 18:49:38.343 [update notice] received unknown MS cmd: 1008b5110100 / 080a030f001f000080
2023-05-26 18:49:38.559 [update notice] received update-read broadcast outsidetemp QQ=10: 14.324
2023-05-26 18:49:40.185 [update notice] sent poll-read bai FanSpeed QQ=31: 0
2023-05-26 18:49:42.136 [update notice] sent poll-read bai PositionValveSet QQ=31: heating
2023-05-26 18:49:44.136 [update notice] sent poll-read bai HwcWaterflow QQ=31: 0.00
2023-05-26 18:49:46.140 [update notice] sent poll-read bai FlowTemp QQ=31: 48.62;ok
2023-05-26 18:49:47.280 [update notice] received read bai Status01 QQ=10: 48.5;47.5;-;-;56.5;off
2023-05-26 18:49:47.547 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0
2023-05-26 18:49:48.173 [update notice] sent poll-read bai ReturnTemp QQ=31: 47.50;64775;ok
2023-05-26 18:49:50.141 [update notice] sent poll-read bai OutdoorstempSensor QQ=31: -60.44;cutoff
2023-05-26 18:49:52.146 [update notice] sent poll-read bai WaterPressure QQ=31: 1.532;ok
2023-05-26 18:49:54.153 [update notice] sent poll-read bai Flame QQ=31: off
2023-05-26 18:49:56.158 [update notice] sent poll-read bai ModulationTempDesired QQ=31: 17.0
[... cut ...]
2023-05-26 18:49:57.605 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0
2023-05-26 18:49:58.195 [update notice] sent poll-read bai PowerValue QQ=31: 11 04 54 15 64 18
2023-05-26 18:50:00.152 [update notice] sent poll-read bai HwcSwitch QQ=31: on
2023-05-26 18:50:02.146 [update error] unable to parse poll-read bai HcPumpMode from 3108b509030d4904 / 00: ERR: invalid position
2023-05-26 18:50:04.188 [update notice] sent poll-read b7v HwcMaxFlowTempDesired QQ=31: 80
2023-05-26 18:50:06.185 [update notice] sent poll-read b7v SystemFlowTemp QQ=31: -
`

john30, 2023-05-30:
sounds like a container config issue to me. please post the output of docker inspect for both running containers in order to check

Mozzarella7, 2023-05-31:
logs for ebusd:devel
`
[{"Id":"fe6671f276c206c3985883fad75edec6d0a926e0ddcb7f415ef1f19289866bc5","Created":"2023-05-31T05:51:50.077308859Z","Path":"/docker-entrypoint.sh","Args":["--foreground","--scanconfig","--device=enh:192.168.5.110:9999","--mqtthost=192.168.5.104","--mqttport=1883","--mqttuser=mqtt-user","--mqttpass=xxxxx","--configpath=/config/ebusd/config/en/","--latency=50","--pollinterval=1","--accesslevel=*","--log=all:notice"],"State":{"Status":"running","Running":true,"Paused":false,"Restarting":false,"OOMKilled":false,"Dead":false,"Pid":299306,"ExitCode":0,"Error":"","StartedAt":"2023-05-31T05:51:52.670422034Z","FinishedAt":"0001-01-01T00:00:00Z"},"Image":"sha256:f06b1587ac1207d534ab96f3815b56ce1f15628974d48abda401236b8679b810","ResolvConfPath":"/var/lib/docker/containers/fe6671f276c206c3985883fad75edec6d0a926e0ddcb7f415ef1f19289866bc5/resolv.conf","HostnamePath":"/var/lib/docker/containers/fe6671f276c206c3985883fad75edec6d0a926e0ddcb7f415ef1f19289866bc5/hostname","HostsPath":"/var/lib/docker/containers/fe6671f276c206c3985883fad75edec6d0a926e0ddcb7f415ef1f19289866bc5/hosts","LogPath":"","Name":"/ebusd_new","RestartCount":0,"Driver":"overlay2","Platform":"linux","MountLabel":"","ProcessLabel":"","AppArmorProfile":"docker-default","ExecIDs":null,"HostConfig":{"Binds":["/usr/share/hassio/homeassistant:/config:rw"],"ContainerIDFile":"","LogConfig":{"Type":"journald","Config":{"tag":"{{.Name}}"}},"NetworkMode":"ebusd_new_default","PortBindings":{},"RestartPolicy":{"Name":"unless-stopped","MaximumRetryCount":0},"AutoRemove":false,"VolumeDriver":"","VolumesFrom":null,"ConsoleSize":[0,0],"CapAdd":null,"CapDrop":null,"CgroupnsMode":"host","Dns":null,"DnsOptions":null,"DnsSearch":null,"ExtraHosts":[],"GroupAdd":null,"IpcMode":"private","Cgroup":"","Links":null,"OomScoreAdj":0,"PidMode":"","Privileged":false,"PublishAllPorts":false,"ReadonlyRootfs":false,"SecurityOpt":null,"UTSMode":"","UsernsMode":"","ShmSize":67108864,"Runtime":"runc","Isolation":"","CpuShares":0,"Memory":0,"NanoCpus":0,"CgroupParent":"","BlkioWeight":0,"BlkioWeightDevice":null,"BlkioDeviceReadBps":null,"BlkioDeviceWriteBps":null,"BlkioDeviceReadIOps":null,"BlkioDeviceWriteIOps":null,"CpuPeriod":0,"CpuQuota":0,"CpuRealtimePeriod":0,"CpuRealtimeRuntime":0,"CpusetCpus":"","CpusetMems":"","Devices":null,"DeviceCgroupRules":null,"DeviceRequests":null,"MemoryReservation":0,"MemorySwap":0,"MemorySwappiness":null,"OomKillDisable":false,"PidsLimit":null,"Ulimits":null,"CpuCount":0,"CpuPercent":0,"IOMaximumIOps":0,"IOMaximumBandwidth":0,"MaskedPaths":["/proc/asound","/proc/acpi","/proc/kcore","/proc/keys","/proc/latency_stats","/proc/timer_list","/proc/timer_stats","/proc/sched_debug","/proc/scsi","/sys/firmware"],"ReadonlyPaths":["/proc/bus","/proc/fs","/proc/irq","/proc/sys","/proc/sysrq-trigger"]},"GraphDriver":{"Data":{"LowerDir":"/var/lib/docker/overlay2/dd52e112bfa27a05cd3c15af2e96bc38c6b0205415f92c9477ea149ba6829880-init/diff:/var/lib/docker/overlay2/d23cc4ac7ba840504fad4d3be39daf6acf1ef646ed0b05f92a4bd67212558bf7/diff:/var/lib/docker/overlay2/a3345a572803312a5f9c31696a5ede3718741c4d9a3b92d8dca1b4ed582f8d6a/diff:/var/lib/docker/overlay2/0cef6692b11e2c4f1b4e97ef4f2f5094e3f6d55253396b6ccad8178cb0339e0b/diff:/var/lib/docker/overlay2/640315df100a18fa82e6491b308749af68ea75050b1bacd6cff4ddad9deb3b0b/diff:/var/lib/docker/overlay2/c49440472cbe7430eaab8434541616c43bd0ecc11d932f93ee9ab140dee5be06/diff","MergedDir":"/var/lib/docker/overlay2/dd52e112bfa27a05cd3c15af2e96bc38c6b0205415f92c9477ea149ba6829880/merged","UpperDir":"/var/lib/docker/overlay2/dd52e112bfa27a05cd3c15af2e96bc38c6b0205415f92c9477ea149ba6829880/diff","WorkDir":"/var/lib/docker/overlay2/dd52e112bfa27a05cd3c15af2e96bc38c6b0205415f92c9477ea149ba6829880/work"},"Name":"overlay2"},"Mounts":[{"Type":"bind","Source":"/usr/share/hassio/homeassistant","Destination":"/config","Mode":"rw","RW":true,"Propagation":"rprivate"}],"Config":{"Hostname":"fe6671f276c2","Domainname":"","User":"","AttachStdin":false,"AttachStdout":true,"AttachStderr":true,"ExposedPorts":{"8888/tcp":{}},"Tty":false,"OpenStdin":false,"StdinOnce":false,"Env":["PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin","EBUSD_ARCH=amd64","EBUSD_VERSION=23.1","EBUSD_FOREGROUND="],"Cmd":["--foreground","--scanconfig","--device=enh:192.168.5.110:9999","--mqtthost=192.168.5.104","--mqttport=1883","--mqttuser=mqtt-user","--mqttpass=xxxxx","--configpath=/config/ebusd/config/en/","--latency=50","--pollinterval=1","--accesslevel=*","--log=all:notice"],"Image":"john30/ebusd:devel","Volumes":null,"WorkingDir":"","Entrypoint":["/docker-entrypoint.sh"],"OnBuild":null,"Labels":{"com.docker.compose.config-hash":"ae4de66b0a88d344f122680cec5e3ed032143eaab493fbc9f33371f7b8665885","com.docker.compose.container-number":"1","com.docker.compose.depends_on":"","com.docker.compose.image":"sha256:f06b1587ac1207d534ab96f3815b56ce1f15628974d48abda401236b8679b810","com.docker.compose.oneoff":"False","com.docker.compose.project":"ebusd_new","com.docker.compose.project.config_files":"/data/compose/17/docker-compose.yml","com.docker.compose.project.working_dir":"/data/compose/17","com.docker.compose.replace":"7282f6ef9bef1c593e669a37465482dd47ef2542c80f5ec54065826c1ab9e7a5","com.docker.compose.service":"ebusd","com.docker.compose.version":"2.17.2","maintainer":"ebusd@ebusd.eu","version":"23.1-amd64-devel"}},"NetworkSettings":{"Bridge":"","SandboxID":"4400f2b9e9a624e3ac83502ebb10876e32e5336f7db2e438476eab51ce6ea5aa","HairpinMode":false," [... cut ...] orkID":"9ba23b064f30e5454a2b7bb5cd9c8a3a19f0751bcda9b7206560366dab9b9e28","EndpointID":"69423a1d7f7f2233b1e0e0841e6514a538afcc5c02276bd5580f7a24c22275f9","Gateway":"172.20.0.1","IPAddress":"172.20.0.2","IPPrefixLen":16,"IPv6Gateway":"","GlobalIPv6Address":"","GlobalIPv6PrefixLen":0,"MacAddress":"02:42:ac:14:00:02","DriverOpts":null}}}}]`

Both containers started from the same portainer stack with changed image. 
Today I pull latest devel version and it looks like error is fixed, I think.
Thanks!

closed 2023-06-02 by Mozzarella7