john30/ebusd #919: Daemon starts, but no connection

pleibling, 2023-06-01:
### Description

I have a EBUScontroller USB Version 3.1connected to VM (ESXI USB Passthrough) with Debian 11.7 (all Updates). I get via lsusb -v this:

```
Bus 002 Device 004: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Couldn’t open device, some information will be missing
Device Descriptor:
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0
bDeviceSubClass 0
bDeviceProtocol 0
bMaxPacketSize0 64
idVendor 0x10c4 Silicon Labs
idProduct 0xea60 CP210x UART Bridge
bcdDevice 1.00
iManufacturer 1 Silicon Labs
iProduct 2 CP2102N USB to UART Bridge Controller
iSerial 3 68639f303adeea11aee2cf149a583cc7
```
"ebusctl i" show this:
```
version: ebusd 23.1.23.1
update check: OK
device: /dev/ttyUSB0
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
And the /var/log/ebusd.log show this:
```
2023-06-01 11:47:02.333 [main notice] ebusd 23.1.23.1 started with auto scan on device /dev/ttyUSB0
2023-06-01 11:47:02.412 [bus notice] bus started with own address 31/36
2023-06-01 11:49:12.560 [main notice] update check: OK
```
The Configuration is:
`EBUSD_OPTS="--device=/dev/ttyUSB0 --scanconfig --configpath=/home/pleibling/ebusd-configuration/ebusd-2.1.x/de"`
Configfile is reachable and have all access:
```
root@ebusd:/home/pleibling/ebusd-configuration# ls -lisa
insgesamt 100
1026  4 drwxrwxrwx  9 root      root       4096  1. Jun 08:22 .
 731  4 drwxr-xr-x 14 pleibling pleibling  4096  1. Jun 08:22 ..
1072  4 -rwxrwxrwx  1 root      root       1217  1. Jun 08:22 ChangeLog.md
1076  4 drwxrwxrwx  6 root      root       4096  1. Jun 08:22 ebusd-0.5.x
1112  4 drwxrwxrwx  4 root      root       4096  1. Jun 08:22 ebusd-1.x.x
1159  4 drwxrwxrwx  4 root      root       4096  1. Jun 08:22 ebusd-2.1.x
1428  4 drwxrwxrwx  4 root      root       4096  1. Jun 08:22 ebusd-2.x.x
1027  4 drwxrwxrwx  8 root      root       4096  1. Jun 08:22 .git
1057  4 -rwxrwxrwx  1 root      root         57  1. Jun 08:22 .gitignore
1561  0 lrwxrwxrwx  1 root      root         11  1. Jun 08:22 latest -> ebusd-2.1.x
1562  4 drwxrwxrwx  4 root      root       4096  1. Jun 08:22 libebus-0.1.x
1579  4 drwxrwxrwx  4 root      root       4096  1. Jun 08:22 libebus-0.2.x
1073 32 -rwxrwxrwx  1 root      root      32473  1. Jun 08:22 LICENSE
1596  4 -rwxrwxrwx  1 root      root         84  1. Jun 08:22 make_all.sh
1597  4 -rwxrwxrwx  1 root      root       2142  1. Jun 08:22 make_debian.sh
1598  4 -rwxrwxrwx  1 root      root       1380  1. Jun 08:22 make_tgz.sh
1074  8 -rwxrwxrwx  1 root      root       6313  1. Jun 08:22 README.md
1075  4 -rwxrwxrwx  1 root      root          3  1. Jun 08:22 VERSION
```
Everything seems to be ok, but i get not "signal: aquire".
Is the controller broken or is this a ebusd problem?
Can anyone help me please?
Thanks a lot.

### Actual behavior

Could not get state of Signal to "aquire".

### Expected behavior

Normale the state of Signal have to be aquire.

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--device=/dev/ttyUSB0 --scanconfig --configpath=/home/pleibling/ebusd-configuration/ebusd-2.1.x/de"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Related integration

other

### Logs

2023-06-01 11:47:02.333 [main notice] ebusd 23.1.23.1 started with auto scan on device /dev/ttyUSB0
2023-06-01 11:47:02.412 [bus notice] bus started with own address 31/36
2023-06-01 11:49:12.560 [main notice] update check: OK

john30, 2023-06-01:
you probably forgot to add the "ens:" prefix to the device string as [documented](https://adapter.ebusd.eu/v31/index.en.html#usb)

pleibling, 2023-06-01:
You are my Hero - and make my Day. Thanks a lot for your fast help - this solve my problem :).

pleibling, 2023-06-01:
One question again, now it works and i get a lot of informations. I want to move it now on my docker Container, how is the commandline for this?

If i use this:
 docker run -d --name=ebusd --device=/dev/ttyUSB0:/dev/ttyUSB0 -p 8888 john30/ebusd

the container start without error, but has the problem before - ebusctl i show no connection, but whe i try to start like this - the container doesnt start (docker: bad mode specified: /dev/ttyUSB0.):
 docker run -d --name=ebusd --device=ens:/dev/ttyUSB0:/dev/ttyUSB0 -p 8888 john30/ebusd

What is the correct Parameter?

Thanks for your help.

john30, 2023-06-03:
these are two completely different ports, don't mix them up! 8888 is the standard port for TCP client connections whereas the httpport is optional and may not overlap with the other one.
anyway, you need to check your docker run config so that dockerd will publish the port as you need it.

john30, 2023-09-23:
closed for inactivity

closed 2023-09-23 by john30