john30/ebusd #923: ebusd scanning only loads all needed csv files when loglevel=debug

DiKa67, 2023-06-03:
### Description

ebusd does scanning but finds only one device (out of six) and therefore loads only ONE out of six csv files.
When I restart my heating device (Vailant) by powering off/on, then all 6 are scanned, but still only ONE CSV file is loaded.

And now the funny part: :-)
As soon as loglevel=debug, then everything is fine, all scans are completed within seconds and all six csv files are loaded accordingly, even without restarting the Vaillant system.

Note: I am using the new v5 adapter (cannot yet be selected in adapter list below), attached with USB to Raspberry 4

_ebusctl info
version: ebusd 23.1.23.1
device: /dev/ttyACM0, enhanced
signal: acquired
symbol rate: 42
max symbol rate: 119
min arbitration micros: 2
max arbitration micros: 53
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 3
messages: 143
conditional: 49
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanning
address 10: master #2
address 23: slave, scanning
address 25: slave, scanning
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanning
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0118;HW=5202", loaded "vaillant/e0.omu.csv"_

After several (10 minutes) the scan will find also other devices, but does not load further the CSV files.

_version: ebusd 23.1.23.1
device: /dev/ttyACM0, enhanced
signal: acquired
symbol rate: 33
max symbol rate: 121
min arbitration micros: 2
max arbitration micros: 53
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 3
messages: 143
conditional: 49
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanning
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901"
address 23: slave, scanning
address 25: slave, scanning
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanning
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0118;HW=5202", loaded "vaillant/e0.omu.csv"__

Also remarkable: ebusctl info does not show the firmware version in the line "device" in the buggy case! 

### Actual behavior

ebusd does scanning but finds only one device (out of six) and therefore loads only ONE out of six csv files.
When I restart my heating device (Vailant) by powering off/on, then all 6 are scanned, but still only ONE CSV file is loaded.
[ebusd_error.log](https://github.com/john30/ebusd/files/11642914/ebusd_error.log)

### Expected behavior

ebusd scans all 6 devices and loads the CSV files accordingly. 

_ebusctl info
version: ebusd 23.1.23.1
device: /dev/ttyACM0, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 36
max symbol rate: 131
min arbitration micros: 2
max arbitration micros: 109
min symbol latency: 3
max symbol latency: 5
reconnects: 0
masters: 3
messages: 741
conditional: 262
poll: 3
update: 60
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0118;HW=5202", loaded "vaillant/e0.omu.csv"_
[ebus.log](https://github.com/john30/ebusd/files/11642912/ebus.log)

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--configpath=http://cfg.ebusd.eu/ --scanconfig=full -d ens:/dev/ttyACM0  -p 8888  --latency=50 --receivetimeout=100000 -l /var/log/ebusd.log  --logareas=bus  --loglevel=debug  --lograwdatasize=100  --httpport=8889 --htmlpath=/var/ebusd/html"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 USB

### Logs

[ebusd_error.log](https://github.com/john30/ebusd/files/11642923/ebusd_error.log)
[ebus_ok.log](https://github.com/john30/ebusd/files/11642924/ebus_ok.log)

DiKa67, 2023-06-03:
Note: in the buggy case, the parameter used is loglevel=info

john30, 2023-06-04:
thats weird indeed. can you post a raw log (without debug level) from the start of ebusd until the first scan message that should succeed (slave 08 I suppose)? then cancel and also post the easi> log

DiKa67, 2023-06-04:
Hi John, 

Thanks for taking the time, very appreciated! :-) 

Her the log files, normal and raw, for BOTH cases (ok and nok):
 
...01_ok_...  when scan and fileloads works (logareas=all) 
with #EBUSD_OPTS="--configpath=http://cfg.ebusd.eu/ --scanconfig=full -d ens:/dev/ttyACM0  -p 8888  --latency=50 --receivetimeout=100000 -l /var/log/ebusd.log  --logareas=all  --lograwdata  --lograwdatafile=/var/lo$

and

...02_nok... when scan takes very long and only one file is loaded finally.
with EBUSD_OPTS="--configpath=http://cfg.ebusd.eu/ --scanconfig=full -d ens:/dev/ttyACM0  -p 8888  --latency=50 --receivetimeout=100000 -l /var/log/ebusd.log  --lograwdata  --lograwdatafile=/var/log/ebusd_raw.log  $

[ebusd_01_ok_raw.log](https://github.com/john30/ebusd/files/11645326/ebusd_01_ok_raw.log)
[ebusd_01_ok.log](https://github.com/john30/ebusd/files/11645327/ebusd_01_ok.log)
[ebusd_02_nok_raw.log](https://github.com/john30/ebusd/files/11645328/ebusd_02_nok_raw.log)
[ebusd_02_nok.log](https://github.com/john30/ebusd/files/11645329/ebusd_02_nok.log)

As well as the easy log:

Build: 20230501
[ebusd device string](https://github.com/john30/ebusd/wiki/2.-Run#device-options): ens:/dev/ttyACM0
WiFi station: xxx, 192.168.xxx.xxx, 92% (-54dBm) (note: xxx to replace real values)
WiFi AP: EBUS, 192.168.4.1
11:59:52	Info	executeCommand: executing command: function start server "led 0 10"
11:59:52		function installed
11:59:52	Info	replExecuteFromFile: executing command: on wifi 1 start
11:59:52	Info	executeCommand: executing command: on wifi 1 start
11:59:52		network trigger installed
11:59:52	Info	replExecuteFromFile: executing command: on ethernet 1 start
11:59:52	Info	executeCommand: executing command: on ethernet 1 start
11:59:52		network trigger installed
11:59:52	Info	replExecuteFromFile: executing command: function aponfor120 "ap EBUS" "after -o 120 \"ap -o\""
11:59:52	Info	executeCommand: executing command: function aponfor120 "ap EBUS" "after -o 120 \"ap -o\""
11:59:52		function installed
11:59:52	Info	replExecuteFromFile: executing command: if "!§isnvs sta.on" "ap EBUS" else "if \"isnvs ap.on\" ap else aponfor120"
11:59:52	Info	executeCommand: executing command: if "!§isnvs sta.on" "ap EBUS" else "if \"isnvs ap.on\" ap else aponfor120"
11:59:52	Info	executeCommand: executing command: isnvs sta.on
11:59:52	Info	executeCommand: executing command: if "isnvs ap.on" ap else aponfor120
11:59:52	Info	executeCommand: executing command: isnvs ap.on
11:59:52	Info	executeCommand: executing command: aponfor120
11:59:52	Info	executeCommand: executing command: ap EBUS
11:59:52	Info	pp: pp rom version: 9387209
11:59:52	Info	net80211: net80211 rom version: 9387209
11:59:52	Info	wifi_init: rx ba win: 6
11:59:52	Info	wifi_init: tcpip mbox: 32
11:59:52	Info	wifi_init: udp mbox: 6
11:59:52	Info	wifi_init: tcp mbox: 6
11:59:52	Info	wifi_init: tcp tx win: 5744
11:59:52	Info	wifi_init: tcp rx win: 5744
11:59:52	Info	wifi_init: tcp mss: 1440
11:59:52	Info	wifi_init: WiFi IRAM OP enabled
11:59:52	Info	wifi_init: WiFi RX IRAM OP enabled
11:59:52	Info	phy_init: phy_version 960,7392adf,Mar 31 2023,16:27:13
11:59:52	Info	esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
11:59:52	Info	esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
11:59:52		AP mode, EBUS
11:59:52	Info	executeCommand: executing command: after -o 120 "ap -o"
11:59:52		time trigger installed
11:59:52	Info	replExecuteFromFile: executing command: if "isnvs sta.on" sta
11:59:52	Info	executeCommand: executing command: if "isnvs sta.on" sta
11:59:52	Info	executeCommand: executing command: isnvs sta.on
11:59:52	Info	executeCommand: executing command: sta
11:59:52		sta connecting to 'FritzHugoAlpe'
11:59:52	Info	cmd_wifi: sta disconnect, reconnect...
11:59:52	Info	repl_event_task: non-io event 1: 0
11:59:52	Info	executeCommand: executing command: led 0 0
11:59:52	Info	statusLedTimerFn: duty set to 0
11:59:55	Info	esp_netif_handlers: sta ip: 192.168.103.128, mask: 255.255.255.0, gw: 192.168.103.1
[... 58 lines cut ...]
11:59:55	Info	replExecuteFromFile: executing command: if "isnvs eth.on" "if \"var eth\" enet"
11:59:55	Info	executeCommand: executing command: if "isnvs eth.on" "if \"var eth\" enet"
11:59:55	Info	executeCommand: executing command: isnvs eth.on
11:59:55	Warning	executeCommand: command failed: isnvs eth.on = 1
11:59:55	Info	replExecuteFromFile: executing command: function ebusrpi "ebus -l" "led 0 50"
11:59:55	Info	executeCommand: executing command: function ebusrpi "ebus -l" "led 0 50"
11:59:55		function installed
11:59:55	Info	replExecuteFromFile: executing command: if "isnvs -s rpi ebus.host" ebusrpi exit
11:59:55	Info	executeCommand: executing command: if "isnvs -s rpi ebus.host" ebusrpi exit
11:59:55	Info	executeCommand: executing command: isnvs -s rpi ebus.host
11:59:55	Info	replExecuteFromFile: executing command: if "isnvs -s usb ebus.host" "ebus -jl" "led 0 50" exit
11:59:55	Info	executeCommand: executing command: if "isnvs -s usb ebus.host" "ebus -jl" "led 0 50" exit
11:59:55	Info	executeCommand: executing command: isnvs -s usb ebus.host
11:59:55	Info	executeCommand: executing command: ebus -jl
11:59:55	Info	init_ebus: switching to enhanced eBUS mode on JTAG
11:59:55	Info	setup_uart: eBUS UART 1 on TX=10(inv) RX=7(inv) with 2400 Bd
11:59:55	Info	uart: ALREADY NULL
11:59:55	Info	init_ebus: timer started
11:59:55	Info	executeCommand: executing command: led 0 50
11:59:55	Info	executeCommand: executing command: exit
11:59:55	Info	executeCommand: command exit
11:59:55	Info	executeCommand: command exit
11:59:55	Info	replExecuteFromFile: successfully executed 46/46 commands from '/data/main.eas'
11:59:55		Welcome to eBUS Adapter Shield Interface! Type 'help' to get the list of commands.
11:59:55	Info	statusLedTimerFn: duty set to 50

john30, 2023-06-11:
very strange. it looks as if the SYN generator completely ignores any traffic, even valid one. which is particular strange as other participants answer to requested data.

please post the log with raw bytes into the same file by using these args:
`--scanconfig=full -d ens:/dev/ttyACM0 --latency=50 --receivetimeout=100 -l /var/log/ebusd.log --lograwdata=bytes`

DiKa67, 2023-06-11:
Will do. As I am currently in holidays, I will perform it at the next weekend!
Thanks a lot.

DiKa67, 2023-06-18:
Hi John, 
After being back from holidays, here the log as requested with the parameters:

`EBUSD_OPTS="--scanconfig=full -d ens:/dev/ttyACM0 --latency=50 --receivetimeout=100 -l /var/log/ebusd.log --lograwdata=bytes"`

[ebusd.log](https://github.com/john30/ebusd/files/11780944/ebusd.log)

Btw. with those parameters it works pretty nice:
Here the result, all devices scanned and csv files loaded successfully after less than one minute:
```
ebusctl info

version: ebusd 23.1.23.1
device: /dev/ttyACM0, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 72
max symbol rate: 144
min arbitration micros: 2
max arbitration micros: 51
min symbol latency: 3
max symbol latency: 13
reconnects: 0
masters: 3
messages: 740
conditional: 262
poll: 3
update: 60
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0118;HW=5202", loaded "vaillant/e0.omu.csv"
```

Thanks for your support! :-)

DiKa67, 2023-07-15:
Hi John,

I now moved to a ebusd docker container on my NAS and all seems to be fine. 
So no further investigations here needed.
:-)

Thanks,
Dieter

john30, 2023-07-29:
so you're using 23.2 now, right?

avanmourik, 2023-08-27:
Hello, 
I am very new to python / ebusd and struggeling with seemingly similar problems as described in #923. There are only 3 elements on my EBUS : heatpump Arotherm plus VWL 75/6 A ; system assistant VWZ AI , and Sensocomfort VRC 720/2. Installed sept 2022. Data are stored on PI 4 , ARM71. USB is used to connect EBUSD with PI.
Using the most simple ebusd in /etc/default/ ebusd:
EBUSD_OPTS="--scanconfig"
EBUSD_OPTS="--scanconfig -d ens:/dev/ttyACM0" 

I get the following outputs, so equipment recognized but no data, very similar to Dika67 in #923 :
Ebusctl find : 
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = 27.250
broadcast signoflife = no data stored
broadcast vdatetime = 16:48:07;24.08.2023
general valuerange = no data stored
hmu ConsumptionThisYear10 = no data stored
hmu ConsumptionThisYear11 = no data stored
hmu ConsumptionThisYear12 = no data stored
hmu ConsumptionThisYear1 = no data stored
hmu ConsumptionThisYear2 = no data stored
hmu ConsumptionThisYear3 = no data stored
hmu ConsumptionThisYear4 = no data stored
hmu ConsumptionThisYear5 = no data stored
hmu ConsumptionThisYear6 = no data stored
hmu ConsumptionThisYear7 = no data stored
hmu ConsumptionThisYear8 = no data stored
hmu ConsumptionThisYear9 = no data stored
hmu ConsumptionTotal = no data stored
hmu CurrentCompressorUtil = no data stored
hmu CurrentConsumedPower = no data stored
hmu currenterror = no data stored
hmu CurrentYieldPower = no data stored
hmu DateTime = no data stored
hmu EnergyCool = no data stored
hmu EnergyHc = no data stored
hmu errorhistory = no data stored
hmu Hours = no data stored
hmu HoursCool = no data stored
hmu HoursHc = no data stored
hmu SetMode = auto;0.0;-;-;1;1;1;0;0;0
hmu State = 0;0;160;0
hmu Status01 = 25.5;24.5;-;-;-;off
hmu Status02 = no data stored
hmu Status16 = no data stored
hmu Status = no data stored
hmu StatusCirPump = off
hmu WaterThroughput = no data stored
hmu YieldThisYear10 = no data stored
hmu YieldThisYear11 = no data stored
hmu YieldThisYear12 = no data stored
hmu YieldThisYear1 = no data stored
hmu YieldThisYear2 = no data stored
hmu YieldThisYear3 = no data stored
hmu YieldThisYear4 = no data stored
hmu YieldThisYear5 = no data stored
hmu YieldThisYear6 = no data stored
hmu YieldThisYear7 = no data stored
hmu YieldThisYear8 = no data stored
hmu YieldThisYear9 = no data stored
hmu YieldTotal = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = Vaillant;HMU00;0607;5103
scan.15  = Vaillant;CTLV2;0514;1104
scan.76  = Vaillant;VWZIO;0202;0103

[... 145 lines cut ...]
2023-08-27 18:17:11.536 [update notice] received unknown MS cmd: 1008b5110101 />
2023-08-27 18:17:11.805 [update info] received MS cmd: 1076b5110101 / 09ffff701>
2023-08-27 18:17:11.805 [update notice] received unknown MS cmd: 1076b5110101 />
2023-08-27 18:17:12.072 [update info] received MS cmd: 1076b512030f0001 / 07008>
2023-08-27 18:17:12.072 [update notice] received unknown MS cmd: 1076b512030f00>
2023-08-27 18:17:12.286 [update info] received BC cmd: 10feb505025c00
2023-08-27 18:17:12.286 [update notice] received unknown BC cmd: 10feb505025c00
2023-08-27 18:17:12.550 [update info] received MS cmd: 1008b51009000000ffffff07>
2023-08-27 18:17:12.550 [update notice] received unknown MS cmd: 1008b510090000>
2023-08-27 18:17:12.814 [update info] received MS cmd: 1076b51009000000ffffff05>
2023-08-27 18:17:12.814 [update notice] received unknown MS cmd: 1076b510090000>
2023-08-27 18:17:13.087 [update info] received MS cmd: 1076b5040100 / 0a0313171>
2023-08-27 18:17:13.087 [update notice] received unknown MS cmd: 1076b5040100 />
2023-08-27 18:17:13.183 [update info] received MS cmd: 7108b5110107 / 0a000000a>

Some of the ctlv2 data however can be found via the read command

ebusctl read Date  27.08.2023
ebusctl read MaintenanceDate  01.01.2019
ebusctl read WaterPressure -
ebusctl read time 19:33:20
ebusctl read HydraulicScheme 8
ebusctl read FrostOverRideTime ERR: SYN received

Lost in the Jungle!!

DiKa67, 2023-09-02:
Hi avanmourik,
Can you please tell us about your setting, hardware, os, interface…
And I solved it by moving ebus to a docker container.
Before, it was installed on a Raspi, together with OpenHAB, influxdb, Grafana and some other stuff. It ran for some years pretty stable and suddenly brought those issues…

DiKa67, 2023-09-02:
> so you're using 23.2 now, right?

Still on 23.1 on docker. Works perfectly since 1,5 months now. :-)

avanmourik, 2023-09-02:
Hi,

Yes running 23.2

When it last ran, I used the ttyACM USB connection, and apart from getting almost no data it seemed to work.

Until it crashed completely and could not even start anymore (systemctl start ebusd  terminated with code 64 error)

 

I thought to get rid of all the hassle I am  in right now, by using the docker option as you suggested.

It actually seems a lot more elegant than the “traditional” procedure

So installed it, on the basis of a web tutorial and that was OK.

Pulled the files from John Baier from : https://hub.docker.com/r/john30/ebusd/

All ok and then the trouble starts : 

I flashed the device, set it up for PI serial, re-installed the bullseye-ebusd software for arm71 (but necessary with the image?)

 

I changed the device to ttyACM0, since I had been using this device in the previous setup.

It would not run , because the docker was looking for ttyUSB0 , I do not know why.

The command on the website was docker run --rm -it --device=/dev/ttyUSB1:/dev/ttyUSB0 -p 8888 john30/ebus.

But I changed it (maybe wrongly) to docker run --rm -it --device=/dev/ttyACM0:/dev/ttyACM1 -p 8888 john30/ebus

Don’t actually know what the 2 devices in 1 command mean

 

But the overruling question is what files are still used from the setup document for non-container setup.

Is /etc/default still used as configuration file (including a path to local csv files)?

I guess since you went through all this , so might know the answers.

 

Were there any other details you wanted to know (your first email)?

 

Kind regards,

john30, 2023-09-03:
if you move to docker, I suggest that you read the docs for that as it is all quite different (no /etc/defaults/ebusd e.g.): https://github.com/john30/ebusd/blob/master/contrib/docker/README.md
the compose file shows all settings available, all of which then have to be passed by environment variables or as arguments to the docker run command: https://github.com/john30/ebusd/blob/master/contrib/docker/docker-compose.example.yaml

john30, 2023-09-03:
besides: having an output of "no data stored" does not mean that there is none. you'd have to actively query these messages to see if the device provides that data. if you don't, then ebusd will answer from cache only. this is specifically true for the "find" command

john30, 2023-09-03:
@DiKa67 this hijacked issue is actually closable, right?
btw: you seem to be using a rather old firmware, maybe updating it might help as well, see [here](https://adapter.ebusd.eu/v5/ChangeLog#20230903)

DiKa67, 2023-09-03:
> @DiKa67 this hijacked issue is actually closable, right?
> btw: you seem to be using a rather old firmware, maybe updating it might help as well, see [here](https://adapter.ebusd.eu/v5/ChangeLog#20230903)

Correct, my issue is not longer an issue :-) Thanks John!

closed 2023-09-03 by DiKa67

avanmourik, 2023-09-03:
I used both the read command and the shell script , with same result. No values

avanmourik, 2023-09-03:
OK,

Understand.

Explains a lot.

I read the readme , but it does not suggest , that the downloadable version is a tailored one (docker pull john30/ebusd) 

Will dive into it