john30/ebusd #929: getting error message "invalid position in decode" when trying to read values from circuit "sc"

UnderPalms, 2023-06-09:
### Description

When I try to read all values from circuit "sc" I get the message "ERR: invalid position in decode". This error occurs for example when executing "ebusctl read SolGain". 
For all other objects which are not related to the circuit "sc" the ebus daemon is working fine and delivering sufficient values.

### Actual behavior

When reading values from circuit "sc" getting error message "ERR: invalid position in decode"

### Expected behavior

I expected proper values to be received. For example for "SolGain"

### ebusd version

23.1

### ebusd arguments

EBUSD_OPTS="--configpath=http://ebusd.eu/config/  -d ens:/dev/ttyAMA0 -p 8888 --latency=50 -receivetimeout=60000 -l /var/log/ebusd.log --scanconfig=full --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=* --enablehex

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2023-05-10 21:47:50.564 [main notice] ebusd 23.1.23.1-9-g1895339e started with auto scan on device /dev/ttyUSB0
2023-05-10 21:47:50.886 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:47:50.889 [bus notice] bus started with own address 31/36
2023-05-10 21:47:50.890 [bus notice] device invalid
2023-05-10 21:47:55.890 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:47:55.898 [bus notice] device invalid
2023-05-10 21:48:00.899 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:00.899 [bus notice] device invalid
2023-05-10 21:48:05.900 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:05.900 [bus notice] device invalid
2023-05-10 21:48:10.900 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:10.901 [bus notice] device invalid
2023-05-10 21:48:15.901 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:15.901 [bus notice] device invalid
2023-05-10 21:48:20.902 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:20.902 [bus notice] device invalid
2023-05-10 21:48:25.903 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:25.903 [bus notice] device invalid
2023-05-10 21:48:30.903 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:30.904 [bus notice] device invalid
2023-05-10 21:48:35.904 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:35.905 [bus notice] device invalid
2023-05-10 21:48:40.905 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:40.905 [bus notice] device invalid
2023-05-10 21:48:45.906 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:45.906 [bus notice] device invalid
2023-05-10 21:48:50.907 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:50.907 [bus notice] device invalid
2023-05-10 21:48:55.907 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:48:55.908 [bus notice] device invalid
2023-05-10 21:49:00.908 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:00.908 [bus notice] device invalid
2023-05-10 21:49:05.909 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:05.909 [bus notice] device invalid
2023-05-10 21:49:10.910 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:10.910 [bus notice] device invalid
2023-05-10 21:49:15.910 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:15.911 [bus notice] device invalid
2023-05-10 21:49:20.911 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:20.911 [bus notice] device invalid
2023-05-10 21:49:25.912 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:25.912 [bus notice] device invalid
2023-05-10 21:49:30.913 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:30.913 [bus notice] device invalid
2023-05-10 21:49:35.914 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:35.914 [bus notice] device invalid
2023-05-10 21:49:40.914 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:40.915 [bus notice] device invalid
2023-05-10 21:49:45.915 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:45.915 [bus notice] device invalid
2023-05-10 21:49:50.916 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:50.916 [bus notice] device invalid
2023-05-10 21:49:55.917 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:49:55.917 [bus notice] device invalid
2023-05-10 21:50:00.917 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:50:00.918 [bus notice] device invalid
2023-05-10 21:50:01.107 [main notice] update check: revision 23.1 available
2023-05-10 21:50:05.918 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:50:05.919 [bus notice] device invalid
2023-05-10 21:50:10.919 [bus error] unable to open /dev/ttyUSB0: ERR: element not found
2023-05-10 21:50:10.919 [bus notice] device invalid
2023-05-10 21:50:15.920 [bus error] unable to open /dev/ttyUSB0: ERR: element not found

john30, 2023-06-11:
the log does not match the start parameters and thus is irrelevant.
anyway, I guess the circuit is not capable of answering to the SolGain query and as such responds empty which will lead to the exact error report. you can check by looking at the hex bytes of that message (e.g. grab command)

UnderPalms, 2023-06-11:
Thanks John, 
indeed there's something wrong with my log. However, I copied that in as it represents the last entries saved in the log and also because the issue-mask required to fill something for the log-part.
I tried the grab command and indeed there're a lot of messages related to that circuit "sc" with the bus address "ec".
Here's the extract:

peter@raspberrypi:~ $ ebusctl grab result
1008b5110100 / 0832041b001f000080 = 731
1008b5160111 / 090105040e0430040a04 = 1
100ab5160111 / 00 = 1
1026b5160111 / 090103010e0130010301 = 1
10edb5160111 / 00 = 1
10feb510020601 = 155
01ecb5040110 / 0a00800080008000800080 = 1991
01ecb5040134 / 03000001 = 701
01feb505054ef51c0080 = 685
1008b5120204ff / 0101 = 150
1008b513020508 / 00 = 156
100ab505020400 / 00 = 1
1006b50903295600 / 065600f51c0000 = 76
1008b5100305ff01 / 0101 = 143
1006b5080707000030006d02 / 00 = 46
1006b50807070000a0031402 / 00 = 102
1008b516081000ffff0400212e / 0b0000ff0400c92e5431344c = 1
1008b516081002ffff0400812d / 0b0207ff04008b2d88b9ba49 = 34
100ab5050433000400 / 00 = 1
1026b5030c0700ffffffffffffffffffff / 0101 = 49
1026b516081000ffff0304212e / 0b2000ff0304c92e00000000 = 2
1026b516081002ffff0304a12e / 0b2201ff0304a12e00000000 = 21
1026b5230b0401010000000000220200 / 0101 = 1
01ecb505084d00040000000000 / 00 = 1
01ecb505084d00050000640100 / 00 = 1
01ecb505084d01030000190100 / 00 = 1
01ecb505084d01050000640100 / 00 = 1
01ecb505084d02040000190100 / 00 = 1
01ecb505084d02050000640100 / 00 = 3
01ecb505084d02090001000000 / 00 = 1
01ecb505084d03020000000000 / 00 = 1
01ecb505084d030500004b0100 / 00 = 2
01ecb505084d06030000000000 / 00 = 1
01ecb505084d060500004b0100 / 00 = 1
01ecb505084d07010000000000 / 00 = 1
01ecb505084d07040000640100 / 00 = 1
01ecb505084d09020000000000 / 00 = 1
01ecb505084d0b010000000000 / 00 = 1
01ecb505084d0b030000000000 / 00 = 2
01ecb505084d0b040000190100 / 00 = 2
01ecb505084d0c030000000000 / 00 = 1
01ecb505084d0c080001000000 / 00 = 1
01ecb505084d0d020000000000 / 00 = 2
01ecb505084d0f010000000000 / 00 = 2
01ecb505084d0f040000000000 / 00 = 1
01ecb505084d0f060001000000 / 00 = 1
01ecb505084d10010000000000 / 00 = 1
01ecb505084d110400004b0100 / 00 = 2
01ecb505084d12020000000000 / 00 = 1
01ecb505084d13040000190100 / 00 = 2
01ecb505084d14010000000000 / 00 = 2
01ecb505084d14050000640100 / 00 = 1
01ecb505084d15030000000000 / 00 = 1
01ecb505084d15050000640100 / 00 = 1
01ecb505084d160500004b0100 / 00 = 1
01ecb505084d17010000000000 / 00 = 1
01ecb505084d17020000000000 / 00 = 3
01ecb505084d17030000000000 / 00 = 1
01ecb505084d17050000640100 / 00 = 2
01ecb505084d1a010000000000 / 00 = 2
01ecb505084d1c010000000000 / 00 = 1
01ecb505084d1c030000000000 / 00 = 1
01ecb505084d1d020000000000 / 00 = 1
01ecb505084d1e010000000000 / 00 = 1
[... 386 lines cut ...]
01ecb505084df1050001000000 / 00 = 1
01ecb505084df20300004b0100 / 00 = 1
01ecb505084df3000000000000 / 00 = 2
01ecb505084df3020000000000 / 00 = 1
01ecb505084df4010000000000 / 00 = 1
01ecb505084df4040000640100 / 00 = 2
01ecb505084df4070001000000 / 00 = 1
01ecb505084df5000000000000 / 00 = 1
01ecb505084df5040000640100 / 00 = 1
01ecb505084df6000000000000 / 00 = 3
01ecb505084df6080001000000 / 00 = 1
01ecb505084df7040000640100 / 00 = 1
01ecb505084df8040000640100 / 00 = 1
01ecb505084df9080001000000 / 00 = 1
01ecb505084dfa000000000000 / 00 = 1
01ecb505084dfa030000000000 / 00 = 1
01ecb505084dfa040000640100 / 00 = 2
01ecb505084dfb040000640100 / 00 = 2
01ecb505084dfc000000000000 / 00 = 2
01ecb505084dfc030000000000 / 00 = 1
01ecb505084dfd010000000000 / 00 = 1
01ecb505084dfe060001000000 / 00 = 1
01ecb505084dff000000000000 / 00 = 1
01ecb505084dff040000640100 / 00 = 1
01ecb505084dff080001000000 / 00 = 1

john30, 2023-06-15:
looks as if the unit is not responding as it should. without proper logging/scan result there is not more to tell

closed 2023-06-15 by john30

UnderPalms, 2023-06-16:
I received now a distinct log for this issue reading the value "Col" from the circuit "sc"

2023-06-16 13:52:51.427 [main debug] >>> read Col
2023-06-16 13:52:51.427 [bus info] send message: 31ecb509030d0400
2023-06-16 13:52:51.427 [network debug] [00025] wait for result
2023-06-16 13:52:51.433 [bus debug] start request 31
2023-06-16 13:52:51.434 [bus debug] arbitration start with 31
2023-06-16 13:52:51.481 [bus debug] arbitration won
2023-06-16 13:52:51.481 [bus debug] arbitration delay 25 micros
2023-06-16 13:52:51.481 [bus debug] switching from ready to send command
2023-06-16 13:52:51.486 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.491 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.496 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.501 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.506 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.511 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.516 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.517 [bus debug] switching from send command to send command CRC
2023-06-16 13:52:51.522 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.522 [bus debug] switching from send command CRC to receive command ACK
2023-06-16 13:52:51.526 [bus debug] switching from receive command ACK to receive response
2023-06-16 13:52:51.530 [bus debug] switching from receive response to receive response CRC
2023-06-16 13:52:51.534 [bus debug] switching from receive response CRC to send response ACK
2023-06-16 13:52:51.539 [bus debug] send/receive symbol latency 4 ms
2023-06-16 13:52:51.540 [update info] sent MS cmd: 31ecb509030d0400 / 00
2023-06-16 13:52:51.542 [update error] unable to parse read sc Col from 31ecb509030d0400 / 00: ERR: invalid position

Does that help anyhow?

UnderPalms, 2023-06-20:
Thanks John, is there any workaround to load a csv that fits bester? The ec.sol.csv (which seems to be the wrong file) is always being loaded automatically

john30, 2023-07-01:
are you sure that the unit is working correctly? since it does answer only on two messages with non-empty response it seems to be rather useless.
the right way to exclude the ec.sol for it would be to adjust the csv to exclude the product number or the HW/SW version

UnderPalms, 2023-07-03:
Yes, I'm sure it's working correctly since another ebus-device (the 700 and display at my gas-heater) is able to visualize the solar yield from the past months. That means there must be a proper communication being established at least among these 2 devices (700 and sc) which means that they must be properly communicating via ebus.