john30/ebusd #936: ebusd data lastup all values 0

msschl, 2023-06-19:
### Description

The data of the ebusd service shows all lastup fields with a 0 value

### Actual behavior

I think no visible errors at start

### Expected behavior

To display the latest data

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--scanconfig=full -d enh:192.168.178.15:9999 -p 8888 --httpport=8889 --htmlpath=/var/ebusd/html -l /var/log/ebusd.log"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

adapter 5 via Ethernet

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2023-06-19 16:31:48.395 [main notice] ebusd 23.1.23.1 started with full scan on enhanced device 192.168.178.15:9999
2023-06-19 16:31:48.919 [bus notice] device status: resetting
2023-06-19 16:31:48.919 [bus notice] bus started with own address 31/36
2023-06-19 16:31:48.936 [bus notice] signal acquired
2023-06-19 16:31:49.245 [bus notice] device status: reset, supports info
2023-06-19 16:31:49.350 [bus notice] device status: extra info: firmware 1.1[3501].1[3501], jumpers 0x0f
2023-06-19 16:31:52.321 [bus notice] new master 10, master count 2
2023-06-19 16:31:52.396 [bus notice] new master 03, master count 3
2023-06-19 16:31:52.396 [update notice] received unknown MS cmd: 1008b5110101 / 09989bf01fff710000ff
2023-06-19 16:31:52.686 [update notice] received unknown MS cmd: 1052b5230103 / 0f91036603008000802b040080c85004
2023-06-19 16:31:52.956 [update notice] received unknown MS cmd: 1008b510090003ffffa0ff010001 / 0101
2023-06-19 16:31:53.217 [update notice] received unknown MS cmd: 1052b523080114ff0014ffffb4 / 0101
2023-06-19 16:31:53.467 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2023-06-19 16:31:58.920 [main notice] starting initial full scan
2023-06-19 16:32:01.551 [bus notice] scan 08: ;Vaillant;BAI00;0105;7803
2023-06-19 16:32:01.551 [update notice] store 08 ident: done
2023-06-19 16:32:01.551 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0105;7803
2023-06-19 16:32:01.551 [bus notice] scan 08: ;Vaillant;BAI00;0105;7803
2023-06-19 16:32:06.806 [bus notice] scan 15: ;Vaillant;70000;0614;6903
2023-06-19 16:32:06.806 [update notice] store 15 ident: done
2023-06-19 16:32:06.806 [update notice] sent scan-read scan.15  QQ=31: Vaillant;70000;0614;6903
2023-06-19 16:32:06.806 [bus notice] scan 15: ;Vaillant;70000;0614;6903
2023-06-19 16:32:06.966 [update notice] received unknown MS cmd: 1008b510090003ffffa0ff010001 / 0101
2023-06-19 16:32:09.474 [main notice] read common config file vaillant/broadcast.csv
2023-06-19 16:32:09.587 [main notice] read common config file vaillant/scan.csv
2023-06-19 16:32:09.721 [main notice] read common config file vaillant/general.csv
2023-06-19 16:32:09.873 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0105, HW7803
2023-06-19 16:32:10.856 [update notice] sent scan-read scan.08 id QQ=31:
[... 86 lines cut ...]
2023-06-19 16:34:42.468 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f48 / 00: ERR: invalid position
2023-06-19 16:34:42.723 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f50 / 00: ERR: invalid position
2023-06-19 16:34:42.958 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f58 / 00: ERR: invalid position
2023-06-19 16:34:43.338 [update notice] received read bai Status01 QQ=10: 75.0;76.5;32.125;-;57.0;off
2023-06-19 16:34:43.630 [update notice] received read vr_70 SensorData QQ=10: 57.44;54.88;-;-;67.12;-;c8 50 04
2023-06-19 16:34:43.908 [update notice] received update-write bai SetMode QQ=10: 3;-;-;160;1;0;0;1;0;0
2023-06-19 16:34:44.168 [update notice] received write vr_70 SetActorState QQ=10: on;-;off;on;-;-;180
2023-06-19 16:34:44.418 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: off;0.0;on;-100
2023-06-19 16:34:52.247 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f00 / 00: ERR: invalid position
2023-06-19 16:34:52.508 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f08 / 00: ERR: invalid position
2023-06-19 16:34:52.747 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f10 / 00: ERR: invalid position
2023-06-19 16:34:52.989 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f18 / 00: ERR: invalid position
2023-06-19 16:34:53.224 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f20 / 00: ERR: invalid position
2023-06-19 16:34:53.468 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f28 / 00: ERR: invalid position
2023-06-19 16:34:53.578 [update notice] received read bai Status01 QQ=10: 75.0;76.5;32.125;-;57.0;off
2023-06-19 16:34:53.758 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f30 / 00: ERR: invalid position
2023-06-19 16:34:54.008 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f38 / 00: ERR: invalid position
2023-06-19 16:34:54.184 [update notice] received read vr_70 SensorData QQ=10: 57.44;54.88;-;-;67.38;-;c8 50 04
2023-06-19 16:34:54.388 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f40 / 00: ERR: invalid position
2023-06-19 16:34:54.647 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f48 / 00: ERR: invalid position
2023-06-19 16:34:54.800 [update notice] received update-write bai SetMode QQ=10: 3;-;-;160;1;0;0;1;0;0
2023-06-19 16:34:54.988 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f50 / 00: ERR: invalid position
2023-06-19 16:34:55.247 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f58 / 00: ERR: invalid position
2023-06-19 16:34:55.388 [update notice] received write vr_70 SetActorState QQ=10: on;-;off;on;-;-;180
2023-06-19 16:34:55.638 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: off;0.0;on;-100

msschl, 2023-06-19:
```
ebusctl grab result
1008b5110100 / 08800417001f080080 = 12
10feb510020601 = 3
1008b5120204ff / 0101 = 3
1008b513020508 / 00 = 3
1008b5100305ff03 / 0101 = 3
1052b5030c0700ffffffffffffffffffff / 0101 = 1
```

msschl, 2023-06-20:
ebusd ip:port/data

```json
{
 "700": {
  "messages": {   "AdaptHeatCurve": {
    "name": "AdaptHeatCurve",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "BankHolidayEndPeriod": {
    "name": "BankHolidayEndPeriod",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "BankHolidayStartPeriod": {
    "name": "BankHolidayStartPeriod",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Friday": {
    "name": "ccTimer.Friday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Monday": {
    "name": "ccTimer.Monday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Saturday": {
    "name": "ccTimer.Saturday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Sunday": {
    "name": "ccTimer.Sunday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Thursday": {
    "name": "ccTimer.Thursday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Tuesday": {
    "name": "ccTimer.Tuesday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ccTimer.Wednesday": {
    "name": "ccTimer.Wednesday",
    "passive": false,
    "write": false,
    "lastup": 0
   },
   "ContinuosHeating": {
    "name": "ContinuosHeating",
    "passive": false,
    "write": false,
    "lastup": 0
[... 2916 lines cut ...]
     "S5": {"value": 46.69},
     "S6": {"value": null},
     "ignore": {"value": "c8 50 04"}
    }
   }
  }
 },
 "global": {
  "version": "23.1.23.1",
  "updatecheck": "OK",
  "signal": true,
  "symbolrate": 23,
  "maxsymbolrate": 108,
  "minarbitrationmicros": 2,
  "maxarbitrationmicros": 178,
  "minsymbollatency": 6,
  "maxsymbollatency": 28,
  "qq": 49,
  "reconnects": 0,
  "masters": 3,
  "messages": 747,
  "lastup": 1687197058
 }
}
```

john30, 2023-06-20:
lastup will only be set for messages that were queried or that arrived due to traffic seen from another device. you can force that by using polling or configure the MQTT integration accordingly

msschl, 2023-06-20:
How would I do that? Setting —pollinterval=5 in the ebusd argument?

john30, 2023-07-01:
you can either query values of intereset individually e.g. by using a cron job or use a single read command with the option to adjust the poll priority, e.g. "ebusctl read -p1 ..."

closed 2023-07-08 by john30