john30/ebusd #937: Split out update checks

tobylorenz, 2023-06-20:
### Description

Hi,

I'm currently setting up systemd security configurations for all my services. I discovered that ebusd does unexpected https requests. I found out that this is because of the default `def_global_updatecheck-payload` lines in `mqtt-hassio.cfg`. I'm wondering why the daemon should automatically update it's own configuration files...

For me this is behavior I don't want from security perspective and in order to have a reliable system without unexpected changes.

So I propose to split out the update functionality in an own executable. If wanted, this can still be run by systemd ExecStartPre. But normally I just update/setup the configuration files once, and as soon as my ebus device is working, I'm keeping this configuration.

With such a split, the pure daemon doesn't need http(s) access and also no write access to configurations at all, which improves maintainability and security overall.

What do you think?
I guess, the config files that get updated are the ones in `/etc/ebusd/ebusd-configuration`?

By the way, this is my systemd configuration extensions:
```
[Service]
# --- Paths ---
# RootDirectory=
# RootImage=
ProtectProc=noaccess
ProcSubset=pid

# --- User/Group Identity ---
# User=
# DynamicUser=
# SupplementaryGroups=

# --- Capabilities ---
# CapabilityBoundingSet=
# AmbientCapabilities==

# --- Security ---
# NoNewPrivileges=

# --- Process Properties ---
UMask=0666
# KeyringMode=

# --- Sandboxing ---
ProtectSystem=full
ProtectHome=true
PrivateTmp=true
PrivateDevices=false
PrivateNetwork=false
# PrivateUsers=
ProtectHostname=true
ProtectClock=true
ProtectKernelTunables=true
ProtectKernelModules=true
ProtectKernelLogs=true
ProtectControlGroups=true
RestrictAddressFamilies=AF_NETLINK AF_UNIX AF_INET AF_INET6
# RestrictNamespaces=
LockPersonality=true
MemoryDenyWriteExecute=true
RestrictRealtime=true
RestrictSUIDSGID=true
# RemoveIPC=
# PrivateMounts=

# --- System Call Filtering ---
SystemCallArchitectures=native
SystemCallFilter=@default @basic-io @file-system @io-event @ipc @network-io @process @signal flock ioctl setsid sysinfo umask

# --- Resource Control ---
IPAddressAllow=202.61.232.73 2a03:4000:30:f9a3::16:5094
IPAddressDeny=any
DeviceAllow=/dev/esera rw
# Delegate=

# --- Service ---
# NotifyAccess=
```

john30, 2023-06-20:
the config files are only grabbed from https when the --configpath is set to an URL which is the case by default. if you don't want that, just point it to local files.
the updatecheck can be disabled as well using cmdline arguments.

tobylorenz, 2023-06-23:
Ok, fair enough.
So I would indeed disable this updatecheck in the config files per default, and would probably request this also as default on distro level as well. I'm using Debian.

So, is there then an easy way to run an update from command line just once? I only see the command line argument `--updatecheck=(on|off)`, but not something to just check and exit. Or is this just a `git pull origin` in the `/etc/ebusd/ebusd-configuration` directory? Is there more than this folder updated?

john30, 2023-07-01:
with local files you'd have to do a git pull followed by ebusctl reload, with files from the webserver ebusdctl reload is enough

closed 2023-07-08 by john30