john30/ebusd #943: strange --mqtttopic handling when / inserted

JanHBade, 2023-06-25:
### Description

setting the topic to --mqtttopic=ebusd/lueftung results in strange behaviour

[image]

### Actual behavior

see picture, the / is ignored

### Expected behavior

subtopic in mqtttopic should supported

### ebusd version

current source from git

### ebusd arguments

sudo ./src/ebusd/ebusd --scanconfig=full -c /etc/ebusd -d /dev/ttyebus --receivetimeout=50000 --pollinterval=30 --mqtthost=isis --mqttport=1883 --mqtttopic=ebusd/lueftung -f

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT generic

### Logs

-none-

john30, 2023-07-01:
is this only for the global topic or for all?

JanHBade, 2023-07-02:
for all

JanHBade, 2023-07-04:
```pi@lueftung:~/ebusd$ sudo ./src/ebusd/ebusd --scanconfig=full -c /etc/ebusd -d /dev/ttyebus --receivetimeout=50000 --pollinterval=30 --mqtthost=isis --mqttport=1883 --mqtttopic=ebusd/lueftung -f
2023-07-04 20:58:49.002 [main notice] ebusd 23.1.23.1-51-ge33810aa started with full scan on device: /dev/ttyebus
2023-07-04 20:58:49.030 [bus notice] bus started with own address 31/36
2023-07-04 20:58:49.067 [bus notice] signal acquired
2023-07-04 20:58:49.070 [mqtt notice] connection established
2023-07-04 20:58:49.073 [mqtt error] subscribe: Invalid function arguments provided.
2023-07-04 20:58:49.803 [bus notice] new master 37, master count 2
2023-07-04 20:58:59.066 [main notice] starting initial full scan
```

[image]

john30, 2023-07-08:
please check again with 43d3599

closed 2023-07-08 by john30

JanHBade, 2023-07-09:
works, thank you

commit 43d3599c (2023-07-08): fix for topic separator check (closes #943)