john30/ebusd #958: wolfberry 23.2 not present in alpine linux repository

xpavli44, 2023-07-15:
### Description

Latest release notes https://github.com/john30/ebusd/releases/tag/23.2
suggest
`For Alpine Linux, run apk --repository=http://dl-cdn.alpinelinux.org/alpine/edge/community add ebusd.`

### Actual behavior

However if I open https://dl-cdn.alpinelinux.org/alpine/edge/community/armv7/
only ebusd-23.1-r1.apk
is present there.

### Expected behavior

@john30 would it be possible to compile and upload 23.2 to the repository please? 

I was wondering why [ebusd addon](https://github.com/LukasGrebe/ha-addons/tree/main/ebusd) from @LukasGrebe for Home Assistant is still running v 23.1 and
[this](https://github.com/LukasGrebe/ha-addons/blob/22172a4522dfd8c399a2f1af4118ee0f5322abc9/ebusd/Dockerfile#L12) seems to be the answer. It is pointing to a correct repo, but the up to date version of the package is not present there.

### ebusd version

23.2

### ebusd arguments

none

### Operating system

other

### CPU architecture

armv7l

### Dockerized

same as ebusd version

### Hardware interface

adapter 5 via WiFi

### Related integration

other

### Logs

no logs

tjorim, 2023-07-15:
I flagged the package as outdated last week, I think we need @tim-devel for this.

tim-devel, 2023-07-16:
Sorry but I don't use the Alpine version any more so am probably not the best person to maintain that version anymore.

It is pretty straight forward to test and release a new version.  Link to the guidance is below.  You just need to change the version number in the APKBUILD file, test it builds and runs and then make a pull request against aports gitlab repo

https://wiki.alpinelinux.org/wiki/Creating_an_Alpine_package
https://gitlab.alpinelinux.org/alpine/aports

tim-devel, 2023-07-16:
@lukasgrebe

xpavli44, 2023-07-16:
@tim-devel I've made the MR https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/48862 but checks are failing with 
`>>> ERROR: ebusd: ebusd-23.2.tar.gz is missing in checksums
`
any idea how to fix it pls?

tim-devel, 2023-07-16:
You need to run the command below to update the checksum in the APKBUILD

abuild checksum

xpavli44, 2023-07-16:
thanks to [@psykose](https://github.com/psykose) it should be done now https://gitlab.alpinelinux.org/alpine/aports/-/commit/a7abee0748bf5c5e9e95bc8fb5c78489917b644d

closed 2023-07-16 by xpavli44