john30/ebusd #960: MQTT writing doesn't work when --mqtttopic starts with "/"

knalltuete86, 2023-07-17:
### Description

ebusd does not react anyhow while writing to MQTT-Topic (e.g. from HomeAssistant).

e.g. `--mqtttopic=/Home/Heatpump/%circuit/%name`

### Actual behavior

Publishing to MQTT-Broker works as expected, subscribing from `/set`-Topic doesn't.

There is no log message during startup or any other hint why the prefixed "/" may cause an issue.

### Expected behavior

Either exit with error code during startup asking the user to fix `--mqtttopic` (remove "/") or get subscription working the same way as publishing do.

### ebusd version

23.2

### ebusd arguments

EBUSD_OPTS="--configpath=/etc/ebusd/configs --device=ens:192.168.178.59:9999 --mqttport=1883 --mqttuser=USER --mqttpass=PASS --mqtttopic=/Home/Heatpump/%circuit/%name --mqttjson --mqttvar=filter-direction=r|u|^w --pollinterval=15

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 5 via WiFi

### Logs

no message about mqtt subscription at loglevel debug

knalltuete86, 2023-07-17:
See related #954

john30, 2023-07-29:
afaik a beginning "/" is not supposed be part of the topic anyway

knalltuete86, 2023-07-29:
That's why it was so strange to find the root cause of this issue :)
I expected it should work as there was no issue while reading values etc.

I guess it's its a minor problem for most of the users - but this issue shall help others may running into the same problem.

commit 949b366f (2023-08-06): deny mqtttopic starting with '/' (fixes #960)
closed 2023-08-06 by john30