john30/ebusd #965: ebusd telling signal connected while ebusd can't connect to ebus network interface

eazy, 2023-07-26:
### Description

From time to time my ebus network interface crashes and ebusd can't connect to it. 

But the MQTT Message of ebusd tells signal = true

After Powercycling the interface everything works fine again. So I need to detect that problem, to auto powercyle the intreface.

### Actual behavior

Can't detect ebusd is realy working because of the wrong signal information.

### Expected behavior

signal should be false
maybe there should be an additional connection state regarding the ebus interface and not just the ebus signal itself.

### ebusd version

23.2

### ebusd arguments

--mqttvar=filter-direction=r|u|^w

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.0 Ethernet

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-07-26 21:02:35.828 [bus error] unable to open 192.168.222.107:9999: ERR: generic I/O error
2023-07-26 21:02:35.828 [bus notice] device invalid

john30, 2023-07-29:
which firmware version is it?

eazy, 2023-07-30:
device status: extra info: firmware 1.1[8419].2[847f], jumpers 0x37

john30, 2023-08-06:
can you ping the ip in this case?
what is the status of the LEDs on the adapter and the W5500?

eazy, 2023-08-07:
the device is responding to icmp.
Adapter LED States:
yellow: on
blue: on
green: normal flickering
red: off

WS550 LED States:
red: on

john30, 2023-08-15:
please check with the updated source as in e3abbf1

john30, 2024-04-27:
closed due to inactivity

closed 2024-04-27 by john30

MartinMichaux, 2025-01-14:
Hey,
I am actually also having this issue. My ebus adapter is connected to the vr94f network module. I receive signals like:

# Try to telnet to the adapter
telnet 192.168.1.100 9999
Trying 192.168.1.100...
Connected to 192.168.1.100.
Escape character is '^]'.
ƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƵƴ
1ƶ*ƽƪƪƪƪƪƪƪƪƪƪƪƪƪƪƵƉ    ?7ƀǿǿǿƪƪƪƪƪvƵ   Aǿ=ǹǿaǿƪƪƪƪƪvƵǼ
ǿǿƪƪƪƪƵ GǿǿǿǢƚƪƪƪƪƪvƵ   ǿǿǿƅƚƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪvƵƶǿǓƪƪƪƪƪƪƪƪƪƵƴ
2ƶ*3ƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪǱƵ$ƪƪƪǱƵ$ǚƕƪƵƴ
2ƶ*3ƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƵƴ
3Ʒ*ǔƪƪƪƪƪƵƉ     ?6ƀǿǿǿƩƪƪƪƪƪvƵ  Aǿ*ǹǿaǿ?ƪvƵƶǿǓƪƪƪƪvƵǼ
                                                      ǿơƪƪƪƪƵ   GǿǿǿǢƚƪƪƪƪƪvƵ   ǿǿǿƅƚƪƪƪƪƪƪƪƪƪƵ|2Ơ|ǀǏǀ{<ưƪƪƪƪƪƪƪƪƵ}3:}ǀƋǰƪƪƪƪƪƪƪƵ~4
                                                                                                                                   ~Ǐ ǀ!9ƪƪƪƪƪƪƪƵ5ƑƐƑƪƪƪƪƪƪƪƪƵƀ6ǩƀǌƪƪƪƪƪƪƪƪƪƵƴ
3Ʒ*ǔƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪƪ

and there doesnt seem to be data parser for it.
Could you help ?

Thanks in advance.

john30, 2025-01-26:
> Hey, I am actually also having this issue. My ebus adapter is connected to the vr94f network module. I receive signals like:
> # Try to telnet to the adapter
>
> telnet 192.168.1.100 9999 Trying [...]

misplaced here on this already closed issue, but anyway: look at this image here:
https://adapter.ebusd.eu/v5-c6/index.en.html#connections
as of today, you need to have ebusd running somewhere in the network for interpreting the traffic