john30/ebusd #975: Home Assistant Update 2024.2.0. Detected entities with a name starting with the device name

edudele, 2023-08-07:
### Description

Need update or modify mqtt-hassio.cfg
[image]

john30, 2023-08-15:
great, phantastic move of HA. the device name is added as prefix in order to assure that is unique, if they remove it now, it will no longer be unique - brilliant. I'll search HA issue tracker if another developer sees it the same way

john30, 2023-08-15:
please check if the warning is gone and the entities remain as previously when changing line 137 of mqtt-hassio.cfg from this:
`   "name":"%prefixn %circuit",`
to this:
`   "name":"%circuit",`

if that yields in the desired result, users of HA before 2023.8 would have to use the old version and everyone else the new version.

edudele, 2023-08-16:
After the change the warning seems to disappear but this one still appears:

[image]

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 10:19:20 (5 occurrences)
Last logged: 10:19:20

MQTT entity name starts with the device name in your config {'unique_id': 'ebusd_global_running', 'device': {'identifiers': ['ebusd'], 'manufacturer': 'ebusd.eu', 'name': 'ebusd', 'sw_version': '23.2', 'suggested_area': 'Heating', 'connections': []}, 'state_topic': 'ebusd/global/running', 'name': 'ebusd running', 'device_class': <BinarySensorDeviceClass.RUNNING: 'running'>, 'payload_on': 'true', 'payload_off': 'false', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'availability_mode': 'latest', 'force_update': False, 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'running'
MQTT entity name starts with the device name in your config {'unique_id': 'ebusd_global_signal', 'device': {'identifiers': ['ebusd'], 'manufacturer': 'ebusd.eu', 'name': 'ebusd', 'sw_version': '23.2', 'suggested_area': 'Heating', 'connections': []}, 'state_topic': 'ebusd/global/signal', 'name': 'ebusd signal', 'device_class': <BinarySensorDeviceClass.CONNECTIVITY: 'connectivity'>, 'payload_on': 'true', 'payload_off': 'false', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'availability_mode': 'latest', 'force_update': False, 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'signal'
MQTT entity name starts with the device name in your config {'unique_id': 'ebusd_global_uptime', 'device': {'identifiers': ['ebusd'], 'manufacturer': 'ebusd.eu', 'name': 'ebusd', 'sw_version': '23.2', 'suggested_area': 'Heating', 'connections': []}, 'state_topic': 'ebusd/global/uptime', 'name': 'ebusd uptime', 'state_class': <SensorStateClass.TOTAL_INCREASING: 'total_increasing'>, 'unit_of_measurement': 's', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'availability_mode': 'latest', 'force_update': False, 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'uptime'
MQTT entity name starts with the device name in your config {'unique_id': 'ebusd_global_updatecheck', 'device': {'identifiers': ['ebusd'], 'manufacturer': 'ebusd.eu', 'name': 'ebusd', 'sw_version': '23.2', 'suggested_area': 'Heating', 'connections': []}, 'state_topic': 'ebusd/global/updatecheck', 'name': 'ebusd updatecheck', 'value_template': Template<template=({% set my_new = value_json|truncate(255)|regex_replace(find=',.*| available|revision v|version v|OK',replace='') %}{% if my_new == '' %}{% set my_new = '23.2' %}{% endif %}{{ {'installed_version':'23.2','latest_version':my_new,'entity_picture':'https://ebusd.eu/logo-32x32.png','release_url':'https://github.com/john30/ebusd/releases/latest'} | tojson }}) renders=0>, 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'availability_mode': 'latest', 'retain': False, 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'updatecheck'
MQTT entity name starts with the device name in your config {'unique_id': 'ebusd_global_scan', 'device': {'identifiers': ['ebusd'], 'manufacturer': 'ebusd.eu', 'name': 'ebusd', 'sw_version': '23.2', 'suggested_area': 'Heating', 'connections': []}, 'state_topic': 'ebusd/global/scan', 'name': 'ebusd scan', 'qos': 0, 'payload_available': 'online', 'encoding': 'utf-8', 'availability_mode': 'latest', 'force_update': False, 'payload_not_available': 'offline', 'enabled_by_default': True}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'scan'

edudele, 2023-08-17:
The warning problem disappears, but a bigger problem comes, we are left without history.

But the unique_id are renamed, losing all the history of Home Assistant, InfluxDB, etc...
Original: number.ebusd_basv0_hwctempdesired_tempv
With Change: number.**basv0**_ebusd_basv0_hwctempdesired_tempv

Make it clear that the creation of the ID is generated by HA based on the device name adding the name prefix.

We can't keep the unique_id as it was and comply with the new requirements?

Danit2, 2023-08-17:
You must change the line 312 to   "name":"%name %fieldname",

Then it works and only some global error rest. 
This i try to remove also.

Danit2, 2023-08-17:
For the global errors you must change the line 341 to
  "name":"%name"

That have fix all the errors for me.

edudele, 2023-08-17:
> For the global errors you must change the line 341 to "name":"%name"
> 
> That have fix all the errors for me.

With those changes that you comment, does it change the ids of the entities?
​

Danit2, 2023-08-18:
No, for me not. 
The entity id is the same and the history is stil there.

estegewr, 2023-08-24:
The change was rolled back in HA release [2023.8.4](https://github.com/home-assistant/core/pull/98768). Therefore, no warning occurs anymore even with the original mqtt configuration.

edudele, 2023-08-25:
Close Remove repair issue for MQTT discovered items #98768

closed 2023-08-25 by edudele

Danit2, 2023-08-27:
No, the Changes is not rolled back. 
Only the repair messgae is gone.

but the developers of programs still need to have this problem fixed by version 2024.2.

Quote: "[The goal of getting MQTT naming up to par with the rest of the Home Assistant integrations remains.](https://github.com/home-assistant/core/pull/98768#:~:text=The%20goal%20of%20getting%20MQTT%20naming%20up%20to%20par%20with%20the%20rest%20of%20the%20Home%20Assistant%20integrations%20remains%3B)"

so, i would be interested to know if my changes have fixed the bug or not?
@edudele?

If so, can I do a PR to implement this. 

Thanks.

edudele, 2023-08-27:
I think the change is valid.
the ids are kept and the warning was fixed.