john30/ebusd #976: Value not retrieved when using the http client interface

teclab-at, 2023-08-07:
### Description

When sending a http request, the json data is retrieved without the value.
ebusd 23.2.23.2

### Actual behavior

Restart the service:
`systemctl restart ebusd.service`

HTTP Request:
`curl http://ebusd/data/700/hwcstoragetemp?exact=true&maxage=0`
Returns:
```
 "700": {
  "messages": {   "HwcStorageTemp": {
    "name": "HwcStorageTemp",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "global": {
  "version": "23.2.23.2",
  "signal": true,
  "symbolrate": 51,
  "maxsymbolrate": 119,
  "minarbitrationmicros": 1,
  "maxarbitrationmicros": 9,
  "minsymbollatency": 7,
  "maxsymbollatency": 9,
  "qq": 49,
  "reconnects": 0,
  "masters": 4,
  "messages": 628,
  "lastup": 0
 }
}
```

Now do:
`ebusctl read -c 700 HwcStorageTemp`
Returns:
`53`

Now HTTP returns correct value:
`curl http://ebusd/data/700/hwcstoragetemp?exact=true&maxage=0`
```
"700": {
  "messages": {   "HwcStorageTemp": {
    "name": "HwcStorageTemp",
    "passive": false,
    "write": false,
    "lastup": 1691439861,
    "zz": 21,
    "fields": {
     "tempv": {"value": 53}
    }
   }
  }
 },
 "global": {
  "version": "23.2.23.2",
  "updatecheck": "OK",
  "signal": true,
  "symbolrate": 23,
  "maxsymbolrate": 119,
  "minarbitrationmicros": 1,
  "maxarbitrationmicros": 21,
[... 202 lines cut ...]
2023-08-07 20:31:18.156 [update notice] received update-write hmu SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0
2023-08-07 20:31:18.423 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-08-07 20:31:21.938 [update notice] received unknown MS cmd: 03e0b5210900002200e4ffff4600 / 0b0000ea00c5000700000600
2023-08-07 20:31:26.993 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2023-08-07 20:31:27.125 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2023-08-07 20:31:27.256 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2023-08-07 20:31:27.397 [update notice] received read hmu Status01 QQ=10: 22.0;21.5;12.625;-;53.0;off
2023-08-07 20:31:27.523 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2023-08-07 20:31:27.523 [main error] scan config 76: ERR: invalid position
2023-08-07 20:31:27.708 [update notice] received unknown MS cmd: 1076b5110101 / 09ff2ba00cff6a0000ff
2023-08-07 20:31:27.974 [update notice] received unknown MS cmd: 1076b512030f0001 / 0756030000801303
2023-08-07 20:31:28.241 [update notice] received update-write hmu SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0
2023-08-07 20:31:28.508 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-08-07 20:31:36.698 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2023-08-07 20:31:36.830 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2023-08-07 20:31:36.961 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2023-08-07 20:31:37.093 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2023-08-07 20:31:37.093 [main error] scan config 76: ERR: invalid position
2023-08-07 20:31:37.449 [update notice] received read hmu Status01 QQ=10: 22.0;21.5;12.625;-;53.0;off
2023-08-07 20:31:37.716 [update notice] received unknown MS cmd: 1076b5110101 / 09ff2ba00cff6a0000ff
2023-08-07 20:31:37.983 [update notice] received unknown MS cmd: 1076b512030f0001 / 0756030000801303
2023-08-07 20:31:38.249 [update notice] received update-write hmu SetMode QQ=10: auto;0.0;-;-;1;1;1;0;0;0
2023-08-07 20:31:38.516 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-08-07 20:31:45.843 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2023-08-07 20:31:45.974 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position

john30, 2023-08-15:
can't reproduce. does the message have an access level associated with it?

teclab-at, 2023-08-17:
Don't know what you mean with "access level". If I set the item to r1, then I also get it with the http method.
So my assumption is, that http only retrieves from the cache, but does not query values from the modbus?

Silverstar, 2023-08-30:
On http, you need to query with "required" parameter (and possibly "maxage") otherwise ebusd just answers from cache: https://github.com/john30/ebusd/wiki/3.2.-HTTP-client

john30, 2023-09-03:
did you pass the URL arg to curl as posted above?
as in this case the shell will run parts of it in background.
you'd have to wrap the URL in quotes like this: `curl "http://ebusd/data/700/hwcstoragetemp?exact=true&maxage=0"`

john30, 2024-04-27:
closed due to inactivity

closed 2024-04-27 by john30