john30/ebusd #979: Typo in one of the csv Files

pene8, 2023-08-08:
### Description

in my config the configpath is:  --configpath=http://ebusd.eu/config/ 
when I read the config of my ebus the retrieve the hwc-timer I'll get the list:
 sudo ebusctl f | grep  hwcTimer.W
700 hwcTimer.Wednesday = ...

this shoud be 
700 hwcTimer.Wendesday = ....

Please correct the Typo in the .csv file.
Thankoyu 

### Actual behavior

sudo ebusctl f | grep  hwcTimer.W
700 hwcTimer.Wednesday = ....

### Expected behavior

sudo ebusctl f | grep  hwcTimer.W
700 hwcTimer.Wendesday = ....

### ebusd version

23.1

### ebusd arguments

 --configpath=http://ebusd.eu/config/ 

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.1 Ethernet

### Related integration

HTTP

### Logs

none

estegewr, 2023-08-13:
Everything is fine, no changes needed: Correct spelling is "Wednesday" as used in the configs.

john30, 2023-08-15:
indeed, Wednesday is the correct spelling

closed 2023-08-15 by john30