john30/ebusd #990: Lots of errors on v5 adapter via WiFi

Xyaren, 2023-08-21:
### Description

My brandnew v5 adapter arrived today 🎉 .
So I flashed the latest firmware,setup Wifi and connected it to my Vaillant Burner.
On my local linux server I created an ebusd docker container and looked at the logs.

Also commands like `ebusdctl scan 09` result in those errors (with varying delay betwen the command and the error).

What I tried (unseccesfull)
- Reflash firmware
- Flash older firmware
- 4 Different power supply
- Thicker & Shorter eBUS Cable (about 15cm long)
- Chaning the arbitration value via the web ui

Then I tried using my laptop and connecting via USB. 
**Suddenly all errors were gone.**

Docker compose ebusd-definition:
```yaml
ebusd:
  image: john30/ebusd
  restart: unless-stopped
  environment:
    EBUSD_SCANCONFIG: ""
    EBUSD_DEVICE: "ens:192.168.2.74:9999"
    EBUSD_MQTTHOST: "mosquitto"
    EBUSD_MQTTPORT: 1883
    EBUSD_MQTTINT: "/etc/ebusd/mqtt-hassio.cfg"
    EBUSD_MQTTJSON: ""
  networks:
    mqtt:
    bridge:
```

Ping (Server -> Adapter:
```
root@lx0001:/services/docker/smarthome# ping 192.168.2.74
PING 192.168.2.74 (192.168.2.74) 56(84) bytes of data.
64 bytes from 192.168.2.74: icmp_seq=1 ttl=255 time=47.5 ms
64 bytes from 192.168.2.74: icmp_seq=2 ttl=255 time=1.95 ms
64 bytes from 192.168.2.74: icmp_seq=3 ttl=255 time=1.44 ms
64 bytes from 192.168.2.74: icmp_seq=4 ttl=255 time=2.29 ms
64 bytes from 192.168.2.74: icmp_seq=5 ttl=255 time=2.07 ms
64 bytes from 192.168.2.74: icmp_seq=6 ttl=255 time=2.03 ms
64 bytes from 192.168.2.74: icmp_seq=7 ttl=255 time=1.99 ms
64 bytes from 192.168.2.74: icmp_seq=8 ttl=255 time=2.32 ms
64 bytes from 192.168.2.74: icmp_seq=9 ttl=255 time=1.63 ms
64 bytes from 192.168.2.74: icmp_seq=10 ttl=255 time=1.50 ms
64 bytes from 192.168.2.74: icmp_seq=11 ttl=255 time=1.56 ms
64 bytes from 192.168.2.74: icmp_seq=12 ttl=255 time=1.63 ms
64 bytes from 192.168.2.74: icmp_seq=13 ttl=255 time=2.16 ms
64 bytes from 192.168.2.74: icmp_seq=14 ttl=255 time=1.61 ms
64 bytes from 192.168.2.74: icmp_seq=15 ttl=255 time=1.57 ms
64 bytes from 192.168.2.74: icmp_seq=16 ttl=255 time=1.55 ms
64 bytes from 192.168.2.74: icmp_seq=17 ttl=255 time=1.77 ms
64 bytes from 192.168.2.74: icmp_seq=18 ttl=255 time=1.85 ms
64 bytes from 192.168.2.74: icmp_seq=19 ttl=255 time=1.58 ms
64 bytes from 192.168.2.74: icmp_seq=20 ttl=255 time=1.61 ms
64 bytes from 192.168.2.74: icmp_seq=21 ttl=255 time=1.60 ms
64 bytes from 192.168.2.74: icmp_seq=22 ttl=255 time=2.00 ms
64 bytes from 192.168.2.74: icmp_seq=23 ttl=255 time=2.27 ms
64 bytes from 192.168.2.74: icmp_seq=24 ttl=255 time=2.09 ms
64 bytes from 192.168.2.74: icmp_seq=25 ttl=255 time=1.72 ms
64 bytes from 192.168.2.74: icmp_seq=26 ttl=255 time=2.17 ms
64 bytes from 192.168.2.74: icmp_seq=27 ttl=255 time=2.19 ms
^C
--- 192.168.2.74 ping statistics ---
27 packets transmitted, 27 received, 0% packet loss, time 26024ms
rtt min/avg/max/mdev = 1.438/3.541/47.505/8.626 ms
[... 436 lines cut ...]
2023-08-21 18:31:56.247 [update notice] sent poll-read f43 Hc1OPMode QQ=31: auto
2023-08-21 18:31:57.320 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:01.306 [update notice] received poll-read bai Status01 QQ=10: 31.0;32.0;-;30.0;31.0;off
2023-08-21 18:32:02.248 [update notice] sent poll-read f43 Hc1QuickVetoTemp QQ=31: 19.0
2023-08-21 18:32:03.298 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;59.0
2023-08-21 18:32:07.328 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:08.248 [update notice] sent poll-read f43 Hc1RoomTempSwitchOn QQ=31: thermostat
2023-08-21 18:32:11.348 [update notice] received poll-read bai Status01 QQ=10: 31.0;32.0;-;30.0;31.0;off
2023-08-21 18:32:13.288 [update notice] received update-read broadcast outsidetemp QQ=10: 24.312
2023-08-21 18:32:14.103 [bus error] poll f43 HolidayRoomTemp failed: ERR: SYN received
2023-08-21 18:32:17.321 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:20.247 [update notice] sent poll-read f43 HwcActualTempDesired QQ=31: 60.0
2023-08-21 18:32:21.320 [update notice] received poll-read bai Status01 QQ=10: 31.0;32.0;-;30.0;31.0;off
2023-08-21 18:32:23.308 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;59.0
2023-08-21 18:32:25.358 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:26.248 [update notice] sent poll-read f43 HwcManualOPTempDesired QQ=31: 60.0
2023-08-21 18:32:31.327 [update notice] received poll-read bai Status01 QQ=10: 31.0;32.0;-;30.0;31.0;off
2023-08-21 18:32:32.093 [bus error] poll f43 HwcOPMode failed: ERR: SYN received
2023-08-21 18:32:33.299 [update notice] received unknown BC cmd: 10feb505020400
2023-08-21 18:32:35.340 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:38.104 [bus error] poll f43 HwcQuickVetoTemp failed: ERR: wrong symbol received
2023-08-21 18:32:41.351 [update notice] received poll-read bai Status01 QQ=10: 31.0;32.0;-;30.0;31.0;off
2023-08-21 18:32:44.113 [bus error] poll f43 HwcTempDesired failed: ERR: SYN received
2023-08-21 18:32:45.337 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;0;0;0;0
2023-08-21 18:32:50.115 [bus error] poll f43 OutsideTemp failed: ERR: wrong symbol received

john30, 2023-08-22:
what is the wifi signal strength as of the adapter web UI? it should be at least 70%.

the max ping round trip time of 47.5 ms is problematic and you need to get that down significantly.

basically rtt needs to be below 4 ms in order to have good enough transfer of eBUS data.

you might want to erase the user data from the adapter (using the erase tool on the firmware page), reflash it, and connect to your wifi when it is physically located at the target position so that the wifi calibration will take that into account.

CommanderROR9, 2023-08-22:
So, I checked an mine had a pretty weak Wifi (42%) despite being just 2m from the Router. I guess it didn't like my Mesh Network.
I connected it to a different Wifi and now it has 100% and perfect connection but the HA ebus daemon shows the same errors and the device log itself keeps throwing my_client_socket_send: send: errno 11 which doesn't tell me anything.
It does provide some data in Home Assistant, but only for the VR71 Device and some bits and pieces beyond, nothing for the VR921, the Arotherm Plus or the Unitower.

JosemyAB, 2023-08-22:
There are more people with the same problem. I opened this thread some weeks ago: https://github.com/john30/ebusd-esp32/issues/3. 
I just wanted to put it here for documentation

frankkaktus, 2023-08-29:
I had the same error message. With a distance to the router of about 4m, the receiving level was at -76dBm (46%). For normal communication an SNR of over 20 dB should be good enough for stable connections, but the ping loop time goes up to 150ms.
The problem was solved by changing the power supply cable! The receiving level goes to -48dBm (100%) without any error messages again. Ping was in 90% by under 2ms, but exception up to 30ms...
Could be, that this is only one possible reason for such a problem, but is works with me...

Xyaren, 2023-08-30:
My pings are below 3 ms except a few 40 ms peaks. 
This could by anything on or off my network and something I would say is normal on a wifi in a populated area.

john30, 2023-09-03:
wrong repo and tracked further in the right one anyway (see [ebusd-esp32](https://github.com/john30/ebusd-esp32/issues/3)), thus closed

closed 2023-09-03 by john30