john30/ebusd #993: Log file is being filled up

HeikoGr, 2023-08-25:
### Description

i found a log file in the size of >1 GB and deleted it.
after a restart of the ebusd daemon a new log file was created and filled up.
despite it's not requested by the EBUSD_OPTS (in "etc/default/ebusd")

i use a self compiled version (23.2.23.2-5-ge3abbf1b) of ebusd without any modification.

### Actual behavior

log file is created and filled up.

### Expected behavior

as far as i understand the log file should only be filled up if the option 

- -l,
- --logfile=FILE, or
- -l, --lograwdatafile=FILE

is set

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="--scanconfig --accesslevel=* --device=ens:192.168.178.133:9999 --mqtthost=home.fritz.box --mqttport=1883 --mqtttopic=ebusd/%circuit/%name --mqttjson"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via WiFi

### Related integration

MQTT generic

### Logs

well... the problem is not in the content of the log file. it's the existence of the log-file :-D

john30, 2023-09-03:
its the default being used as (documented)[https://github.com/john30/ebusd/wiki/2.-Run#log-options]. you can set it to empty explicitly to switch to syslog and then make syslog drop that. or you could install logrotate, for which a config file is also part of the installation

HeikoGr, 2023-09-03:
I thought the target ('var/log/ebusd.log') is the default, not the writing of the logfile itself.

john30, 2023-09-03:
then where would be the switch to turn it off?
anyway, it is a daemon and as such needs to have logging options. if you don't want that, change it to syslog and drop the messages there. or you might as well use /dev/null as target file, but then you're basically blind.
or you could set the log levels to error only

closed 2023-09-03 by john30

HeikoGr, 2023-09-03:
I thought the log would only be created if the logfile option is set. the option would then be the switch. But i understand that a daemon should have a default logging. I will setup logrotate and only error logging. 

Thank you!