john30/ebusd #997: Error reading config files

tjarb, 2023-08-29:
### Description

2023-08-29 14:16:37.618 [main notice] replaced old configPath http://ebusd.eu/config/ with new one: http://cfg.ebusd.eu/
2023-08-29 14:16:37.618 [main notice] ebusd 23.2.23.2 started with full scan on device: IP:9999, enhanced
2023-08-29 14:16:47.628 [network error] HTTP send: timed out after 4 sec
2023-08-29 14:16:47.628 [main error] HTTP failure, repeating: send error
2023-08-29 14:16:48.628 [network error] HTTP connect: 537456642=system lib
2023-08-29 14:16:48.628 [main error] error reading config files from http://cfg.ebusd.eu/: ERR: element not found, 

### Actual behavior

Download of config files?

### Expected behavior

Expected to work?

### ebusd version

current source from git

### ebusd arguments

--scanconfig=full --pollinterval=180 --latency 5000 --acquiretimeout=15000 --configpath=http://ebusd.eu/config/ --device=ens:IP:9999 --mqtthost=192.168.3.90 --mqttport=1883 --mqtttopic=vaillant --mqttretain --mqttinsecure

### Operating system

other

### CPU architecture

armv7l

### Dockerized

latest

### Hardware interface

adapter 5 via WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

not present

tjarb, 2023-08-29:
Maybe a real IP could help, instead of 'ens:IP:9999'

john30, 2023-09-03:
can't reproduce. maybe related to IPv6. do you have an v4 address on the host or a v6?

john30, 2024-04-27:
closed due to inactivity

closed 2024-04-27 by john30