john30/ebusd #1001: Stable connection only with --loglevel=debug

alpxp, 2023-09-01:
### Description

I have a weird situation where ebusd works stable only when I add `--loglevel=debug` to the arguments. Without the debug flag it takes very long to discover the device and most of the communications end up with `ERR: read timeout`. Must be some hidden configuration differences between debug and production modes?

### Actual behavior

with 
`EBUSD_OPTS="--scanconfig -d ens:/dev/ttyACM0 --scanconfig=08  -c /etc/ebusd/"
`
the device discovery may take very long and most (but not all) of reads/writes fail, i.e.

`$ ebusctl read -f -c recov TempOutsideAir
`
end up with 
`ERR: read timeout
`
see Log-no-debug below.

When I add `--loglevel=debug` to the EBUSD_OPTS, the communication is stable.
i.e. with 
`-d ens:/dev/ttyACM0 --loglevel=debug --scanconfig=08 -c /etc/ebusd/`
 most of reads are successfully.
see Log-debug below.

I tried playing with `latency` and `receivetimeout` parameters for production runs but saw no improvements. i.e. to be concrete `--latency=10 --receivetimeout=50` does not help. 

### Expected behavior

The communication quality is not expected to depend on log level. Also a stable communication with with the adapter is expected.

### ebusd version

23.2

### ebusd arguments

EBUSD_OPTS="--scanconfig -d ens:/dev/ttyACM0 --scanconfig=08 -c /etc/ebusd/"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via USB

### Logs

Log-no-debug:
```
alpxp@raspberrypi4:~ $ sudo ebusd -f -d ens:/dev/ttyACM0 --scanconfig=08 -c /etc/ebusd/
2023-09-01 11:41:31.053 [main notice] ebusd 23.2.23.2 started with single scan on device: /dev/ttyACM0, enhanced high speed
2023-09-01 11:41:31.058 [bus notice] device status: resetting
2023-09-01 11:41:31.058 [bus notice] bus started with own address 31/36
2023-09-01 11:41:31.069 [bus notice] signal acquired
2023-09-01 11:41:31.258 [bus notice] device status: reset, supports info
2023-09-01 11:41:31.369 [bus notice] device status: extra info: firmware 1.1[380f].1[380f], jumpers 0x03
2023-09-01 11:41:41.059 [main notice] starting initial scan for 08
2023-09-01 11:41:41.182 [main error] initial scan failed: ERR: read timeout
2023-09-01 11:41:50.650 [main notice] starting initial scan for 08
2023-09-01 11:41:50.786 [bus notice] new master 03, master count 2
2023-09-01 11:41:50.786 [bus notice] scan 08: ;Vaillant;recoV;0229;0203
[... 489 lines cut ...]
2023-09-01 11:13:36.271 [bus debug] arbitration won
2023-09-01 11:13:36.271 [bus debug] arbitration delay 17 micros
2023-09-01 11:13:36.271 [bus debug] switching from ready to send command
2023-09-01 11:13:36.275 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.279 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.284 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.288 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.292 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.296 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.301 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.301 [bus debug] switching from send command to send command CRC
2023-09-01 11:13:36.305 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.305 [bus debug] switching from send command CRC to receive command ACK
2023-09-01 11:13:36.310 [bus debug] switching from receive command ACK to receive response
2023-09-01 11:13:36.341 [bus debug] switching from receive response to receive response CRC
2023-09-01 11:13:36.345 [bus debug] switching from receive response CRC to send response ACK
2023-09-01 11:13:36.349 [bus debug] send/receive symbol latency 4 ms
2023-09-01 11:13:36.349 [update info] sent MS cmd: 3108b50903290c0e / 050c0e310100
2023-09-01 11:13:36.350 [update notice] sent read recov TempOutsideAir QQ=31: 19.06;ok
2023-09-01 11:13:36.350 [bus debug] notify request: done
2023-09-01 11:13:36.350 [bus debug] read res: 050c0e310100
2023-09-01 11:13:36.350 [bus debug] switching from send response ACK to send SYN
2023-09-01 11:13:36.350 [main info] read recov TempOutsideAir: 19.06;ok
2023-09-01 11:13:36.350 [main debug] <<< 19.06;ok
```

john30, 2023-09-03:
I can only imagine that the timing is special for this recov device. you could check that now with the current firmware as stated here:
https://github.com/john30/ebusd-esp32/issues/14

alpxp, 2023-09-03:
Here is the output of the `ebus -v`
```executeCommand: executing command: ebus -v
error: unable to load from NVS
init_ebus: already switched to enhanced eBUS mode on TCP port. updating verbosity.
init_ebus: clock frq 80000000 div 33 1/3, src clk 4
init_ebus: detected: 2412 Hz on 18 edges with 1989 +/1989 - edge width, 969 H/1019 L pulse width, i.e. 404 us H/425 us L period
```
I don't know how to judge. Does this look ok?

s0129, 2023-09-04:
This is strange, I have also been having problems and just tried adding `--loglevel=debug` and it has also sprung to life!

I am also using an adapter 5 via USB with a Vaillant boiler `initial scan result: 08;Vaillant;BAI00;0202;8001`.

I adjusted the baud rate using the new feature (+24) and it didn't seem to make a difference.
```
init_ebus: clock frq 80000000 div 33 1/3, src clk 4
init_ebus: detected: 2424 Hz on 19 edges with 1979 +/1979 - edge width, 1001 H/977 L pulse width, i.e. 417 us H/407 us L period
init_ebus: master 8 0x31 delay=198 us [197-203]
init_ebus: master 11 0x03 delay=147 us [147-197]
```

john30, 2023-09-17:
I've found an issue hidden in the ESP32-C3 UART that reports a falsy non-TX-readiness. this seems to be an issue especially if ebusd handles the traffic too quickly as it seems to be the case on your host without the debug logging.

could you check with the [new firmware 20230917](https://adapter.ebusd.eu/v5/ChangeLog#20230917) just released?

alpxp, 2023-09-17:
I confirm that with the [new firmware](https://adapter.ebusd.eu/v5/ChangeLog#20230917) the issue is resolved. Quick test showed identically good behaviour with and without debug log.
Thank you

closed 2023-09-17 by alpxp