john30/ebusd #1005: scanconfig / autoscan can't be disabled when using docker-compose

LordShadowen, 2023-09-06:
### Description

There doesn't seem to be any way to disable "scanconfig" when using Docker Compose.

When using the command line, it seems to be off by default unless we use --scanconfig to enable it.
But in docker-compose file, regardless if EBUSD_SCANCONFIG  is present or not, it always runs.

This is probably because the Dockerfile has this line at the end

```
CMD ["--scanconfig"]
```

Without a parameter to explicitly disable it, and by being assumed, there is no "nice" way to do it.
One workaround I've found is to override the CMD with something else that is not really needed, just to remove it.

e.g. 
```
command: "-c /etc/ebusd/configuration"
```

but it's a hack...

### Actual behavior

If EBUSD_SCANCONFIG is not specified, scanconfig is still active

### Expected behavior

If EBUSD_SCANCONFIG is not specified, there should not be any auto scan

### ebusd version

23.2

### ebusd arguments

n/a

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 5 via USB

### Logs

n/a

john30, 2023-09-09:
should be possible to just specify replacement command like this in your own Dockerfile using the image from hub as base:
`CMD []`

LordShadowen, 2023-09-09:
Yes, that could work. But it's the same type of solution I'm doing with my docker-compose. 
(and in general I try not to tinker with custom Dockerfiles...)

But the more general point is that it should not be needed for an end user to do such thing. If it's a feature that can be turned on/off (which this is), it should be possible to configure it in a "standard" way.

It seems the most logical to me that it's either:
- not providing the EBUSD_SCANCONFIG does not run Auto Scan (replicating the command-line behavior)
- or if that's not desirable for some reason, then provide a new ENV variable that explicitly disables it (something like EBUSD_DISABLESCAN or such...)

john30, 2023-09-23:
added with b0cfb52e3b1005cf340f241d1ebc17c4b653086b

closed 2023-09-23 by john30