john30/ebusd #1007: scan config 08: ERR: element not found

john4242, 2023-09-09:
### Description

I have a Wolf FGB-35 boiler, connected it up to a recently purchased ebus adapter v5.2. The green light is active and I guess that means wiring is good (UTP twisted wire was used and about 3m long).
I am having issues with scan config and communication. All I see is received unknown BC cmd: 03fe050308010000004320383f
and scan is also strange: scan 08: ;65; &;
Tried to read up on all documentations but didn't find any solution. Also enabled debug level to have more info.

### Actual behavior

It does not seem to properly scan and detect

### Expected behavior

It should detect

### ebusd version

23.2

### ebusd arguments

--scanconfig -d ens:192.168.1.2:12999 --loglevel=debug

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 5 via WiFi

### Related integration

other

### Logs

2023-09-09 10:21:16.384 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.1.2:12999, enhanced
2023-09-09 10:21:17.136 [bus notice] device status: resetting
2023-09-09 10:21:17.136 [bus notice] bus started with own address 31/36
2023-09-09 10:21:17.184 [bus notice] signal acquired
2023-09-09 10:21:17.386 [bus notice] device status: reset, supports info
2023-09-09 10:21:17.516 [bus notice] device status: extra info: firmware 1.1[3903].1[3903], jumpers 0x0f
2023-09-09 10:21:19.659 [bus notice] new master 03, master count 2
2023-09-09 10:21:19.659 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:21:27.280 [main error] scan config 08: ERR: read timeout
2023-09-09 10:21:29.732 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:21:39.715 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:21:49.768 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:21:59.821 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:22:09.815 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:22:19.852 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:22:29.612 [update notice] store 08 ident: ERR: argument value out of valid range
2023-09-09 10:22:29.612 [update error] unable to parse scan-read scan.08  from 3108070400 / 0a410126004000002a0001: ERR: argument value out of valid range
2023-09-09 10:22:29.612 [bus notice] scan 08: ;65; &;
2023-09-09 10:22:29.612 [main error] scan config 08: ERR: argument value out of valid range
2023-09-09 10:22:29.951 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:22:40.040 [update notice] received unknown BC cmd: 03fe050308010000004421383f
2023-09-09 10:22:50.061 [update notice] received unknown BC cmd: 03fe050308010000004421383f
2023-09-09 10:23:00.095 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:23:10.190 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:23:20.146 [update notice] received unknown BC cmd: 03fe050308010000004421393f
2023-09-09 10:23:21.796 [main notice] update check: invalid request
2023-09-09 10:23:30.163 [update notice] received unknown BC cmd: 03fe050308010000004421393f
..
2023-09-09 10:33:23.129 [main info] successfully read file broadcast.csv
2023-09-09 10:33:23.130 [main info] read config files, got 11 messages
2023-09-09 10:33:23.263 [bus notice] device status: resetting
2023-09-09 10:33:23.264 [bus notice] bus started with own address 31/36
2023-09-09 10:33:23.264 [main info] registering data handlers
2023-09-09 10:33:23.264 [main info] registered data handlers
2023-09-09 10:33:23.322 [bus debug] ERR: SYN received during no signal, switching to ready
2023-09-09 10:33:23.322 [bus notice] signal acquired
2023-09-09 10:33:23.394 [bus notice] device status: reset, supports info
2023-09-09 10:33:23.571 [bus notice] device status: extra info: firmware 1.1[3903].1[3903], jumpers 0x0f
2023-09-09 10:33:32.901 [bus notice] new master 03, master count 2
2023-09-09 10:33:32.902 [update info] received BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:32.902 [update notice] received unknown BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:33.264 [main debug] performing regular tasks
2023-09-09 10:33:33.265 [bus info] scan 08 cmd: 3108070400
2023-09-09 10:33:33.302 [bus debug] start request 31
2023-09-09 10:33:33.303 [bus debug] arbitration start with 31
2023-09-09 10:33:33.392 [bus debug] arbitration won
2023-09-09 10:33:33.392 [bus debug] arbitration delay 13 micros
2023-09-09 10:33:33.392 [bus info] arbitration delay 13 - 13 micros
2023-09-09 10:33:33.392 [bus debug] switching from ready to send command
2023-09-09 10:33:33.412 [bus debug] send/receive symbol latency 19 ms
2023-09-09 10:33:33.412 [bus info] send/receive symbol latency 19 - 19 ms
2023-09-09 10:33:33.432 [bus debug] send/receive symbol latency 19 ms
2023-09-09 10:33:33.452 [bus debug] send/receive symbol latency 19 ms
2023-09-09 10:33:33.469 [bus debug] send/receive symbol latency 17 ms
2023-09-09 10:33:33.469 [bus info] send/receive symbol latency 17 - 19 ms
2023-09-09 10:33:33.469 [bus debug] switching from send command to send command CRC
2023-09-09 10:33:33.490 [bus debug] send/receive symbol latency 20 ms
2023-09-09 10:33:33.490 [bus info] send/receive symbol latency 17 - 20 ms
2023-09-09 10:33:33.490 [bus debug] switching from send command CRC to receive command ACK
2023-09-09 10:33:33.492 [bus debug] switching from receive command ACK to receive response
2023-09-09 10:33:33.536 [bus debug] switching from receive response to receive response CRC
2023-09-09 10:33:33.542 [bus debug] switching from receive response CRC to send response ACK
2023-09-09 10:33:33.562 [bus debug] send/receive symbol latency 19 ms
2023-09-09 10:33:33.562 [update info] sent MS cmd: 3108070400 / 0a410126004000002a0001
2023-09-09 10:33:33.562 [update notice] store 08 ident: ERR: argument value out of valid range
2023-09-09 10:33:33.562 [update error] unable to parse scan-read scan.08  from 3108070400 / 0a410126004000002a0001: ERR: argument value out of valid range
2023-09-09 10:33:33.562 [bus debug] notify request: done
2023-09-09 10:33:33.562 [bus notice] scan 08: ;65; &;
2023-09-09 10:33:33.562 [bus debug] switching from send response ACK to send SYN
2023-09-09 10:33:33.563 [main error] scan config 08: ERR: argument value out of valid range
2023-09-09 10:33:33.582 [bus debug] send/receive symbol latency 19 ms
2023-09-09 10:33:33.582 [bus debug] switching from send SYN to ready
2023-09-09 10:33:35.563 [main debug] performing regular tasks
2023-09-09 10:33:42.975 [update info] received BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:42.975 [update notice] received unknown BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:45.564 [main debug] performing regular tasks
...
2023-09-09 10:33:45.753 [main error] HTTP failure: receive error: 404 Not Found
2023-09-09 10:33:45.753 [main error] unable to load scan config 08: list files in 65 ERR: element not found
2023-09-09 10:33:45.753 [main error] scan config 08: ERR: element not found
2023-09-09 10:33:47.753 [main debug] performing regular tasks
2023-09-09 10:33:52.963 [update info] received BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:52.964 [update notice] received unknown BC cmd: 03fe050308010000004320383f
2023-09-09 10:33:53.171 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2023-09-09 10:33:57.753 [main debug] performing regular tasks
2023-09-09 10:34:03.020 [update info] received BC cmd: 03fe050308010000004320383f
2023-09-09 10:34:03.021 [update notice] received unknown BC cmd: 03fe050308010000004320383f
2023-09-09 10:34:07.754 [main debug] performing regular tasks
2023-09-09 10:34:13.055 [update info] received BC cmd: 03fe050308010000004320383f
2023-09-09 10:34:13.056 [update notice] received unknown BC cmd: 03fe050308010000004320383f
2023-09-09 10:34:17.755 [main debug] performing regular tasks

john4242, 2023-09-09:
Things are not totally bad as I do see data change when I turn on the heating (radiators):
It looks like "0503" is some kind of Wolf command the boiler emits regularly and when heating is on you can see the "00400" in the middle (vs. "00000"), and I guess some other parts might be temperatures?

2023-09-09 11:22:45.958 [update notice] received unknown BC cmd: 03fe050308010000007b3a403f
2023-09-09 11:22:55.984 [update notice] received unknown BC cmd: 03fe050308010000007b3a403f
Heating (radiator) started:
2023-09-09 11:23:06.086 [update notice] received unknown BC cmd: 03fe050308010040007b38403f
2023-09-09 11:23:11.609 [main notice] update check: invalid request
2023-09-09 11:23:16.154 [update notice] received unknown BC cmd: 03fe05030801004000751d403f
2023-09-09 11:23:26.211 [update notice] received unknown BC cmd: 03fe050308010040005a24403f
2023-09-09 11:23:36.196 [update notice] received unknown BC cmd: 03fe050308010040005129403f
2023-09-09 11:23:46.182 [update notice] received unknown BC cmd: 03fe050308010040005327403f
2023-09-09 11:23:50.016 [bus error] signal lost
2023-09-09 11:23:51.267 [bus notice] signal acquired
2023-09-09 11:23:56.278 [update notice] received unknown BC cmd: 03fe050308010040005225403f
2023-09-09 11:24:06.403 [update notice] received unknown BC cmd: 03fe050308010040004f24403f
2023-09-09 11:24:16.401 [update notice] received unknown BC cmd: 03fe050308010040004d23403f
2023-09-09 11:24:26.405 [update notice] received unknown BC cmd: 03fe050308010040004b23403f
2023-09-09 11:24:36.551 [update notice] received unknown BC cmd: 03fe050308010040004922403f
2023-09-09 11:24:46.508 [update notice] received unknown BC cmd: 03fe050308010040004821403f
2023-09-09 11:24:56.534 [update notice] received unknown BC cmd: 03fe050308010040004620403f
2023-09-09 11:25:06.635 [update notice] received unknown BC cmd: 03fe050308010040004520403f
2023-09-09 11:25:16.658 [update notice] received unknown BC cmd: 03fe05030801004000431f403f
2023-09-09 11:25:26.707 [update notice] received unknown BC cmd: 03fe05030801004000421f403f
2023-09-09 11:25:36.730 [update notice] received unknown BC cmd: 03fe05030801004000411e403f
Heating (radiator) turned off:
2023-09-09 11:25:46.810 [update notice] received unknown BC cmd: 03fe05030801004000401e3f3f
2023-09-09 11:25:56.768 [update notice] received unknown BC cmd: 03fe050308010040003f1d3f3f
2023-09-09 11:26:06.839 [update notice] received unknown BC cmd: 03fe050308010040003e1d3f3f
2023-09-09 11:26:16.955 [update notice] received unknown BC cmd: 03fe050308010000003d1d403f
2023-09-09 11:26:26.953 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:26:37.064 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:26:47.046 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:26:57.148 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:27:07.087 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:27:17.184 [update notice] received unknown BC cmd: 03fe050308010000003e1d403f
2023-09-09 11:27:27.257 [update notice] received unknown BC cmd: 03fe050308010000003e1d3f3f
2023-09-09 11:27:37.276 [update notice] received unknown BC cmd: 03fe050308010000003e1d3f3f
2023-09-09 11:27:47.382 [update notice] received unknown BC cmd: 03fe050308010000003e1d3f3f
2023-09-09 11:27:57.420 [update notice] received unknown BC cmd: 03fe050308010000003e1e3f3f
2023-09-09 11:28:07.479 [update notice] received unknown BC cmd: 03fe050308010000003e1e403f
2023-09-09 11:28:17.430 [update notice] received unknown BC cmd: 03fe050308010000003e1e403f
2023-09-09 11:28:27.528 [update notice] received unknown BC cmd: 03fe050308010000003e1e403f
2023-09-09 11:28:37.494 [update notice] received unknown BC cmd: 03fe050308010000003e1e403f

john4242, 2023-09-09:
I think I found a wolf.csv in github, but why wasn't this picked up?
https://github.com/john30/ebusd-configuration/blob/master/ebusd-0.5.x/wolf/wolf.csv

eh-gh, 2023-09-15:
HI
it is not clear to me how you added the wolf config files to ebusd.
Afaik the default webservice  [https://cfg.ebusd.eu/] contains only the vaillant configs, if you want to connect to wolf, you have to specify it with:

-c, --configpath=PATH
Read CSV config files from PATH (local folder or HTTPS URL)

john4242, 2023-09-15:
> HI it is not clear to me how you added the wolf config files to ebusd. Afaik the default webservice [https://cfg.ebusd.eu/] contains only the vaillant configs, if you want to connect to wolf, you [...]

I didn't add wolf config files, I don't think there is one yet. I might do one.
My issue is that the scan says "value out of valid range".

2023-09-09 10:33:33.562 [update info] sent MS cmd: 3108070400 / 0a410126004000002a0001
2023-09-09 10:33:33.562 [update notice] store 08 ident: ERR: argument value out of valid range
2023-09-09 10:33:33.562 [update error] unable to parse scan-read scan.08 from 3108070400 / 0a410126004000002a0001: ERR: argument value out of valid range
2023-09-09 10:33:33.562 [bus debug] notify request: done
2023-09-09 10:33:33.562 [bus notice] scan 08: ;65; &;

I found useful ebus docu at: https://ebus-wiki.org/doku.php/ebus/ebusdoku
and especially the PDF: https://ebus-wiki.org/lib/exe/fetch.php/ebus/spec_prot_7_v1_6_1_e.pdf
which seems to correspond with the broadcast: 05 03 Operational Data of Burner Control unit to Room Control Unit (Service 05h 03h Block 1)
The only thing is I cannot do an auto scan config until the above scan works. Currently it says "value out of valid range"

For 05 03 broadcast status message, this should be it:
[image]
[image]

eh-gh, 2023-09-15:
there are some basic configs for wolf under:

https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de
what if you give a try with -c 'https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de'?

I agree some specific commands need to be analyzed/adapted based on log and current state of the boiler.

john4242, 2023-09-15:
> there are some basic configs for wolf under:
>
> https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de what if you give a try with -c [...]

It looks to be good now!

I have gone with:
`EBUSD_OPTS="-c /etc/ebusd/configuration -d ens:192.168.1.2:12999 --httpport 8889"`
(obviously first downloading the https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/de into /etc/ebusd/configuration, removed the non-wolf folders) and now it works.
It recognizes the broadcast messages and decodes it! woohoo!

Thank you! This is good enough for me now.

2023-09-15 14:47:30.049 [main notice] ebusd stopped
2023-09-15 14:47:30.127 [main notice] ebusd 23.2.23.2 started on device: 192.168.1.2:12999, enhanced
2023-09-15 14:47:30.272 [bus notice] device status: resetting
2023-09-15 14:47:30.273 [bus notice] bus started with own address 31/36
2023-09-15 14:47:30.278 [bus notice] signal acquired
2023-09-15 14:47:30.349 [bus notice] device status: reset, supports info
2023-09-15 14:47:30.566 [bus notice] device status: extra info: firmware 1.1[3903].1[3903], jumpers 0x0f
2023-09-15 14:47:39.340 [bus notice] new master 03, master count 2
2023-09-15 14:47:39.341 [update notice] received update-read broadcast betrd QQ=03: 0;00;0;29.5;28;56;63
2023-09-15 14:47:40.273 [main notice] found messages: 56 (0 conditional on 0 conditions, 0 poll, 6 update)
2023-09-15 14:47:43.003 [bus error] signal lost
2023-09-15 14:47:44.305 [bus notice] signal acquired
2023-09-15 14:47:49.419 [update notice] received update-read broadcast betrd QQ=03: 0;00;0;29.5;28;56;63
2023-09-15 14:47:59.398 [update notice] received update-read broadcast betrd QQ=03: 0;00;0;29.5;28;56;63

john4242, 2023-09-15:
Well, I got it working with MQTT, I can see the broadcast message there, I might just take it on with Node-red from there for now.

```
EBUSD_OPTS="-c /etc/ebusd/configuration -d ens:192.168.1.2:12999 --httpport 8889 --mqtthost=127.0.0.1 --mqttport
=1883 --mqttint=/etc/ebusd/mqtt-integration.cfg --mqtttopic=ebusd"
```
 
$ mosquitto_sub -h 127.0.0.1 -p 1883 -t "ebusd/#" -v
ebusd/global/version ebusd 23.2.23.2
ebusd/global/running true
ebusd/global/scan finished
ebusd/global/signal true
ebusd/global/uptime 48
**ebusd/broadcast/betrd 0;7c;100;62.5;47;47;63**
ebusd/global/uptime 64
ebusd/global/signal false
ebusd/global/uptime 80
ebusd/global/signal true
ebusd/broadcast/betrd 0;7c;75;73.0;55;45;63

john30, 2023-09-17:
wrong repo (use ebusd-configuration instead) and depending on someone to provide at least half-decent config files for wolf that will be served once it is part of the [latest](https://github.com/john30/ebusd-configuration/blob/master/latest) link.

closed 2023-09-17 by john30