john30/ebusd #1013: limit scope of scanning device - lot of errors "scan config 05: ERR: read timeout"

Sw4nty, 2023-09-20:
### Description

I have eBUS adapter 5 connected throw wifi - Build: [20230917] 
Ebusd version 23.2.23.2 in container (podman)  

current ebusd_opts are:
"--enablehex --enabledefine --device ens:X.X.X.X:9999 --pollinterval=1 --receivetimeout=100 --sendretries=3 --accesslevel=* --configpath /ebusd-config --latency=50 "

In ebus network i have :
saunier duval gas boiler  Thelia Condens
saunier duval red-3 - similar to VR_70
saunier duval Mipro - similar to VRC 700
ebus adapter

after modify cvs files which exist in repo "john30/ebusd-configuration" all devices mostlly work fine (still in testing)
and are discover correctly :

08;Vaillant;BAI00;0712;1303;21;22;40;0010016097;1610;007790;N0
15;Vaillant;E7000;0608;7003;21;22;07;0020218373;0082;005713;N7
52;Vaillant;VR_70;0101;5804;21;22;31;0020257508;0082;006869;N2

from 
ebusctl  info
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0712;HW=1303", loaded "vaillant/bai.0010016097.inc" ([PROD='0010016097']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7000;SW=0608;HW=7003", loaded "vaillant/15.e7000.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0101;HW=5804", loaded "vaillant/52.vr_70.csv"

Problem which bother me  if there is an option to remove "05" from scanning if it always end as error timeout ? 

Similar when I run full scan but then it also found :
ec;Vaillant;SOL00;0608;7003;21;22;07;0020218373;0082;005713;N7;21;22;07;0020218373;0082;005713;N7
which is not important for me - (not in use)  - maybe it is inside bai or inside red-3 ??? I don;t have more devices ;)

### Actual behavior

every scan end up with : scan config 05: ERR: read timeout

### Expected behavior

exclude 05

or set manual device mapping to files and disable scanning totally ???
when i try to run without scanning (mayby i do something wrong) it end up  with error :
"[main error] error reading config files from /ebusd-config/: ERR: duplicate entry, last error: vaillant/05.vd3.csv:8: ERR: duplicate entry, duplicate ID"

### ebusd version

23.2

### ebusd arguments

--enablehex --enabledefine --device ens:X.X.X.X:9999 --pollinterval=1 --receivetimeout=100 --sendretries=3 --accesslevel=* --configpath /ebusd-config --latency=50

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via USB

### Related integration

TCP (cmdline client like ebusctl or netcat), MQTT Home Assistant via mqtt-hassio.cfg

### Logs

logs from last hour 

podman logs --since 1h ebusd_mq_debug | grep "scan config 05: ERR: read timeout"
2023-09-20 20:29:26.809 [main error] scan config 05: ERR: read timeout
2023-09-20 20:30:34.745 [main error] scan config 05: ERR: read timeout
2023-09-20 20:31:47.130 [main error] scan config 05: ERR: read timeout
2023-09-20 20:33:00.180 [main error] scan config 05: ERR: read timeout
2023-09-20 20:34:12.552 [main error] scan config 05: ERR: read timeout
2023-09-20 20:35:24.949 [main error] scan config 05: ERR: read timeout
2023-09-20 20:36:37.340 [main error] scan config 05: ERR: read timeout
2023-09-20 20:37:49.970 [main error] scan config 05: ERR: read timeout
2023-09-20 20:39:02.391 [main error] scan config 05: ERR: read timeout
2023-09-20 20:40:14.781 [main error] scan config 05: ERR: read timeout
2023-09-20 20:41:27.171 [main error] scan config 05: ERR: read timeout
2023-09-20 20:42:39.721 [main error] scan config 05: ERR: read timeout
2023-09-20 20:43:52.245 [main error] scan config 05: ERR: read timeout
2023-09-20 20:45:04.622 [main error] scan config 05: ERR: read timeout
2023-09-20 20:46:16.992 [main error] scan config 05: ERR: read timeout
2023-09-20 20:47:29.382 [main error] scan config 05: ERR: read timeout
2023-09-20 20:48:41.842 [main error] scan config 05: ERR: read timeout
2023-09-20 20:49:54.273 [main error] scan config 05: ERR: read timeout
2023-09-20 20:51:06.653 [main error] scan config 05: ERR: read timeout
2023-09-20 20:52:19.053 [main error] scan config 05: ERR: read timeout
2023-09-20 20:53:31.643 [main error] scan config 05: ERR: read timeout
2023-09-20 20:54:44.031 [main error] scan config 05: ERR: read timeout
2023-09-20 20:55:56.404 [main error] scan config 05: ERR: read timeout
2023-09-20 20:57:08.784 [main error] scan config 05: ERR: read timeout
2023-09-20 20:58:21.164 [main error] scan config 05: ERR: read timeout
2023-09-20 20:59:33.735 [main error] scan config 05: ERR: read timeout
2023-09-20 21:00:46.145 [main error] scan config 05: ERR: read timeout
2023-09-20 21:01:58.535 [main error] scan config 05: ERR: read timeout
2023-09-20 21:03:10.955 [main error] scan config 05: ERR: read timeout
2023-09-20 21:04:23.335 [main error] scan config 05: ERR: read timeout
2023-09-20 21:05:35.716 [main error] scan config 05: ERR: read timeout
2023-09-20 21:06:48.476 [main error] scan config 05: ERR: read timeout
2023-09-20 21:08:00.857 [main error] scan config 05: ERR: read timeout
2023-09-20 21:09:13.237 [main error] scan config 05: ERR: read timeout
2023-09-20 21:10:25.667 [main error] scan config 05: ERR: read timeout
2023-09-20 21:11:38.037 [main error] scan config 05: ERR: read timeout
2023-09-20 21:12:50.407 [main error] scan config 05: ERR: read timeout
2023-09-20 21:14:02.777 [main error] scan config 05: ERR: read timeout
2023-09-20 21:15:15.168 [main error] scan config 05: ERR: read timeout

edsub, 2023-09-21:
Similar issue here since I have the ebusd v5 adapter, but with the timeout on the VR720
`[main error] scan config 15: ERR: read timeout`

In addition my ebusd daemon does not discover ANY of my components:
08.hmu
15.ctlv2 (also tried 15.700 and 15.720)
76.vwz

My isues are in a way similar to when I had the Esera adapter. I bought the ebusd adapter thinking that would solve my issues . . . .

john30, 2023-09-23:
@edsub providing a log file would help, hijacking another issue does not

john30, 2023-09-23:
@Sw4nty when using files instead, you have to reduce the files to those explicitly needed (and remove others), as otherwise duplication errors occur when loading all of them.
anyway, just added an option to limit the scan retries, so ebusd will give up scanning after some time

john30, 2023-09-23:
fixed with 0698906cd14672417ed119978ee89138c267f316

closed 2023-09-23 by john30

Sw4nty, 2023-09-23:
@john30 Thanks for fast response 

Both resolutions fit my needs but second is even better ;)

Thanks again