john30/ebusd #1020: Ebusd strange behaviour or wrong record definition ??

Sw4nty, 2023-09-22:
### Description

I have eBUS adapter 5 connected throw wifi - Build: [20230917]
Ebusd version 23.2.23.2 in container (podman)

current ebusd_opts are:
"--enablehex --enabledefine --device ens:X.X.X.X:9999 --pollinterval=1 --receivetimeout=100 --sendretries=3 --accesslevel=* --configpath /ebusd-config --latency=50 "

affected definition in bai:

# HC operation mode2,,,,,,,,,,,,,
*r,,,,,,"B511",,,,,,,
*w,,,,,,"B510",,,,,,,
w,,SetMode,Operation Mode,,,,00,,,hcmode,,,,flowtempdesired,,temp1,,,,hwctempdesired,,temp1,,,,hwcflowtempdesired,,temp0,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2,,,,currentmode,s,UCH,0=off;1=standby;2=heat;3=water,,

### Actual behavior

I found strange behaviour :

When i restart or start ebusd it is able to catch write base on definition first time (after rescan of devices ):

ebusctl  find -V -a |  grep -i setmode
bai SetMode = hcmode=auto [boiler mode];flowtempdesired=0.0 °C [temperature];hwctempdesired=- °C [temperature];hwcflowtempdesired=- °C [temperature];disablehc=1;disablehwctapping=0;disablehwcload=0;remoteControlHcPump=0;releaseBackup=0;releaseCooling=0;currentmode=standby [ZZ=08, lastup=2023-09-22 22:24:06, active write]

ebusctl  grab result
**1008b51009000000ffffff010000 / 0101 = 1**
1008b51009000000ffffff010000 / 0101 = 13

and every next iteration of write is skipped - (13) not updated in "ebusctl  find" 
after next ebusd restart it again will show in ebusctl  find "actually" values once

example 
podman logs  --since 2m ebusd_mq_debug  | grep -i -E "setmode | ebusd 23.2.23.2 | received unknown MS cmd: 1008b51009"
2023-09-22 22:42:51.369 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:00.808 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced
2023-09-22 22:43:11.029 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:21.050 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby
2023-09-22 22:43:31.099 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:41.139 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:51.159 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:01.493 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:04.851 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced
2023-09-22 22:44:11.269 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:21.610 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby
2023-09-22 22:44:31.331 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:41.359 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101

### Expected behavior

????

### ebusd version

23.2

### ebusd arguments

--enablehex --enabledefine --device ens:X.X.X.X:9999 --pollinterval=1 --receivetimeout=100 --sendretries=3 --accesslevel=* --configpath /ebusd-config --latency=50

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 5 via WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat), MQTT Home Assistant via mqtt-hassio.cfg

### Logs

podman logs  --since 2m ebusd_mq_debug  | grep -i -E "setmode | ebusd 23.2.23.2 | received unknown MS cmd: 1008b51009"
2023-09-22 22:42:51.369 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:00.808 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced
2023-09-22 22:43:11.029 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:21.050 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby
2023-09-22 22:43:31.099 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:41.139 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:51.159 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:01.493 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:04.851 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced
2023-09-22 22:44:11.269 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:21.610 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby
2023-09-22 22:44:31.331 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:44:41.359 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101

john30, 2023-09-23:
I don't understand what you're missing

Sw4nty, 2023-09-23:
I should write maybe more straight
when i start ebusd and i have definition :

definition in bai:

*r,,,,,,"B511",,,,,,,
*w,,,,,,"B510",,,,,,,
w,,SetMode,Operation Mode,,,,00,,,hcmode,,,,flowtempdesired,,temp1,,,,hwctempdesired,,temp1,,,,hwcflowtempdesired,,temp0,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2,,,,currentmode,s,UCH,0=off;1=standby;2=heat;3=water,,

after scan of devices completed ebus it is able first time to show in logs that he parse this pattern (received write bai SetMode)
and every next appearance show  as "received unknown MS cmd: 1008b51009000000ffffff010000 / 0101" until next restart of ebusd and again first time it show corrretly values but next not 

podman logs --since 2m ebusd_mq_debug | grep -i -E "setmode | ebusd 23.2.23.2 | received unknown MS cmd: 1008b51009"
2023-09-22 22:42:51.369 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:00.808 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced **< start**
2023-09-22 22:43:11.029 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-22 22:43:21.050 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby < **pattern found first time after scan** 
2023-09-22 22:43:31.099 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **not found** 
2023-09-22 22:43:41.139 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **not found** 
2023-09-22 22:43:51.159 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **not found** 
2023-09-22 22:44:01.493 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **not found** 
2023-09-22 22:44:04.851 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced **< next restart** 
2023-09-22 22:44:11.269 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 
2023-09-22 22:44:21.610 [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby < **pattern found first time after scan completed and file definition was included ** 
2023-09-22 22:44:31.331 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101< **not found**
2023-09-22 22:44:41.359 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **not found**

I didn;t see such behaviour in other write definitions - so i was wondering if this definition is wrong or maybe it is bug ???

An logs from today 

[user@fedora-podman en]$ podman logs --since 3m ebusd_mq_debug   | grep -i -E "setmode | ebusd 23.2.23.2 | received unknown MS cmd: 1008b51009"
2023-09-23 15:00:56.794 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:01:06.684 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:01:16.479 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:01:27.526 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:01:36.987 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 **< before restart**
2023-09-23 15:01:52.694 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.7.130:9999, enhanced < **restart**
2023-09-23 15:01:57.645 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **before scan**
**2023-09-23 15:02:07.405** [update notice] received write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0;standby < **after scan when configuration from files is loaded - 1 attept**
2023-09-23 15:02:16.885 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101 < **2 attempt not recognise**
2023-09-23 15:02:27.041 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:02:36.995 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:02:46.995 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:02:56.995 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:03:07.715 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:03:17.455 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:03:27.025 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 15:03:37.075 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101

ebusctl  grab result after today restart
1008b51009000000ffffff010000 / 0101 = 1
1008b51009000000ffffff010000 / 0101 = 3

from find also from  today 

ebusctl  find -V -a -h |  grep -i setmode
bai SetMode = 1008b51009000000ffffff010000 / 0101 [ZZ=08, lastup=2023-09-23 15:02:07, active write]
ebusctl  find -V -a |  grep -i setmode
bai SetMode = hcmode=auto [boiler mode];flowtempdesired=0.0 °C [temperature];hwctempdesired=- °C [temperature];hwcflowtempdesired=- °C [temperature];disablehc=1;disablehwctapping=0;disablehwcload=0;remoteControlHcPump=0;releaseBackup=0;releaseCooling=0;currentmode=standby [ZZ=08, **lastup=2023-09-23 15:02:07**, 

It so strange why after restart 1 attempt is ok ;)

john30, 2023-09-23:
are you sure the custom csvs are always reachable from within the container in the /ebusd-config path? maybe this is a podman issue or something with mounting a local volume/path into the container

Sw4nty, 2023-09-23:
Yes they are otherwise probably all configuration will not be readable inside container 

i use in podman "-v /home/user/ebusd-configuration/ebusd-2.1.x/en:/ebusd-config"  - there is whole directory with all cvs files 

and in ebus_opt: "--configpath /ebusd-config"

thats why i was a bit confuse maybe that definition was not right but still not explain why after restart it is able to show values once ;O

Maybe i will try to remove it from file and add manually via "ebusctl define" and see what happens 
Or maybe it is somehow replace via other files definitions i will also look into it also ...

john30, 2023-09-24:
you should be able to check if the files are still there once it occurs again by issuing a "podman exec ls -la /ebusd-config" or so

Sw4nty, 2023-09-26:
Yes files are there ;/ 

podman exec ebusd_mq_debug ls -la /ebusd-config/vaillant/hcmode.inc
-rw-r--r--. 1 root root 2279 Sep 23 22:30 /ebusd-config/vaillant/hcmode.inc

john30, 2023-10-08:
maybe just a rights issue. start ebusd with additional "--log=debug" and check the output. it should give lines like
```
2023-10-08 06:17:49.571 [main info] loading configuration files from /ebusd-config/
2023-10-08 06:17:49.571 [main debug] reading directory /ebusd-config/
2023-10-08 06:17:49.571 [main debug] file type of /ebusd-config/vaillant is dir
2023-10-08 06:17:49.571 [main debug] file type of /ebusd-config/_templates.csv is file
```

Sw4nty, 2023-11-06:
I will check it and verify 
I also try to change a bit definition in bai line - i found some examples in other issue and improve list so i also testing that approche

john30, 2024-04-27:
closed due to inactivity

closed 2024-04-27 by john30