john30/ebusd #1022: Incorrect interpretation of log configuration parameter

zyzakin, 2023-09-23:
### Description

I run the daemon according to the following configuration

`/usr/bin/ebusd -d ens:192.168.1.25:9999 --latency=80 --configlang=EN --scanconfig=full --configpath=http://cfg.ebusd.eu/ --log=main error --mqtthost=192.168.1.88 --mqttport=1883 --mqttjson -l /var/log/ebusd.log`

Unfortunately I also get other areas and levels in the logs

> ihome@hajime:~$ /etc/init.d/ebusd status
● ebusd.service - ebusd, the daemon for communication with eBUS heating systems.
           Loaded: loaded (/lib/systemd/system/ebusd.service; disabled; vendor preset: enabled)
              Active: active (running) since Sat 2023-09-23 10:57:28 CEST; 1min 49s ago
           Process: 2356610 ExecStart=/usr/bin/ebusd $EBUSD_OPTS (code=exited, status=0/SUCCESS)
        Main PID: 2356611 (ebusd)
               Tasks: 5 (limit: 1774)
          Memory: 3.0M
            CGroup: /system.slice/ebusd.service
                           └─2356611 /usr/bin/ebusd -d ens:192.168.1.25:9999 --latency=80 --configlang=EN --scanconfig=full --configpath=http://cfg.ebusd.eu/ --log=main error --mqtthost=192.168.1.88 --mqttport=1883 --mqttjson -l 

### Actual behavior

It gets all areas and levels

### Expected behavior

I would like to get only the area and level that I defined

### ebusd version

23.2

### ebusd arguments

EBUSD_OPTS="-d ens:192.168.1.25:9999 --latency=80 --configlang=EN --scanconfig=full --configpath=http://cfg.ebusd.eu/ --log=main:error  --mqtthost=192.168.1.88 --mqttport=1883 --mqttjson -l /var/log/ebusd.log"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

`2023-09-23 10:57:53.522 [update notice] sent scan-read scan.08 id QQ=31: 
2023-09-23 10:57:53.632 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2023-09-23 10:57:54.442 [update notice] sent scan-read scan.08 id QQ=31: 21;18;33;0010021887;0001;008098;N7
2023-09-23 10:57:55.932 [update notice] sent scan-read scan.08 id QQ=31: 21;18;33;0010021887;0001;008098;N7
2023-09-23 10:57:56.792 [update notice] sent scan-read scan.08 id QQ=31: 21;18;33;0010021887;0001;008098;N7
2023-09-23 10:57:57.662 [update notice] sent scan-read scan.08 id QQ=31: 21;18;33;0010021887;0001;008098;N7
2023-09-23 10:57:58.532 [update notice] sent scan-read scan.08 id QQ=31: 21;18;33;0010021887;0001;008098;N7
2023-09-23 10:57:58.532 [bus notice] scan 08: ;21;18;33;0010021887;0001;008098;N7
2023-09-23 10:58:01.232 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231313734323030
2023-09-23 10:58:02.152 [update notice] sent scan-read scan.15 id QQ=31: 
2023-09-23 10:58:02.271 [update notice] received read bai Status01 QQ=10: 22.5;23.5;19.812;-;54.0;off
2023-09-23 10:58:03.073 [update notice] sent scan-read scan.15 id QQ=31: 
2023-09-23 10:58:04.092 [update notice] sent scan-read scan.15 id QQ=31: 21;17;42;0020171316;0082;018020;N2
2023-09-23 10:58:04.093 [bus notice] scan 15: ;21;17;42;0020171316;0082;018020;N2
2023-09-23 10:58:05.653 [update notice] sent read bai HwcTemp QQ=31: 116.06;circuit
2023-09-23 10:58:07.274 [update notice] sent read bai OutdoorstempSensor QQ=31: 19.81;ok
2023-09-23 10:58:08.447 [update notice] sent read bai StorageTemp QQ=31: 54.31;ok
2023-09-23 10:58:09.461 [update notice] sent read bai StorageTempDesired QQ=31: 55.00
2023-09-23 10:58:10.543 [update notice] sent read bai WaterPressure QQ=31: 2.462;ok
2023-09-23 10:58:11.301 [bus error] send to 08: ERR: SYN received, retry
2023-09-23 10:58:12.673 [bus error] send to 08: ERR: SYN received, retry
2023-09-23 10:58:13.742 [update notice] sent read bai HeatingSwitch QQ=31: off
2023-09-23 10:58:13.879 [update notice] received read bai Status01 QQ=10: 22.5;23.5;19.812;-;54.0;off
2023-09-23 10:58:14.805 [update notice] sent read bai FlowTempDesired QQ=31: 0.00
2023-09-23 10:58:15.658 [bus error] send to 08: ERR: SYN received, retry
2023-09-23 10:58:16.728 [update notice] sent read bai FlowTemp QQ=31: 22.88;ok
2023-09-23 10:58:17.835 [update notice] sent read bai Flame QQ=31: off
2023-09-23 10:58:18.846 [update notice] sent read bai PrEnergySumHc1 QQ=31: 2290549618
2023-09-23 10:58:19.903 [update notice] sent read bai PrEnergySumHwc1 QQ=31: 107970633
2023-09-23 10:58:21.030 [update notice] sent read bai DCRoomthermostat QQ=31: on
2023-09-23 10:58:22.033 [update error] unable to parse read bai PartloadHcKW from 3108b509030d0704 / 00: ERR: invalid position
2023-09-23 10:58:22.034 [main error] read bai PartloadHcKW: decode ERR: invalid position
2023-09-23 10:58:22.855 [update notice] received read bai Status01 QQ=10: 22.5;23.5;19.812;-;54.0;off
2023-09-23 10:58:23.136 [update notice] sent read bai Statenumber QQ=31: 31
2023-09-23 10:58:23.332 [mqtt error] decode bai PartloadHcKW: ERR: invalid position
2023-09-23 10:58:24.175 [update notice] sent read bai ModulationTempDesired QQ=31: 16.0
2023-09-23 10:58:24.301 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0
`

john30, 2023-09-23:
either use "--logareas" and "--loglevel" to set for one area only, or use "--log=all:none" followed by "--log="main:error" to set it individually

closed 2023-09-23 by john30

zyzakin, 2023-09-23:
Thank you very much for your prompt reply. This is the effect I wanted to achieve.