john30/ebusd #1027: New datatype

MaciekIzdebski, 2023-09-27:
### Description

Hi,
I am trying to complete the ebus message definitions for the SM1 solar module. I managed to decrypt the data in the broadcast message, but I can't match the data type. This is 4 byte data. The first two bytes are the FLT data type and the next two bytes are the UIN data type. The resulting value is the sum of the FLT and UIN data.
For example:
Input data 71 00 3f 00
decode flt 7100 -> 0.113
decode uin 3f00 -> 63
manual addition 63+0.113
Output: 63.113

Is it possible to add such a type? Or maybe there is already one and I just can't find it ;-).

Regards,
Maciek

john30, 2023-10-08:
are you sure the value is really supposed to be 63.113 degrees (or whatever)?
I'm nearly 100% sure that the second 2 bytes are just something else and not related to the first 2.

MaciekIzdebski, 2023-10-08:
Yes I'm sure that it is correct value.

john30, 2024-02-11:
provide some sample values with those readings please so I can verify. hex message sequence plus value read visually from the hardware

MaciekIzdebski, 2024-02-17:
Ok. But we have to wait for some sun ;-)

MaciekIzdebski, 2024-03-29:
Her you are
2024-03-29 09:40:01.458 [update notice] received update-read broadcast solarYield QQ=71: 25.69;0.878;39;0.922;903;0.004;0			Device reading 39.8
2024-03-29 09:40:01.498 [bus notice] <71fe50180e9b016e0327009a0387030400000022
2024-03-29 09:52:00.257 [update notice] received update-read broadcast solarYield QQ=71: 26.69;0.154;45;0.922;903;0.004;0			Device reading 45.1
2024-03-29 09:52:00.307 [bus notice] <71fe50180eab019a002d009a0387030400000064
2024-03-29 09:57:00.280 [update notice] received update-read broadcast solarYield QQ=71: 27.00;0.407;47;0.922;903;0.004;0			Device reading 47.4
2024-03-29 09:57:00.320 [bus notice] <71fe50180eb00197012f009a0387030400000072
2024-03-29 09:57:30.411 [update notice] received update-read broadcast solarYield QQ=71: 27.00;0.632;47;0.922;903;0.004;0			Device reading 47.6
2024-03-29 09:57:30.459 [bus notice] <71fe50180eb00178022f009a03870304000000d9