john30/ebusd #1038: No writable sensors in mqtt autodiscovery for Home Assistant

pattyland, 2023-10-12:
### Description

I'm using the [Home Assistant Addon](https://github.com/LukasGrebe/ha-addons) together with MQTT and a Vaillant atmoTEC exclusive boiler.
Auto discovery works perfect for a few binary sensors and a lof ot sensors. 
To control my heater I added `--mqttvar=filter-direction=r|u|^w` to the additional parameters from the addon. In the arguments field below you can validate that it gets passed through to ebus, but:

### Actual behavior

only sensors and binary sensors in /homeassistant/

### Expected behavior

Good question, switches or range inputs?

### ebusd version

23.2

### ebusd arguments

--foreground --httpport=8889 --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=mosquitto --mqttpass=mosquitto --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.178.8:9999 --log=all:notice --log=other:notice --mqttvar=filter-direction=r|u|^w

### Operating system

other

### CPU architecture

x64

### Dockerized

other

### Hardware interface

adapter 3.1 WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-10-12 23:55:00.228 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.178.8:9999, enhanced
2023-10-12 23:55:01.732 [bus notice] device status: resetting
2023-10-12 23:55:01.732 [bus notice] bus started with own address 31/36
2023-10-12 23:55:01.735 [mqtt notice] connection established
2023-10-12 23:55:01.764 [bus notice] signal acquired
2023-10-12 23:55:02.113 [bus notice] new master 03, master count 2
2023-10-12 23:55:02.456 [bus notice] device status: reset, supports info
2023-10-12 23:55:02.470 [bus notice] device status: extra info: firmware 1.1[8419].2[847f], jumpers 0x3b
2023-10-12 23:55:06.004 [bus notice] new master 10, master count 3
2023-10-12 23:55:06.062 [update notice] received unknown MS cmd: 1008b5110101 / 092c2c0080ffff0000ff
2023-10-12 23:55:06.344 [update notice] received unknown MS cmd: 1008b5100900000078ffff050000 / 0101
2023-10-12 23:55:06.621 [update notice] received unknown MS cmd: 1008b5110102 / 05013c6e501e
2023-10-12 23:55:11.966 [bus notice] scan 08: ;Vaillant;BAI00;0730;7401
2023-10-12 23:55:11.966 [update notice] store 08 ident: done
2023-10-12 23:55:11.966 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0730;7401
2023-10-12 23:55:11.966 [bus notice] scan 08: ;Vaillant;BAI00;0730;7401
2023-10-12 23:55:12.536 [main notice] read common config file vaillant/broadcast.csv
2023-10-12 23:55:12.677 [main notice] read common config file vaillant/scan.csv
2023-10-12 23:55:12.813 [main notice] read common config file vaillant/general.csv
2023-10-12 23:55:12.952 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0730, HW7401
2023-10-12 23:55:13.422 [update notice] sent scan-read scan.08 id QQ=31: 
2023-10-12 23:55:13.631 [update notice] sent scan-read scan.08 id QQ=31: 
2023-10-12 23:55:13.818 [update notice] sent scan-read scan.08 id QQ=31: 
2023-10-12 23:55:14.031 [update notice] sent scan-read scan.08 id QQ=31: 21;22;01;0010018741;0001;005306;N4
2023-10-12 23:55:14.451 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
2023-10-12 23:55:14.710 [update notice] sent scan-read scan.08 id QQ=31: 21;22;01;0010018741;0001;005306;N4
[... 49 lines cut ...]
2023-10-12 23:56:16.033 [update notice] received read bai Status01 QQ=10: 22.0;22.0;-;-;-;off
2023-10-12 23:56:16.309 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;1;0;0;0
2023-10-12 23:56:17.253 [update notice] sent poll-read bai FanStarts QQ=31: 0
2023-10-12 23:56:19.249 [update notice] sent scan-read scan.15 id QQ=31: 
2023-10-12 23:56:19.350 [main error] scan config 15: ERR: SYN received
2023-10-12 23:56:23.211 [update notice] sent poll-read bai FlowTemp QQ=31: 22.12;ok
2023-10-12 23:56:26.009 [update notice] received read bai Status01 QQ=10: 22.0;22.0;-;-;-;off
2023-10-12 23:56:26.274 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;1;0;0;0
2023-10-12 23:56:29.172 [update notice] sent poll-read bai FlowTempDesired QQ=31: 0.00
2023-10-12 23:56:31.538 [update notice] sent scan-read scan.15 id QQ=31: 
2023-10-12 23:56:31.636 [main error] scan config 15: ERR: wrong symbol received
2023-10-12 23:56:35.214 [update notice] sent poll-read bai FlowTempMax QQ=31: 66.12
2023-10-12 23:56:36.006 [update notice] received read bai Status01 QQ=10: 22.0;22.0;-;-;-;off
2023-10-12 23:56:36.334 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;1;0;0;0
2023-10-12 23:56:36.532 [update notice] received update-read broadcast vdatetime QQ=10: 22:12:02;12.10.2023
2023-10-12 23:56:36.808 [update notice] received poll-read bai Status02 QQ=10: on;60;55.0;80;15.0
2023-10-12 23:56:37.085 [update notice] received unknown MS cmd: 1008b5110100 / 0860010d001f000000
2023-10-12 23:56:37.310 [update notice] received update-read broadcast outsidetemp QQ=10: -40.000
2023-10-12 23:56:41.180 [update notice] sent poll-read bai HcHours QQ=31: 510
2023-10-12 23:56:43.809 [update notice] sent scan-read scan.15 id QQ=31: 
2023-10-12 23:56:43.914 [main error] scan config 15: ERR: SYN received
2023-10-12 23:56:46.004 [update notice] received read bai Status01 QQ=10: 22.0;22.0;-;-;-;off
2023-10-12 23:56:46.288 [update notice] received update-write bai SetMode QQ=10: auto;0.0;60.0;-;1;0;1;0;0;0
2023-10-12 23:56:47.107 [bus error] poll bai HcPumpMode failed: ERR: wrong symbol received
2023-10-12 23:56:47.110 [bus error] device status: eBUS comm error: framing

john30, 2023-10-20:
do you have writable messages at all? check with "ebusctl find -w"

inoxsk, 2023-10-20:
Hi, same issue here. Trying to send a mqtt message results in 
`2023-10-20 21:41:41.616 [mqtt error] write message bai FlowTempDesired not found`
Not sure if my formatting is wrong, or some other mistake..

stefaanderoeck, 2023-10-21:
Found this github issue while trying to figure out why I can't set anything (first time setup), I think I'm having similar problems:

```
root@58709eeec8de:/# grep ^filter-direction /etc/ebusd/mqtt-hassio.cfg
filter-direction = r|u|^w
root@58709eeec8de:/# ebusctl info
version: ebusd 23.2.23.2
device: ebus-62b5c8.lan:9999, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 23
max symbol rate: 81
min arbitration micros: 8
max arbitration micros: 90
min symbol latency: 9
max symbol latency: 21
scan: finished
reconnects: 0
masters: 2
messages: 202
conditional: 2
poll: 68
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0405;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

root@58709eeec8de:/# ebusctl find -w
broadcast id = no data stored
broadcast queryexistence = no data stored
broadcast vdatetime = no data stored

root@58709eeec8de:/# ebusctl find -r | wc -l
164
root@58709eeec8de:/# 
```

Pradar123, 2023-10-21:
I have the same issue. However i cant seem to get the ebusctl to run anywhere (its possibly on my end) ssh'ing the HA it sends back unknown command _ebusctl_.  tried to ssh the adapter itself (just to make sure) but i couldnt., Sorry im just a layman at this point. i have HA running on VM, all the sensors from ebus adapter arriving but no option to write anything.
im using the _LukasGrebe ha-addons_ rep for connection.

john30, 2023-10-22:
> FlowTempDesired

look out for the message definition using "ebusctl f -a -l * -w". if it appears as writable there, then you have to check your settings (e.g. acl file missing)

john30, 2023-10-22:
> I have the same issue. However i cant seem to get the ebusctl to run anywhere (its possibly on my end) ssh'ing the HA it sends back unknown command _ebusctl_. tried to ssh the adapter itself (just [...]

you don't need ebusctl, a simple netcat is doing quite the same. the adapter can't do ssh (would be fancy though).

inoxsk, 2023-10-22:
The adapter should communicate over netcat? I get no response trying to access it by ncat..

john30, 2023-10-25:
not the adapter, you can communicate with ebusd using netcat.
didn't I suggest a different command to use for checking?

inoxsk, 2023-10-26:
You did, at first it did not work, it took me some time to get it to work - had to change it to "ebusctl f -a -l "*" -w". Result 

> bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai APCLegioProtection = no data stored (message not available due to condition)
bai BlockTimeHcMax = no data stored
bai clearerrorhistory = no data stored
bai DSNOffset = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FlowsetHcMax = no data stored
bai FlowsetHwcMax = no data stored
bai HcPumpMode = no data stored
bai HoursTillService = no data stored
bai HwcPostrunTime = no data stored
bai HwcTempMax = no data stored
bai PartloadHcKW = no data stored
bai PartloadHwcKW = no data stored
bai PrEnergyCountHc1 = no data stored
bai PrEnergyCountHc2 = no data stored
bai PrEnergyCountHc3 = no data stored
bai PrEnergyCountHwc1 = no data stored
bai PrEnergyCountHwc2 = no data stored
bai PrEnergyCountHwc3 = no data stored
bai PrEnergySumHc1 = no data stored
bai PrEnergySumHc2 = no data stored
bai PrEnergySumHc3 = no data stored
bai PrEnergySumHwc1 = no data stored
bai PrEnergySumHwc2 = no data stored
bai PrEnergySumHwc3 = no data stored
bai ReturnRegulation = no data stored
bai SecondPumpMode = no data stored
bai SetFactoryValues = no data stored
bai SolPostHeat = no data stored
bai StorageLoadTimeMax = no data stored
bai ValveMode = no data stored
bai WarmstartOffset = no data stored
bai WPPostrunTime = no data stored
broadcast id = no data stored
broadcast queryexistence =  (empty for 31fe07fe00 / )
broadcast vdatetime = no data stored
memory eeprom = no data stored
memory ram = no data stored

GrumpyMeow, 2023-10-28:
I think i have the same issue. I do see "select" and "number" entities being created for my thermostat (350), but nothing for my Vaillant-boiler (bai).
```
ebusctl info
version: ebusd 23.2.p20230716
update check: revision 23.2 available
device: /dev/ttyUSB0
access: *
signal: acquired
symbol rate: 23
max symbol rate: 123
min arbitration micros: 1
max arbitration micros: 4
min symbol latency: 5
max symbol latency: 7
scan: finished
reconnects: 0
masters: 3
messages: 369
conditional: 3
poll: 262
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=35000;SW=0109;HW=7102", loaded "vaillant/15.350.HW7102.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
I'm using the Home Assistant addon and configured: ``

In MQTT-explorer i see the following "sensor" (while i would expect it as a writable-type):
```
{
  "unique_id":"ebusd_bai_HwcPostrunTime_0",
  "name":"ebusd bai HwcPostrunTime ",
  "device":{
   "identifiers":"ebusd_bai",
   "manufacturer":"ebusd.eu",
   "name":"ebusd bai",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"CV-Hok"
  },
  "value_template":"{{value_json[\"0\"].value}}",
  "state_topic":"ebusd/bai/HwcPostrunTime",
  "unit_of_measurement":"s"
 }
```
in the config definition file it is specified as:
```
r;wi,,HwcPostrunTime,d.72 WW Pumpnenachlaufzeit,,,,"1104",,,UCH,-10,s,Pumpennachlaufzeit nach Speicherladung
```

GrumpyMeow, 2023-10-28:
I'm pretty sure it has got something to do with the "%direction" token replacement. This as the entity "Hc12stepanalogCtrl" is being registered as a sensor, while it's registered as "r;wi" here: https://github.com/john30/ebusd-configuration/blob/4675cd03b2763867cf74bd12426b1990bebb6204/ebusd-2.1.x/de/vaillant/15.370.csv#L34

Other writable fields with "r;w" do work.

john30, 2023-12-17:
if ws is used, then it is a matter of assigning more rights to the default user by using an ACL file and define a "mqtt" user there or just loosen the default rights (by using the "--accesslevel=..." option)

dialerp, 2023-12-20:
Is there a way for the data from the "bai" section to refresh and be sent via MQTT? Currently, only data from the VR71 section and "bai Status01";"bai Status01" are correctly sent.
[image]
log
[image]

The next issue is that attempts to query gas consumption are returning an error.
[image]

dialerp, 2024-01-11:
Perhaps someone will find my solution useful. On Debian, where ebusd is installed, I've run a script that sends the necessary data via MQTT to my MQTT broker
`#!/bin/bash
while true; do
    result_temp=$(ebusctl read StorageTemp)
    sleep 1
    result_temp_desired=$(ebusctl read StorageTempDesired)

    /usr/bin/mosquitto_pub -h 192.168.X.X -u YOUR_USER -P YOUR_PASS -t "ebus/StorageTemp" -m "$result_temp"
    /usr/bin/mosquitto_pub -h 192.168.X.X -u YOUR_USER -P YOUR_PASS -t "ebus/StorageTempDesired" -m "$result_temp_desired"

    sleep 60
done`

The only issue I can't solve is obtaining data on gas consumption.
`ebusctl read PrEnergyCountHc1
ERR: invalid position in decode
`

xenorio, 2025-12-18:
currently running into the exact same issue, although i run ebusd directly (systemd service on arch) instead of via some HA addon. got my adapter (C6 stick) connected to ebusd, got ebusd connected to HA via a mosquitto instance in a docker container, HA only sees sensors i can't interact with:

[image]

my heater is a Vaillant Atmotec Plus. i pretty much just ripped the existing Vaillant calorMATIC 350 thermostat off the wall and connected the adapter stick to the wire i was left with.

i have set the filter-direction parameter and ``accesslevel=*``!

have tried messing with the csv configuration by cloning the config repo, changing what i wanna be able to set from ReadonlyRegister to just Register in ``vaillant/bai.308523_inc.tsp``, then building that and then pointing --configpath towards the result, however whatever i did i never got it to make anything writable. should mention that i don't know anything about ebus communication, i just wanna set my apartment temperature via network.

config:
```
# /etc/conf.d/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="-d ens:192.168.3.9:9999 \
  --scanconfig \
  --accesslevel=* \
  --logareas=bus,main \
  --loglevel=debug \
  --port 8088 \
  --mqttjson \
  --mqttvar=filter-direction=r|u|^w \
  --mqtthost 127.0.0.1 \
  --mqttport 1883 \
  --mqttint /etc/ebusd/mqtt-hassio.cfg"
```

ebusd info:
```
version: ebusd 25.1.25.1-16-g255f1768
update check: revision 25.1 available
device: 192.168.3.9:9999, TCP, enhanced, firmware 1.1[5c0e].1[5c0e]
access: *
signal: acquired
symbol rate: 22
max symbol rate: 48
min arbitration micros: 0
max arbitration micros: 13
min symbol latency: 4
max symbol latency: 11
scan: finished
reconnects: 0
masters: 2
messages: 235
conditional: 3
poll: 73
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([id_hw=7401]), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
```

``find`` output:
```
bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai ACRoomthermostat = no data stored
bai AverageIgnitiontime = no data stored
bai BlockTimeHcMax = no data stored
bai BoilerType = no data stored
bai ChangesDSN = no data stored
bai CirPump = no data stored
bai CounterStartattempts1 = 0
bai CounterStartattempts2 = 0
bai CounterStartAttempts3 = 0
bai CounterStartAttempts4 = 0
bai Currenterror = no data stored
[... 199 lines cut ...]
bai PrEnergyCountHwc1 = no data stored
bai PrEnergyCountHwc2 = no data stored
bai PrEnergyCountHwc3 = no data stored
bai PrEnergySumHc1 = no data stored
bai PrEnergySumHc2 = no data stored
bai PrEnergySumHc3 = no data stored
bai PrEnergySumHwc1 = no data stored
bai PrEnergySumHwc2 = no data stored
bai PrEnergySumHwc3 = no data stored
bai PumpPowerDesired = no data stored
bai ReturnRegulation = no data stored
bai SecondPumpMode = no data stored
bai SetFactoryValues = no data stored
bai SetMode = no data stored
bai SolPostHeat = no data stored
bai StorageLoadTimeMax = no data stored
bai ValveMode = no data stored
bai WarmstartOffset = no data stored
bai WPPostrunTime = no data stored
Broadcast IdQuery = no data stored
Broadcast Queryexistence =  (empty for 31fe07fe00 / )
Broadcast Vdatetime = no data stored
Memory Eeprom = no data stored
Memory Ram = no data stored
```