john30/ebusd #1046: Vaillant Config 76 missing?

apbarratt, 2023-10-21:
### Description

I'm afraid I'm feeling rather out of my depth right now, but until recently I was able to see things like my hot water temperature and when my heat pump was running.  It all stopped appearing a couple of weeks ago so today I've taken a look around and am finding nothing is being sent to my MQTT server at all anymore.

I've pulled latest from the ebusd-configuration repo and then restarted with error logging.  It seems to complain about a missing 76 config file then just starts interating through various sensors it's erroring on.

I'm running eBusd within the Home Assistant Addon.  I'm not sure if that's at all relevant though.

I'm using the eBus Adapter version 5.

Please advice if there is anything useful I can provide in terms of additional logs or settings.

### Actual behavior

It seems to be complaining about a missing Config 76.  I can confirm no such file seems to exist in the vaillant directory.

### Expected behavior

I suppose I'd expect there to be a config 76 file if it's looking for one.

### ebusd version

23.2

### ebusd arguments

ebusd --foreground --mqtthost=homeassistant.local --mqttport=1883 --mqttuser=homeassistant --mqttpass=MYPASSWORD --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en --latency=10 --accesslevel=* --pollinterval=5 --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.155:9999 --log=all:error

### Operating system

other

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 5 via USB

### Logs

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=homeassistant.local --mqttport=1883 --mqttuser=homeassistant --mqttpass=MY_PASSWORD --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en --latency=10 --accesslevel=* --pollinterval=5 --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.155:9999 --log=all:error
2023-10-21 17:38:32.444 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2023-10-21 17:38:32.444 [main error] scan config 76: ERR: element not found
2023-10-21 17:39:12.182 [bus error] poll basv Hc1HeatCurveAdaption failed: ERR: SYN received
2023-10-21 17:40:00.136 [bus error] poll basv Hc2AutoOffMode failed: ERR: SYN received
2023-10-21 17:41:24.198 [bus error] poll basv Hc3FlowTemp failed: ERR: SYN received
2023-10-21 17:42:07.082 [bus error] poll basv Hc3Status failed: ERR: arbitration lost
2023-10-21 17:42:11.011 [bus error] signal lost
2023-10-21 17:42:14.022 [bus error] poll basv Hc3SummerTempLimit failed: ERR: no signal
2023-10-21 17:42:14.022 [bus error] signal lost
2023-10-21 17:42:19.635 [bus error] arbitration start error
2023-10-21 17:42:29.636 [bus error] unable to open 192.168.1.155:9999: ERR: generic I/O error
2023-10-21 17:42:46.333 [update error] unable to parse poll-read basv HolidayTemp from 3115b52406020000008200 / 00: ERR: invalid position
2023-10-21 17:42:50.628 [mqtt error] decode basv HolidayTemp: ERR: invalid position
2023-10-21 17:43:04.405 [bus error] poll basv HwcOpMode failed: ERR: read timeout
2023-10-21 17:43:34.126 [bus error] poll basv HwcTempDesired failed: ERR: wrong symbol received
[... 42 lines cut ...]
2023-10-21 17:50:25.342 [mqtt error] decode hmu YieldThisYear3: ERR: invalid position
2023-10-21 17:50:28.299 [update error] unable to parse poll-read hmu YieldThisYear4 from 3108b516041a010400 / 00: ERR: invalid position
2023-10-21 17:50:30.351 [mqtt error] decode hmu YieldThisYear4: ERR: invalid position
2023-10-21 17:50:34.286 [update error] unable to parse poll-read hmu YieldThisYear5 from 3108b516041a010500 / 00: ERR: invalid position
2023-10-21 17:50:40.276 [update error] unable to parse poll-read hmu YieldThisYear6 from 3108b516041a010600 / 00: ERR: invalid position
2023-10-21 17:50:40.410 [mqtt error] decode hmu YieldThisYear5: ERR: invalid position
2023-10-21 17:50:45.462 [mqtt error] decode hmu YieldThisYear6: ERR: invalid position
2023-10-21 17:50:46.302 [update error] unable to parse poll-read hmu YieldThisYear7 from 3108b516041a010700 / 00: ERR: invalid position
2023-10-21 17:50:50.476 [mqtt error] decode hmu YieldThisYear7: ERR: invalid position
2023-10-21 17:50:52.277 [update error] unable to parse poll-read hmu YieldThisYear8 from 3108b516041a010800 / 00: ERR: invalid position
2023-10-21 17:50:55.526 [mqtt error] decode hmu YieldThisYear8: ERR: invalid position
2023-10-21 17:50:58.117 [bus error] poll hmu YieldThisYear9 failed: ERR: SYN received
2023-10-21 17:51:04.256 [update error] unable to parse poll-read hmu YieldTotal from 3108b516021801 / 00: ERR: invalid position
2023-10-21 17:51:10.553 [mqtt error] decode hmu YieldTotal: ERR: invalid position
2023-10-21 17:51:58.153 [bus error] poll basv Hc1MaxFlowTempDesired failed: ERR: SYN received
2023-10-21 17:52:04.490 [bus error] poll basv Hc1MinFlowTempDesired failed: ERR: read timeout
2023-10-21 17:52:07.000 [bus error] signal lost
2023-10-21 17:52:09.054 [bus error] signal lost
2023-10-21 17:52:13.030 [bus error] poll basv Hc1PumpStatus failed: ERR: no signal
2023-10-21 17:52:13.030 [bus error] signal lost
2023-10-21 17:52:18.411 [bus error] arbitration start error
2023-10-21 17:52:28.417 [bus error] unable to open 192.168.1.155:9999: ERR: generic I/O error
2023-10-21 17:52:52.147 [bus error] poll basv Hc2ActualFlowTempDesired failed: ERR: SYN received
2023-10-21 17:52:58.118 [bus error] poll basv Hc2AutoOffMode failed: ERR: SYN received
2023-10-21 17:53:58.181 [bus error] poll basv Hc2SummerTempLimit failed: ERR: SYN received

apbarratt, 2023-10-21:
Couple of edits to above to give clearer logs and more detail of setup.

apbarratt, 2023-10-21:
I figured it'd be sensible to try updating the firmware on the adapter, so it is now at build [20231015](https://adapter.ebusd.eu/v5/ChangeLog#20231015) (glad to see the endless empty password prompt has been fixed!)

Anyway, after doing that, the logs have a new interesting message in them that I suspect may shed some clues:

`2023-10-21 19:10:56.485 [bus error] device status: eBUS comm error: overrun`

I'm now starting to wonder if my adapter might have burnt something out after a few months of usage?  Either that or something on the heat pump setup itself, which is itself relatively new too.

A longer chunk of those new logs:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ge9fashooM5eiraeja9ohr6oo8jaegood5igh3eeghahkobei4pai1odie2eichu --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en --latency=10 --accesslevel=* --pollinterval=5 --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.155:9999 --log=all:error
2023-10-21 19:10:56.485 [bus error] device status: eBUS comm error: overrun
2023-10-21 19:10:58.073 [main error] scan config 15: ERR: read timeout
2023-10-21 19:11:01.087 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2023-10-21 19:11:01.088 [main error] scan config 76: ERR: read timeout
2023-10-21 19:11:49.232 [update error] unable to parse poll-read hmu Status02 from 3108b5110102 / 00: ERR: invalid position
2023-10-21 19:11:53.496 [mqtt error] decode hmu Status02: ERR: invalid position
2023-10-21 19:11:55.222 [update error] unable to parse poll-read hmu Status16 from 3108b5040116 / 00: ERR: invalid position
2023-10-21 19:11:58.505 [mqtt error] decode hmu Status16: ERR: invalid position
2023-10-21 19:12:01.353 [update error] unable to parse poll-read hmu Status from 3108b5110103 / 00: ERR: invalid position
2023-10-21 19:12:04.403 [mqtt error] decode hmu Status: ERR: invalid position
2023-10-21 19:12:07.283 [update error] unable to parse poll-read hmu YieldThisYear10 from 3108b516041a010a00 / 00: ERR: invalid position
2023-10-21 19:12:11.803 [mqtt error] decode hmu YieldThisYear10: ERR: invalid position
2023-10-21 19:12:13.294 [update error] unable to parse poll-read hmu YieldThisYear11 from 3108b516041a010b00 / 00: ERR: invalid position
2023-10-21 19:12:16.824 [mqtt error] decode hmu YieldThisYear11: ERR: invalid position
2023-10-21 19:12:19.283 [update error] unable to parse poll-read hmu YieldThisYear12 from 3108b516041a010c00 / 00: ERR: invalid position
2023-10-21 19:12:21.835 [mqtt error] decode hmu YieldThisYear12: ERR: invalid position
2023-10-21 19:12:25.263 [update error] unable to parse poll-read hmu YieldThisYear1 from 3108b516041a010100 / 00: ERR: invalid position
2023-10-21 19:12:26.883 [mqtt error] decode hmu YieldThisYear1: ERR: invalid position
2023-10-21 19:12:31.273 [update error] unable to parse poll-read hmu YieldThisYear2 from 3108b516041a010200 / 00: ERR: invalid position
2023-10-21 19:12:36.944 [mqtt error] decode hmu YieldThisYear2: ERR: invalid position
2023-10-21 19:12:37.254 [update error] unable to parse poll-read hmu YieldThisYear3 from 3108b516041a010300 / 00: ERR: invalid position
2023-10-21 19:12:41.953 [mqtt error] decode hmu YieldThisYear3: ERR: invalid position
2023-10-21 19:12:43.295 [update error] unable to parse poll-read hmu YieldThisYear4 from 3108b516041a010400 / 00: ERR: invalid position
2023-10-21 19:12:47.004 [mqtt error] decode hmu YieldThisYear4: ERR: invalid position
2023-10-21 19:12:49.283 [update error] unable to parse poll-read hmu YieldThisYear5 from 3108b516041a010500 / 00: ERR: invalid position
2023-10-21 19:12:52.013 [mqtt error] decode hmu YieldThisYear5: ERR: invalid position
2023-10-21 19:12:55.333 [update error] unable to parse poll-read hmu YieldThisYear6 from 3108b516041a010600 / 00: ERR: invalid position
2023-10-21 19:12:57.062 [mqtt error] decode hmu YieldThisYear6: ERR: invalid position
2023-10-21 19:13:01.355 [update error] unable to parse poll-read hmu YieldThisYear7 from 3108b516041a010700 / 00: ERR: invalid position
2023-10-21 19:13:07.122 [mqtt error] decode hmu YieldThisYear7: ERR: invalid position
2023-10-21 19:13:07.395 [update error] unable to parse poll-read hmu YieldThisYear8 from 3108b516041a010800 / 00: ERR: invalid position
2023-10-21 19:13:12.130 [mqtt error] decode hmu YieldThisYear8: ERR: invalid position
2023-10-21 19:13:13.274 [update error] unable to parse poll-read hmu YieldThisYear9 from 3108b516041a010900 / 00: ERR: invalid position
2023-10-21 19:13:17.180 [mqtt error] decode hmu YieldThisYear9: ERR: invalid position
2023-10-21 19:13:19.238 [update error] unable to parse poll-read hmu YieldTotal from 3108b516021801 / 00: ERR: invalid position
2023-10-21 19:13:22.189 [mqtt error] decode hmu YieldTotal: ERR: invalid position
2023-10-21 19:13:49.599 [bus error] poll basv Hc1ExcessTemp failed: ERR: arbitration lost
```

apbarratt, 2023-10-21:
Hmmm, never mind, that seems to have gone away after I did a general reboot of everything including the heatpump power itself:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ge9fashooM5eiraeja9ohr6oo8jaegood5igh3eeghahkobei4pai1odie2eichu --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en --latency=10 --accesslevel=* --pollinterval=5 --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.155:9999 --log=all:error
2023-10-21 19:19:50.008 [main error] scan config 15: ERR: SYN received
2023-10-21 19:19:52.933 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2023-10-21 19:19:52.933 [main error] scan config 76: ERR: read timeout
2023-10-21 19:19:59.043 [bus error] poll hmu CurrentYieldPower failed: ERR: no signal
2023-10-21 19:19:59.043 [bus error] signal lost
2023-10-21 19:20:09.316 [bus error] poll hmu EnergyHc failed: ERR: SYN received
2023-10-21 19:20:15.206 [bus error] poll hmu EnergyIntegral failed: ERR: SYN received
2023-10-21 19:20:27.474 [bus error] poll hmu HoursCool failed: ERR: read timeout
2023-10-21 19:20:39.227 [update error] unable to parse poll-read hmu Status02 from 3108b5110102 / 00: ERR: invalid position
2023-10-21 19:20:43.258 [mqtt error] decode hmu Status02: ERR: invalid position
2023-10-21 19:20:45.269 [update error] unable to parse poll-read hmu Status16 from 3108b5040116 / 00: ERR: invalid position
2023-10-21 19:20:48.265 [mqtt error] decode hmu Status16: ERR: invalid position
2023-10-21 19:20:51.197 [update error] unable to parse poll-read hmu Status from 3108b5110103 / 00: ERR: invalid position
2023-10-21 19:20:53.314 [mqtt error] decode hmu Status: ERR: invalid position
2023-10-21 19:20:57.132 [bus error] poll hmu YieldThisYear10 failed: ERR: SYN received
2023-10-21 19:21:03.277 [update error] unable to parse poll-read hmu YieldThisYear11 from 3108b516041a010b00 / 00: ERR: invalid position
2023-10-21 19:21:08.340 [mqtt error] decode hmu YieldThisYear11: ERR: invalid position
2023-10-21 19:21:09.278 [update error] unable to parse poll-read hmu YieldThisYear12 from 3108b516041a010c00 / 00: ERR: invalid position
2023-10-21 19:21:13.388 [mqtt error] decode hmu YieldThisYear12: ERR: invalid position
2023-10-21 19:21:15.268 [bus error] poll hmu YieldThisYear1 failed: ERR: read timeout
2023-10-21 19:21:17.015 [bus error] signal lost
2023-10-21 19:21:20.091 [bus error] signal lost
2023-10-21 19:21:28.942 [update error] unable to parse poll-read hmu YieldThisYear2 from 3108b516041a010200 / 00: ERR: invalid position
2023-10-21 19:21:33.424 [mqtt error] decode hmu YieldThisYear2: ERR: invalid position
2023-10-21 19:21:34.290 [update error] unable to parse poll-read hmu YieldThisYear3 from 3108b516041a010300 / 00: ERR: invalid position
2023-10-21 19:21:38.447 [mqtt error] decode hmu YieldThisYear3: ERR: invalid position
2023-10-21 19:21:40.318 [update error] unable to parse poll-read hmu YieldThisYear4 from 3108b516041a010400 / 00: ERR: invalid position
2023-10-21 19:21:43.508 [mqtt error] decode hmu YieldThisYear4: ERR: invalid position
2023-10-21 19:21:46.288 [update error] unable to parse poll-read hmu YieldThisYear5 from 3108b516041a010500 / 00: ERR: invalid position
2023-10-21 19:21:48.556 [mqtt error] decode hmu YieldThisYear5: ERR: invalid position
2023-10-21 19:21:52.376 [bus error] poll hmu YieldThisYear6 failed: ERR: read timeout
2023-10-21 19:21:58.203 [bus error] poll hmu YieldThisYear7 failed: ERR: SYN received
2023-10-21 19:22:05.009 [bus error] poll hmu YieldThisYear8 failed: ERR: no signal
2023-10-21 19:22:05.009 [bus error] signal lost
2023-10-21 19:22:09.088 [bus error] signal lost
2023-10-21 19:22:10.187 [bus error] poll hmu YieldThisYear9 failed: ERR: SYN received
2023-10-21 19:22:16.228 [update error] unable to parse poll-read hmu YieldTotal from 3108b516021801 / 00: ERR: invalid position
2023-10-21 19:22:18.542 [mqtt error] decode hmu YieldTotal: ERR: invalid position
2023-10-21 19:22:22.197 [bus error] poll basv AdaptHeatCurve failed: ERR: SYN received
2023-10-21 19:22:28.895 [bus error] poll basv DisplayedOutsideTemp failed: ERR: arbitration lost
```

apbarratt, 2023-10-21:
But now that the UI for the adapter is being more accommodating, I can see a number of errors on the Log page for it.  I don't know if this indicates the issue lies with the adapter, if it does, I suppose I'm in the wrong repo:

Time | Severity | Log
19:18:33 | Info | execute: var isnew 1
-- | -- | --
19:18:33 | Info | execute: if "§isnvs ebus.host" "var isnew 0"
19:18:33 | Info | execute: isnvs ebus.host
19:18:33 | Info | execute: var isnew 0
19:18:33 | Info | execute: if "§isnvs sta.on" "var isnew 0"
19:18:33 | Info | execute: isnvs sta.on
19:18:33 | Info | execute: var isnew 0
19:18:33 | Info | execute: if "§isnvs eth.on" "var isnew 0"
19:18:33 | Info | execute: isnvs eth.on
19:18:33 | Warning | execute: command failed: isnvs eth.on = 1
19:18:33 | Info | execute: on -lo iol 9 "led 0 255" "led 1 255" "led 2 255" "sleep 500" restart
19:18:33 |   | io trigger installed
19:18:33 | Info | execute: on iol 9 "led 2 50 0"
19:18:33 |   | io trigger installed
19:18:33 | Info | execute: if "var skipstart" "led 0 255 0 5000" "led 2 50" "ap EBUS" server exit
19:18:33 | Info | execute: var skipstart
19:18:33 |   | variable skipstart = 0
19:18:33 | Info | execute: function start server "led 0 10"
19:18:33 |   | function installed
19:18:33 | Info | execute: on wifi 1 start
19:18:33 |   | network trigger installed
19:18:33 | Info | execute: on eth 1 start
19:18:33 |   | network trigger installed
19:18:33 | Info | execute: if "isnvs sta.on" "led 0 10 255 5000" "sleep 100" sta
19:18:33 | Info | execute: isnvs sta.on
19:18:33 | Info | execute: led 0 10 255 5000
19:18:33 | Info | execute: sleep 100
19:18:33 | Info | statusLedTimerFn: duty set to 10
19:18:33 | Info | execute: sta
19:18:33 | Info | pp: pp rom version: 9387209
19:18:33 | Info | net80211: net80211 rom version: 9387209
19:18:33 | Info | wifi_init: tcpip mbox: 32
19:18:33 | Info | wifi_init: udp mbox: 6
19:18:33 | Info | wifi_init: tcp mbox: 6
19:18:33 | Info | wifi_init: tcp tx win: 5744
19:18:33 | Info | wifi_init: tcp rx win: 5744
19:18:33 | Info | wifi_init: tcp mss: 1440
19:18:33 | Info | wifi_init: WiFi IRAM OP enabled
19:18:33 | Info | wifi_init: WiFi RX IRAM OP enabled
19:18:33 | Info | wifi_init: LWIP IRAM OP enabled
19:18:33 | Info | phy_init: phy_version 1130,b4e4b80,Sep 5 2023,11:09:30
19:18:33 |   | sta connecting to 'Vote Yes for Scotland'
19:18:36 | Info | cmd_wifi: sta connect
19:18:38 | Info | esp_netif_handlers: sta ip: 192.168.1.155, mask: 255.255.255.0, gw: 192.168.1.254
19:18:38 | Info | cmd_wifi: sta got ip
19:18:38 | Info | repl_event_task: wifi event: 1
19:18:38 | Info | execute: led 0 50
19:18:38 | Info | execute: start
19:18:38 | Info | execute: server
19:18:38 | Info | esp-rest: Starting HTTP Server
19:18:38 |   | server started
19:18:38 | Info | execute: led 0 10
19:18:38 | Info | execute: if "var isnew" "ap EBUS" else "if \"isnvs ap.on\" ap"
19:18:38 | Info | execute: var isnew
19:18:38 |   | variable isnew = 0
19:18:38 | Info | execute: if "isnvs ap.on" ap
19:18:38 | Info | execute: isnvs ap.on
19:18:38 | Info | execute: var eth 0
19:18:38 | Info | execute: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""
19:18:38 | Info | execute: eth -t
19:18:38 | Error | w5500.mac: w5500_reset(302): reset timeout
19:18:38 | Error | w5500.mac: emac_w5500_init(793): reset w5500 failed
19:18:38 | Error | esp_eth: esp_eth_driver_install(228): init mac failed
19:18:38 | Error | eth: initialise_ethernet(250): install driver
19:18:38 |   | Ethernet on USR-ES1 not available
19:18:38 | Warning | execute: command failed: eth -t = 1
[... 37 lines cut ...]
19:18:38 | Info | host_task_tcp: listening
19:18:38 | Info | init_ebus: timer started
19:18:38 | Info | execute: led 0 50
19:18:38 | Info | execute: exit
19:18:38 | Info | execute: command exit
19:18:38 | Info | execute: command exit
19:18:38 | Info | replExecuteFromFile: successfully executed 41/41 commands from '/data/main.eas'
19:18:38 |   | Welcome to eBUS Adapter Shield Interface! Type 'help' to get the list of commands.
19:18:38 |   | Your terminal application does not support escape sequences.
19:18:38 | Info | main_task: Returned from app_main()
19:18:38 | Info | statusLedTimerFn: duty set to 10
19:18:38 | Info | statusLedTimerFn: duty set to 50
19:18:44 | Info | sntpTimeCb: got NTP time: 2023-10-21 18:18:44
19:19:35 | Info | host_task_tcp: accepted: 192.168.1.199
19:21:16 | Info | statusLedTimerFn: btn short pressed 0
19:21:16 | Info | execute: led 2 50 0
19:21:16 |   | 2 duty set to 50
19:21:16 |   | 2 fade started
19:21:25 | Error | ebus_tcp_server_loop: recv: errno 113
19:21:25 | Error | ebus_tcp_server_loop: recv: errno 128
19:21:25 | Info | host_task_tcp: closed: 192.168.1.199
19:21:25 | Info | host_task_tcp: listening
19:21:28 | Info | host_task_tcp: accepted: 192.168.1.199
19:21:31 | Info | statusLedTimerFn: btn short pressed 1
19:21:33 | Info | statusLedTimerFn: btn long pressed 1

munkien, 2023-10-23:
It looks like its using a local configuration path?
Maybe the files there are corrupt/deleted/missing?
Can you try to remove the following from the ebusd launch? 
Alternatively replace it with "--configpath=https://cfg.ebusd.eu/"

--configpath=/config/ebusd-configuration/latest/en

apbarratt, 2023-10-23:
Thanks @munkien, it's good to know I can just link to a live link instead of having to update my repo all the time.  Unfortunately this does not resolve my issues as a vaillant file prefixed with 76 simply does not exist.

I guess my bigger concern though is why this used to work and has since stopped :/

apbarratt, 2023-10-23:
[image]

munkien, 2023-10-23:
Can you try to increase logging level to all:notice ? In hope that something useful is hidden? 
I have a valliant boiler with a 76-config too and that seems to work.

apbarratt, 2023-10-23:
Just starting it back up with notice log level now.  I'll give it about 15 minutes to get a good chunk of stuff then will investigate how to access a shell to run the find all command :)

apbarratt, 2023-10-23:
Well that was a bad idea, turns out home assistant limits the tail of the log so I lost the beginning.  Started again and this is what I'm getting:
```
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ge9fashooM5eiraeja9ohr6oo8jaegood5igh3eeghahkobei4pai1odie2eichu --scanconfig --mqttjson --configpath=https://cfg.ebusd.eu/ --latency=10 --accesslevel=* --pollinterval=5 --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.155:9999 --log=all:notice
2023-10-23 11:22:35.668 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.155:9999, enhanced
2023-10-23 11:22:38.408 [bus notice] device status: resetting
2023-10-23 11:22:38.410 [bus notice] bus started with own address 31/36
2023-10-23 11:22:38.430 [bus notice] signal acquired
2023-10-23 11:22:38.746 [mqtt notice] connection established
2023-10-23 11:22:39.000 [bus notice] device status: reset, supports info
2023-10-23 11:22:39.038 [bus notice] device status: extra info: firmware 1.1[3a0f].1[3a0f], jumpers 0x0f
2023-10-23 11:22:41.442 [bus notice] new master 10, master count 2
2023-10-23 11:22:41.491 [bus notice] new master 03, master count 3
2023-10-23 11:22:41.491 [update notice] received unknown MS cmd: 1008b5110101 / 0913120080ffff0000ff
2023-10-23 11:22:41.758 [bus notice] new master 71, master count 4
2023-10-23 11:22:41.758 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff6c000019
2023-10-23 11:22:41.856 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:22:42.081 [update notice] received unknown MS cmd: 1076b512030f0201 / 07620300c900ff03
2023-10-23 11:22:42.336 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2023-10-23 11:22:42.600 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-10-23 11:22:45.281 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:22:46.161 [update notice] received unknown MS cmd: 0376b51206130011000006 / 0200ff
2023-10-23 11:22:47.154 [update notice] received unknown MS cmd: 7108b51a03048f32 / 0e8f07c0cfc07b3c00901b20000000
2023-10-23 11:22:47.538 [update notice] received unknown MS cmd: 7108b51a03049033 / 0e900000c083030000000000000000
2023-10-23 11:22:47.919 [update notice] received unknown MS cmd: 7108b51a03049134 / 0e91cf2000c0022101001800061800
2023-10-23 11:22:48.347 [update notice] received unknown MS cmd: 7108b51a03049235 / 0e9200000000000100900100000000
2023-10-23 11:22:48.558 [main error] scan config 08: ERR: SYN received
2023-10-23 11:22:48.843 [update notice] received unknown MS cmd: 7108b51a03049336 / 0e9300000000000000000000000000
2023-10-23 11:22:49.290 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:22:50.802 [bus notice] scan 15: ;Vaillant;BASV2;0562;6304
2023-10-23 11:22:50.802 [update notice] store 15 ident: done
2023-10-23 11:22:50.802 [update notice] sent scan-read scan.15  QQ=31: Vaillant;BASV2;0562;6304
2023-10-23 11:22:50.802 [bus notice] scan 15: ;Vaillant;BASV2;0562;6304
2023-10-23 11:22:51.380 [update notice] received unknown MS cmd: 1008b5110101 / 0913120080ffff0000ff
2023-10-23 11:22:51.535 [main notice] read common config file vaillant/broadcast.csv
2023-10-23 11:22:51.652 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff6c000019
2023-10-23 11:22:51.708 [main notice] read common config file vaillant/scan.csv
2023-10-23 11:22:51.880 [main notice] read common config file vaillant/general.csv
2023-10-23 11:22:51.913 [update notice] received unknown MS cmd: 1076b512030f0201 / 07620300c900ff03
2023-10-23 11:22:52.131 [main notice] read scan config file vaillant/15.basv.csv for ID "basv2", SW0562, HW6304
2023-10-23 11:22:52.175 [update notice] received unknown MS cmd: 1008b51009000000ffffff070000 / 0101
2023-10-23 11:22:52.335 [main notice] found messages: 457 (0 conditional on 0 conditions, 0 poll, 8 update)
2023-10-23 11:22:52.443 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-10-23 11:22:52.651 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323234363030
2023-10-23 11:22:52.761 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:22:52.950 [update notice] sent scan-read scan.15 id QQ=31: 
2023-10-23 11:22:53.011 [bus notice] max. symbols per second: 113
2023-10-23 11:22:53.226 [update notice] sent scan-read scan.15 id QQ=31: 
2023-10-23 11:22:53.515 [update notice] sent scan-read scan.15 id QQ=31: 21;22;46;0020262148;0082;022846;N2
2023-10-23 11:22:53.515 [bus notice] scan 15: ;21;22;46;0020262148;0082;022846;N2
2023-10-23 11:22:54.151 [update notice] received unknown MS cmd: 0376b51206130011000006 / 0200ff
2023-10-23 11:22:54.911 [update notice] sent poll-read basv AdaptHeatCurve QQ=31: no
2023-10-23 11:22:55.790 [bus notice] scan 76: ;Vaillant;VWZIO;0202;0103
2023-10-23 11:22:55.790 [update notice] store 76 ident: done
2023-10-23 11:22:55.790 [update notice] sent scan-read scan.76  QQ=31: Vaillant;VWZIO;0202;0103
2023-10-23 11:22:55.791 [bus notice] scan 76: ;Vaillant;VWZIO;0202;0103
2023-10-23 11:22:56.090 [update notice] sent unknown MS cmd: 3176b5090124 / 09003231323234333030
2023-10-23 11:22:56.255 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:22:56.288 [main error] scan config 76: ERR: SYN received
[... cut ...]
2023-10-23 11:23:16.830 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:23:18.581 [bus error] poll basv Hc1ExcessTemp failed: ERR: read timeout
2023-10-23 11:23:20.301 [update notice] received unknown MS cmd: 7108b5110107 / 0a000000c0003300000000
2023-10-23 11:23:21.090 [update notice] received unknown MS cmd: 1008b5110101 / 0913130080ffff0000ff
2023-10-23 11:23:21.362 [update notice] received unknown MS cmd: 1076b5110101 / 09ffff0080ff6c000019
2023-10-23 11:23:21.625 [update notice] received unknown MS cmd: 1076b512030f0201 / 07620300c900ff03
```

apbarratt, 2023-10-23:
Right, here's the find all from the docker shell:

```
2ad9b828-ebusd:/# ebusctl find -a
basv AdaptHeatCurve = no
basv AdaptHeatCurve = no data stored
basv BankHolidayEndPeriod = no data stored
basv BankHolidayEndPeriod = no data stored
basv BankHolidayStartPeriod = no data stored
basv BankHolidayStartPeriod = no data stored
basv ccTimer.Friday = no data stored
basv ccTimer.Friday = no data stored
basv ccTimer.Monday = no data stored
basv ccTimer.Monday = no data stored
basv ccTimer.Saturday = no data stored
basv ccTimer.Saturday = no data stored
basv ccTimer.Sunday = no data stored
basv ccTimer.Sunday = no data stored
basv ccTimer.Thursday = no data stored
basv ccTimer.Thursday = no data stored
basv ccTimer.Tuesday = no data stored
basv ccTimer.Tuesday = no data stored
basv ccTimer.Wednesday = no data stored
basv ccTimer.Wednesday = no data stored
basv clearerrorhistory = no data stored
basv ContinuosHeating = no data stored
basv ContinuosHeating = no data stored
basv currenterror = no data stored
basv CylinderChargeHyst = no data stored
basv CylinderChargeHyst = no data stored
basv CylinderChargeOffset = no data stored
basv CylinderChargeOffset = no data stored
basv Date = no data stored
basv Date = no data stored
basv DisplayedOutsideTemp = 6.28906
basv errorhistory = no data stored
basv FrostOverRideTime = no data stored
basv FrostOverRideTime = no data stored
basv GlobalSystemOff = no data stored
basv GlobalSystemOff = no data stored
basv Hc1ActualFlowTempDesired = 0.0
basv Hc1AutoOffMode = night
basv Hc1AutoOffMode = no data stored
basv Hc1CircuitType = no data stored
basv Hc1ExcessTemp = no data stored
basv Hc1ExcessTemp = no data stored
basv Hc1FlowTemp = 12.5
basv Hc1HeatCurve = 1.2
basv Hc1HeatCurve = no data stored
basv Hc1HeatCurveAdaption = 0.0
basv Hc1MaxFlowTempDesired = 55
basv Hc1MaxFlowTempDesired = no data stored
basv Hc1MinFlowTempDesired = 15
basv Hc1MinFlowTempDesired = no data stored
basv Hc1MixerMovement = no data stored
basv Hc1PumpStatus = 0
basv Hc1PumpStatus = no data stored
basv Hc1RoomTempSwitchOn = thermostat
basv Hc1RoomTempSwitchOn = no data stored
basv Hc1Status = 0
basv Hc1Status = no data stored
basv Hc1SummerTempLimit = 30
basv Hc1SummerTempLimit = no data stored
basv Hc2ActualFlowTempDesired = 0.0
basv Hc2AutoOffMode = eco
basv Hc2AutoOffMode = no data stored
basv Hc2CircuitType = no data stored
basv Hc2ExcessTemp = 0.0
basv Hc2ExcessTemp = no data stored
basv Hc2FlowTemp = -
[... 417 lines cut ...]
hmu WaterThroughput = no data stored
hmu YieldThisYear10 =  (ERR: invalid position for 3108b516041a010a00 / 00)
hmu YieldThisYear11 =  (ERR: invalid position for 3108b516041a010b00 / 00)
hmu YieldThisYear12 =  (ERR: invalid position for 3108b516041a010c00 / 00)
hmu YieldThisYear1 =  (ERR: invalid position for 3108b516041a010100 / 00)
hmu YieldThisYear2 =  (ERR: invalid position for 3108b516041a010200 / 00)
hmu YieldThisYear3 =  (ERR: invalid position for 3108b516041a010300 / 00)
hmu YieldThisYear4 =  (ERR: invalid position for 3108b516041a010400 / 00)
hmu YieldThisYear5 =  (ERR: invalid position for 3108b516041a010500 / 00)
hmu YieldThisYear6 =  (ERR: invalid position for 3108b516041a010600 / 00)
hmu YieldThisYear7 =  (ERR: invalid position for 3108b516041a010700 / 00)
hmu YieldThisYear8 =  (ERR: invalid position for 3108b516041a010800 / 00)
hmu YieldThisYear9 =  (ERR: invalid position for 3108b516041a010900 / 00)
hmu YieldTotal =  (ERR: invalid position for 3108b516021801 / 00)
memory eeprom = no data stored
memory eeprom = no data stored
memory ram = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = Vaillant;HMU00;0607;5103
scan.15  = Vaillant;BASV2;0562;6304
scan.15 id = 21;22;46;0020262148;0082;022846;N2
scan.76  = Vaillant;VWZIO;0202;0103
scan.76 id = 21;22;43;0010031644;0082;011034;N6
```

apbarratt, 2023-10-23:
That shows all sensors etc that I was receiving before.  They just don't seem to be able to get any data anymore.

apbarratt, 2023-10-23:
Seems whatever 76 is is this thingy.  This manual seems to include a big long appendix full of setting and error codes, but I don't know if those easily translate to whatever the byte code like thingies are that ebus uses :/
https://www.vaillant.co.uk/downloads/aproducts/renewables-1/arotherm-plus/appliance-interface-operating-and-installation-instructions-1799367.pdf

john30, 2023-10-25:
if a csv is missing, then this is the wrong repo anyway. so please continue in ebusd-configuration

closed 2023-10-25 by john30

apbarratt, 2023-11-27:
Hi guys, seems I figured out why my data was no longer coming through (config 76 is of course still missing but I've been able to confirm this was not the cause of the communication breakdown).

It seems that a wee while back when my ebusd_addon updated , a breaking change must have been added to the requirements of `mqtt-hassio.cfg`

The changes that I can see are the following:
`filter-name` now includes a comparison string of 'timer'
`type_map-time =` has changed to `type_map-time = time`
`type_switch-w-number =` gained the values `number,temperature, = temp|,K$` and `number,, = integral|,°min$`
along with a significant number of additions for time based stuff.

By updating my `mqtt-hassio.cfg` to match the latest and re-editing it with the two changes I actually need, I've suddenly got my ebusd back up and running :)

I still don't seem to be able to control anything yet, but I expect that's my need for more understanding there :)

I just wanted to share this update for others who may encounter the same issue, do check to see if you've got an outdated version of mqtt-hassio.cfg as of course it will not necessarily update with the rest of the addon if you have a custom version :)