john30/ebusd #1050: ERR: SYN received during receive command, switching to ready

eyk87, 2023-10-25:
### Description

Hi,

I managed to get the ebus adapter with the newest FW to work with HA and ebusd. When I try to read the bus, I only get the mentioned error.
I also tried different options with no success. The Wifi has a signal strength of -45dbm (100%).
Any ideas what else could cause this issue?
Thank you very much.

### Actual behavior

-

### Expected behavior

-

### ebusd version

23.2

### ebusd arguments

--latency 20000 --acquiretimeout=15000 --receivetimeout=80

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-10-25 23:49:36.160 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2023-10-25 23:49:36.160 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2023-10-25 23:49:36.209 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.259 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.299 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.343 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.430 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2023-10-25 23:49:36.479 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.524 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.570 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.609 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.681 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2023-10-25 23:49:36.692 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2023-10-25 23:49:36.744 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.789 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.829 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.870 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:36.940 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2023-10-25 23:49:36.946 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2023-10-25 23:49:37.009 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.050 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.093 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.139 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.183 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.229 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.269 [bus debug] ERR: SYN received during receive command, switching to ready
2023-10-25 23:49:37.309 [bus debug] ERR: SYN received during receive command, switching to ready

john30, 2023-11-04:
what is the device string you're using? it needs to match what the adapter tells on the top of the UI

TeXiCiTy, 2023-12-10:
Is this on the new Raspberry OS bookworm? I'm getting the same when switching to the newer version.

closed 2023-12-25 by john30