john30/ebusd #1053: Temperaturwerte werden nicht gelesen bei Ebusd Vaillant RecovAir 260/4 - Standalone

hermannf1976, 2023-10-30:
### Description

Hallo John30,
ich verwende das Ebusd Plugin in Loxberry für die Übermittlung der Parameter einer Vaillant RecovAIR 260/4 Wohnraumlüftung.
Soweit hat alles geklappt, ich kann die Daten lesen, ich kann auch Daten schreiben. Lediglich die Parameter mit Temperaturwerten werden nicht übergeben. Wenn diese Parameter in der 08.recov.csv Datei vorhanden sind, werden nur teilweise andere Werte übergeben - die mit Temperaturwerden nie. Wenn die Parameter mit Temperaturwerten ausgeblendet werden, dann werden alle anderen Parameter vollständig gelesen.
Ich habe schon in diversen Fehlerbeschreibungen gesucht - leider hat keine Lösung funktioniert. 
Vielleicht kannst du mir dazu helfen? Vielen Dank

[08.recov.csv](https://github.com/john30/ebusd/files/13208486/08.recov.csv)
[image]

### Actual behavior

Temperaturwerte werden nicht gelesen:
.....
r;w,,BypassDiffTemp,BypassDiffTemp,,,,2a0e,,,temp, ,°C,
.....
r;w,,Desiredtemp,Wunschtemp,,,,140e,,,temp,,°C,
r,,OutsideAirTemp,AussenTemp,,,,0c0e,,,temp,,°C,
r,,ExhaustAirTemp,FortLuft,,,,0d0e,,,temp,,°C,
r,,OutletAirTemp,AbLuft,,,,b40d,,,temp,,°C,
r,,InletAirTemp,ZuLuft,,,,b30d,,,temp,,°C,
.....

### Expected behavior

Ebusd Argumente sollten wie alle anderen angezeigt werden:
[image]

### ebusd version

other

### ebusd arguments

--scanconfig --localhost --accesslevel='*' --enabledefine --device=enh:192.168.1.131:9999 --configpath=/opt/loxberry/data/plugins/ebusdplugin --latency=50000

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 3.0 Ethernet

### Related integration

HTTP

### Logs

2023-10-30 19:13:07.550 [main notice] ebusd 21.3.v21.3 started with auto scan on enhanced device 192.168.1.131:9999
2023-10-30 19:13:07.551 [main error] error reading config files from /opt/loxberry/data/plugins/ebusdplugin: ERR: invalid numeric argument, last error: 08.recov.csv:25: ERR: invalid numeric argument, divisor   in field 0
2023-10-30 19:13:07.576 [bus notice] device status: resetting
2023-10-30 19:13:07.577 [bus notice] bus started with own address 31/36
2023-10-30 19:13:07.620 [bus notice] signal acquired
2023-10-30 19:13:07.850 [bus notice] device status: reset, supports info
2023-10-30 19:13:14.915 [bus notice] new master 03, master count 2
2023-10-30 19:13:14.915 [update notice] sent read recov FanLevelDay QQ=31: 4
2023-10-30 19:13:15.071 [update notice] sent read recov FilterChange QQ=31: 286
2023-10-30 19:13:21.362 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2023-10-30 19:13:21.362 [update notice] store 08 ident: done
2023-10-30 19:13:21.362 [update notice] sent scan-read scan.08  QQ=31: Vaillant;recoV;0225;0203
2023-10-30 19:13:21.362 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2023-10-30 19:13:21.363 [main error] unable to load scan config 08: no file from vaillant with prefix 08 matches ID "recov", SW0225, HW0203
2023-10-30 19:13:21.363 [main error] scan config 08: ERR: element not found
2023-10-30 19:13:21.525 [update notice] sent read recov DayKwh QQ=31: 0.0
2023-10-30 19:13:21.676 [update notice] sent read recov Disbalance QQ=31: 3
2023-10-30 19:13:21.827 [update notice] sent read recov FreezProtect QQ=31: k ein
2023-10-30 19:13:21.978 [update notice] sent read recov FanLevelNight QQ=31: 1
2023-10-30 19:13:22.130 [update notice] sent read recov GroundCollector QQ=31: nein
2023-10-30 19:13:22.285 [update notice] sent read recov InstallationHight QQ=31: 450
2023-10-30 19:13:22.436 [update notice] sent read recov IntensVolumeFlow QQ=31: 130
2023-10-30 19:13:22.600 [update notice] sent read recov MonthKwh QQ=31: 0.0
2023-10-30 19:13:22.755 [update notice] sent read recov maintenance QQ=31: 285
2023-10-30 19:13:22.907 [update notice] sent read recov RedVolumeFlow QQ=31: 70
2023-10-30 19:13:23.011 [bus notice] max. symbols per second: 108
2023-10-30 19:13:23.071 [update notice] sent read recov TotalKwh QQ=31: 8218
2023-10-30 19:13:23.225 [update notice] sent read recov VolumeFlow QQ=31: 160
2023-10-30 19:13:23.376 [update notice] sent read recov WTTyp QQ=31: Standart
2023-10-30 19:13:23.540 [update notice] sent read recov YearKwh QQ=31: 0
2023-10-30 19:13:23.703 [update notice] sent read recov YesterdayKwh QQ=31: 0.0

ojundt, 2023-11-11:
Hi @hermannf1976 , can you try my config from https://github.com/john30/ebusd-configuration/pull/364?

hermannf1976, 2023-11-12:
Hi Oliver,
thanks for your answer. I tried it in the past but didn´t work for my configuration. Do you have an other idea what it´s going wrong? Thanks in advanced. Hermann

john30, 2023-12-17:
as you can see in the log, the csv is not valid, so that needs to be fixed:
08.recov.csv:25: ERR: invalid numeric argument, divisor in field 0
as this is no ebusd issue, closing here. please use [ebusd-configuration](https://github.com/john30/ebusd-configuration) for such

closed 2023-12-17 by john30