john30/ebusd #1068: Problem: Vaillant ecoTEC Plus VC 20CS/1-5 20 & SensoComfort 720/2 + temp outiside + Edbusd V5

haarkop, 2023-11-26:
### Description

Installation in various ways does not work with loading the correct data from the furnace and the controller. Below screenshots: 
Screen from web edbusd: 
[image]

Add-on used for Home Assist
[image]

Configuring the add-on in HA:

scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
http: true
network_device: enh:192.168.1.204:9999
latency: 10
configpath: /config/ebusd-configuration/ebusd-2.1.x/en
Access level: "*"

[image]

File ripped from github to config HA location:
[image]

I've looked through a million sites and tutorials, tried on Docker Synology, etc. and lime. Please help. 
Below what I get after 10 min !!!

I am missing data from the 720 and, for example, gas consumption or kwh, system pressure, etc. 

### Actual behavior

Dane z MGTT:
[image]
[image]

### Expected behavior

Dane z MGTT HA oraz edbusd:

[image]

### ebusd version

23.2

### ebusd arguments

> ebusd --foreground --httpport=8889 --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=PhohSeipaip0oahie0yo0airaquee8eoqu2uo8iegu1eiV0chie4upoh3paiNg8e --scanconfig --mqttjson --configpath=/config/ebusd-configuration/ebusd-2.1.x/en --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.204:9999 --log=all:notice

### Operating system

other

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --httpport=8889 --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=PhohSeipaip0oahie0yo0airaquee8eoqu2uo8iegu1eiV0chie4upoh3paiNg8e --scanconfig --mqttjson --configpath=/config/ebusd-configuration/ebusd-2.1.x/en --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.204:9999 --log=all:notice
2023-11-26 19:31:13.309 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.204:9999, enhanced
2023-11-26 19:31:13.449 [bus notice] device status: resetting
2023-11-26 19:31:13.449 [bus notice] bus started with own address 31/36
2023-11-26 19:31:13.453 [bus notice] signal acquired
2023-11-26 19:31:13.560 [mqtt notice] connection established
2023-11-26 19:31:13.721 [bus notice] device status: reset, supports info
2023-11-26 19:31:13.789 [bus notice] device status: extra info: firmware 1.1[3a0f].1[3a0f], jumpers 0x0f
2023-11-26 19:31:13.836 [bus notice] new master 03, master count 2
2023-11-26 19:31:13.894 [bus notice] new master 10, master count 3
2023-11-26 19:31:13.894 [bus notice] scan 15: ;Vaillant;CTLV2;0514;1104
2023-11-26 19:31:13.894 [update notice] store 15 ident: done
2023-11-26 19:31:13.894 [update notice] received scan-read scan.15  QQ=03: Vaillant;CTLV2;0514;1104
2023-11-26 19:31:18.615 [update notice] received unknown MS cmd: 1008b5110101 / 09393990ffff470000ff
2023-11-26 19:31:18.891 [update notice] received unknown MS cmd: 1008b51009000047ffffff000000 / 0101
2023-11-26 19:31:23.701 [bus notice] scan 08: ;Vaillant;BAI00;1003;7603
2023-11-26 19:31:23.701 [update notice] store 08 ident: done
2023-11-26 19:31:23.701 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;1003;7603
2023-11-26 19:31:23.701 [bus notice] scan 08: ;Vaillant;BAI00;1003;7603
2023-11-26 19:31:23.707 [main notice] read common config file vaillant/general.csv
2023-11-26 19:31:23.708 [main notice] read common config file vaillant/scan.csv
2023-11-26 19:31:23.709 [main notice] read common config file vaillant/broadcast.csv
2023-11-26 19:31:23.710 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW1003, HW7603
2023-11-26 19:31:23.961 [update notice] sent scan-read scan.08 id QQ=31: 
2023-11-26 19:31:24.241 [update notice] sent scan-read scan.08 id QQ=31: 
2023-11-26 19:31:24.521 [update notice] sent scan-read scan.08 id QQ=31: 
2023-11-26 19:31:24.790 [update notice] sent scan-read scan.08 id QQ=31: 21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:24.807 [main notice] found messages: 225 (3 conditional on 28 conditions, 0 poll, 10 update)
2023-11-26 19:31:25.060 [update notice] sent scan-read scan.08 id QQ=31: 21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:25.331 [update notice] sent scan-read scan.08 id QQ=31: 21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:25.601 [update notice] sent scan-read scan.08 id QQ=31: 21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:25.870 [update notice] sent scan-read scan.08 id QQ=31: 21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:25.870 [bus notice] scan 08: ;21;23;15;0010024647;0001;006864;N0
2023-11-26 19:31:28.141 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323231353030
2023-11-26 19:31:28.400 [update notice] sent scan-read scan.15 id QQ=31: 
2023-11-26 19:31:28.660 [update notice] sent scan-read scan.15 id QQ=31: 
2023-11-26 19:31:28.777 [update notice] received read bai Status01 QQ=10: 28.5;28.5;-0.438;-;35.5;off
2023-11-26 19:31:28.970 [update notice] sent scan-read scan.15 id QQ=31: 21;22;15;0020260916;0953;018537;N0
2023-11-26 19:31:28.970 [bus notice] scan 15: ;21;22;15;0020260916;0953;018537;N0
2023-11-26 19:31:28.972 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv2", SW0514, HW1104
2023-11-26 19:31:28.972 [main error] scan config 15: ERR: element not found
2023-11-26 19:31:29.032 [bus notice] max. symbols per second: 104
2023-11-26 19:31:29.210 [update notice] received update-write bai SetMode QQ=10: auto;35.5;-;-;0;0;0;0;0;0
2023-11-26 19:31:30.170 [update notice] sent poll-read bai CounterStartattempts1 QQ=31: 0
2023-11-26 19:31:35.282 [update notice] sent poll-read bai CounterStartattempts2 QQ=31: 0
2023-11-26 19:31:38.605 [update notice] received read bai Status01 QQ=10: 28.5;28.5;-0.438;-;35.5;off
2023-11-26 19:31:38.881 [update notice] received update-write bai SetMode QQ=10: auto;35.5;-;-;0;0;0;0;0;0
2023-11-26 19:31:39.140 [update notice] received poll-read bai Status02 QQ=10: auto;60;75.0;70;70.0

kst84, 2023-11-28:
Cześć @haarkop ,
So I had the same issue as there is no config file for your wall controler 15.ctlv2.csv
[main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv2", SW0514, HW1104

I've just created a symlink to 15.700.csv, it started working but I don't have few items like z1DayTemp which prevents me from controlling the temperature in the room. So please try again and let me know if u have any luck and if you can control temperature. 
Create link as below:
lrwxrwxrwx 1 root root 10 Sep 15 13:27 15.ctlv2.csv -> 15.700.csv

https://github.com/john30/ebusd/issues/1063

haarkop, 2023-11-29:
Something managed to sćiagnąć, please verify :) 
Usunąłem plik 15.700csv i zastąpiłem nową nazwą 15.ctlv2.csv. Unfortunately, there is still no information about the water pressure in the system and from the side of the furnace and the controller

AdaptHeatCurve = { "yesno": {"value": "yes"}}
DisplayedOutsideTemp = { "tempv": {"value": -4.25}}
Hc1ActualFlowTempDesired = { "tempv": {"value": 41.6247}}
Hc1AutoOffMode = { "offmode": {"value": "eco"}}
Hc1FlowTemp = { "tempv": {"value": 21.5}}
Hc1HeatCurve = { "0": {"name": "", "value": 0.6}}
Hc1HeatCurveAdaption = { "0": {"name": "", "value": 0.0}}
Hc1MaxFlowTempDesired = { "tempv": {"value": 70}}
Hc1MinFlowTempDesired = { "tempv": {"value": 15}}
Hc1PumpStatus = { "0": {"name": "", "value": 1}}
Hc1RoomTempSwitchOn = { "rcmode": {"value": "modulating"}}
Hc1Status = { "0": {"name": "", "value": 1}}
Hc1SummerTempLimit = { "tempv": {"value": 18}}
Hc2ActualFlowTempDesired = { "tempv": {"value": 0.0}}
Hc2AutoOffMode = { "offmode": {"value": "eco"}}
Hc2ExcessTemp = { "calibrationv": {"value": 0.0}}
Hc2FlowTemp = { "tempv": {"value": null}}
Hc2HeatCurve = { "0": {"name": "", "value": 0.6}}
Hc2HeatCurveAdaption = { "0": {"name": "", "value": 0.0}}
Hc2MaxFlowTempDesired = { "tempv": {"value": 90}}
Hc2MinFlowTempDesired = { "tempv": {"value": 15}}
Hc2PumpStatus = { "0": {"name": "", "value": 0}}
Hc2RoomTempSwitchOn = { "rcmode": {"value": "off"}}
Hc2Status = { "0": {"name": "", "value": 0}}
Hc2SummerTempLimit = { "tempv": {"value": 21}}
Hc3ActualFlowTempDesired = { "tempv": {"value": 0.0}}
Hc3AutoOffMode = { "offmode": {"value": "eco"}}
Hc3ExcessTemp = { "calibrationv": {"value": 0.0}}
Hc3FlowTemp = { "tempv": {"value": null}}
Hc3HeatCurve = { "0": {"name": "", "value": 1.2}}
Hc3HeatCurveAdaption = { "0": {"name": "", "value": 0.0}}
Hc3MaxFlowTempDesired = { "tempv": {"value": 90}}
Hc3MinFlowTempDesired = { "tempv": {"value": 15}}
Hc3PumpStatus = { "0": {"name": "", "value": 0}}
Hc3RoomTempSwitchOn = { "rcmode": {"value": "off"}}
Hc3Status = { "0": {"name": "", "value": 0}}
Hc3SummerTempLimit = { "tempv": {"value": 21}}
HcStorageTempTop = { "tempv": {"value": null}}
HwcFlowTemp = { "tempv": {"value": 0.0}}
HwcMaxFlowTempDesired = { "tempv": {"value": 80}}
HwcOpMode = { "opmode": {"value": "day"}}
HwcSFMode = { "sfmode": {"value": "auto"}}
HwcStorageTemp = { "tempv": {"value": 35.5}}
HwcStorageTempBottom = { "tempv": {"value": null}}
HwcStorageTempTop = { "tempv": {"value": null}}
HwcTempDesired = { "tempv": {"value": 40}}
OutsideTempAvg = { "tempv": {"value": -1.73047}}
PrEnergySumHc = { "energy4": {"value": 1}}
PrEnergySumHcLastMonth = { "energy4": {"value": 0}}
PrEnergySumHcThisMonth = { "energy4": {"value": 1}}
PrEnergySumHwc = { "energy4": {"value": 0}}
PrEnergySumHwcLastMonth = { "energy4": {"value": 0}}
PrEnergySumHwcThisMonth = { "energy4": {"value": 0}}
SolarYieldTotal = { "energy4": {"value": 0}}
SystemFlowTemp = { "tempv": {"value": null}}
YieldTotal = { "energy4": {"value": 0}}
z1ActualRoomTempDesired = { "tempv": {"value": 20.5}}
z1CoolingTemp = { "tempv": {"value": 24}}
z1HolidayTemp = { "tempv": {"value": 15}}
z1NightTemp = { "tempv": {"value": 21}}
z1OpMode = { "opmode": {"value": "day"}}
z1OpModeCooling = { "opmode": {"value": "auto"}}
z1QuickVetoTemp = { "tempv": {"value": 21.5}}
z1RoomTemp = { "tempv": {"value": 20.475}}
z1SFMode = { "sfmode": {"value": "auto"}}
z1ValveStatus = { "0": {"name": "", "value": 1}}
z2ActualRoomTempDesired = { "tempv": {"value": 0.0}}
z2CoolingTemp = { "tempv": {"value": 24}}
z2HolidayTemp = { "tempv": {"value": 15}}
z2NightTemp = { "tempv": {"value": 15}}
z2OpMode = { "opmode": {"value": "auto"}}
z2OpModeCooling = { "opmode": {"value": "auto"}}
z2RoomTemp = { "tempv": {"value": null}}
z2SFMode = { "sfmode": {"value": "auto"}}
z2ValveStatus = { "0": {"name": "", "value": 0}}
z3ActualRoomTempDesired = { "tempv": {"value": 0.0}}
z3CoolingTemp = { "tempv": {"value": 24}}
z3HolidayTemp = { "tempv": {"value": 15}}
z3NightTemp = { "tempv": {"value": 15}}
z3OpMode = { "opmode": {"value": "auto"}}
z3OpModeCooling = { "opmode": {"value": "auto"}}
z3QuickVetoTemp = { "tempv": {"value": 21}}
z3RoomTemp = { "tempv": {"value": null}}
z3SFMode = { "sfmode": {"value": "auto"}}
z3ValveStatus = { "0": {"name": "", "value": 0}}
Hc1ExcessTemp = { "calibrationv": {"value": 0.0}}
HcStorageTempBottom = { "tempv": {"value": null}}
z2QuickVetoTemp = { "tempv": {"value": 21}}

kst84, 2023-11-29:
Yeah it reads the data now from your controler.

It doesn't show water pressure for me either, maybe @john30 will be kind enough to put some light on it.  However If I manualy request ebusd to fetch me the water pressure - it works. The same goes with flame status in bai circut.

Try 
```
[krzysiek@mini ~]$ echo "r  -c ctlv2 WaterPressure" | nc -q 1 127.0.0.1 8888
1.9

[krzysiek@mini ~]$ echo "r  -c bai Flame" | nc -q 1 127.0.0.1 8888
on

[krzysiek@mini ~]$
```

Koky05, 2023-11-30:
I think the flame and WaterPressure are not in filter for mqtt-hassio.cfg, Try add values into filter line (108):
`filter-name = status|temp|yield|fuel|count|energy|power|runtime|hours|starts|mode|curve|load|party|cooling|heat|sensor|timer|pressure|flame`

haarkop, 2023-11-30:
please check if I did it right , my config 

scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
http: true
network_device: ens:192.168.1.204:9999
latency: 10
Access level: "*"
configpath: /config/ebusd-configuration/ebusd-2.1.x

below structure on HA drive, not sure if I have the mqtt-hassio.cfdg file in the right place

[image]

Koky05, 2023-11-30:
According your configuration I will try to change it (to same path as config):
mqttint: /config/ebusd-configuration/mqtt-hassio.cfg

haarkop, 2023-11-30:
[image]

After changing something like this, if I leave the default it is ok

Koky05, 2023-11-30:
You can look into MQTT what date it returns.

Fastes way is via MQTT Explorer:
[MQTT Explorer as Addon](https://github.com/adamoutler/mqtt-explorer)

haarkop, 2023-12-13:
I checked and unfortunately nothing helped, I'm thinking about buying an original Vaillant and ebusd to the trash or sell, it's a pity that the author of the device did not provide information that it does not work 100% with new stoves

Koky05, 2023-12-13:
@haarkop can we connect directly? (WhatsApp, Messenger, TeamViewer, ....)
It looks like you use incorrect configuration file and problem could be in way you download it.

k-fil, 2023-12-14:
> @haarkop can we connect directly? (WhatsApp, Messenger, TeamViewer, ....) It looks like you use incorrect configuration file and problem could be in way you download it.

Hello Koky05 - i'm having similar issues but not using HA when I try to read energy data fom my VC20CS i'm getting error invalid position in decode. I think its something about missconfig I would appreciate if You would finde 15 minutes to validate my debian/ebusd/ebusctl setup remotely.

Koky05, 2023-12-14:
@k-fil sorry for misunderstanding, I could help with MQTT or file encoding problems. But I don't know how to change configuration filed to match Vaillant data.  I'm on same boat as you I have couple of messages with invalid position in decode.
And if I run `ebusd --checkconfig` I got no errors.

john30, 2023-12-17:
not an ebusd issue but missing config, so closing this one. please use [ebusd-configuration](https://github.com/john30/ebusd-configuration) for such

closed 2023-12-17 by john30

pvyleta, 2024-03-22:
Hmmm, I am kind of convinced  that the `Malformed UTF-8` is not a consequence of missing config, but rather "malformed" 'mqtt-hassio.cfg. In my case, I received this error when my `mqtt-hassio.cfg` had `CRLF` line ends compared to the original `LF` line ends. This can easily happen when copypasting on a Windows machine.

Pantalooney, 2024-12-03:
> Hmmm, I am kind of convinced that the `Malformed UTF-8` is not a consequence of missing config, but rather "malformed" 'mqtt-hassio.cfg. In my case, I received this error when my `mqtt-hassio.cfg` [...]

@pvyleta You're a hero! I've had this problem about a year ago and spent hours resolving it, but couldn't get it to work. Now found your resolution, tried it, and it works! Before I had just copy-pasted the code of mqtt-integration.cfg and mqtt-hassio.cfg from github into the Home Assistant File Editor. Now I have instead downloaded the files and uploaded it via the File Editor into Home Assistant, the "Malformed UTF-8" message is gone and I am finally receiving values from the sensors! 

Many thanks for that!