john30/ebusd #1074: knx, docker: negative temperatures are not sent correctly to knx bus

mivola, 2023-12-01:
### Description

- the value of a temparature (e.g. OutsideTemp or `broadcast/datetime/outsidetemp`) on the knx bus differs from the value sent via MQTT or fetched via ebusctl if the value is negative
- see:
```
ebusctl read -c ehp OutsideTemp
-1.62;ok
```
[image]

- but on the knx bus the value is different (by factor 10?)
-- see on the wiregate log:
[image]
2023-11-29 09:42:58.702,A_GroupValue_Write,15.15.22,9/2/10,80 D4,**-18.36**,,9,0,low,5,T_DATA_XXX_REQ,0
-- also in NodeRed:
[image]

- the incorrect value is also nicely visibly in a graph when compared to other (not ebusd based) sensors:
[image]

- the ebusd is started in docker with these env vars:
```
-e EBUSD_SCANCONFIG= \
-e EBUSD_DEVICE=/dev/ttyUSB0 \
-e EBUSD_VERSION=23.2 \
-e EBUSD_MQTTPORT=1883 \
-e EBUSD_MQTTHOST=mediaserver \
-e EBUSD_MQTTVAR="filter-direction=r|u|^w" \
-e EBUSD_MQTTINT=/etc/ebusd/mqtt-hassio.cfg \
-e EBUSD_MQTTJSON= \
-e EBUSD_KNXURL= \
-e EBUSD_KNXINT=/etc/ebusd/knx.cfg \
-e EBUSD_LOG=other:debug \

```

### Actual behavior

- incorrect temparature values when the temperature is below zero

### Expected behavior

- the value should be correct even if negative

### ebusd version

23.2

### ebusd arguments

see description

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat), MQTT generic, KNX

### Logs

I couldnt find any log that shows the incorrect value that is sent out to the KNX bus

c-huetter, 2023-12-04:
What a relief, I have exactly the same issue :-)

---
# Description
Negative temperature values are sent incorrectly to KNX. Positive temperature values are sent correctly.
# Actual behavior
ebusd sends telegrams to knxd with incorrect values for negative temperatures:
```
time: 2023-12-04T09:02:20Z
value: -19.78
description: OutsideAirTemperature
dpt: 9.001
dptdesc: Temperature
event: GroupValue_Write
```
# Expected behavior
The actual air temperature was -0.7 °C
```
# ebusctl read -V outsideairtemperature                        
kwl OutsideAirTemperature data=-0.7 °C [OutsideTemperature]
```
# ebusd version
`23.2`
# ebusd arguments
`-c /etc/ebusd -d ens:/dev/ttyUSB0 --knxurl=ip:localhost --knxint=/etc/ebusd/knx.cfg`
# Operating system
Synology DSM Linux 4.4.302+
# CPU architecture
`arm64`
# Dockerized
same as ebusd version
# Hardware interface
adapter 3.0 USB
# Related integration
KNX
# Logs
`2023-12-04 09:02:14.139 [update notice] sent poll-read kwl OutsideAirTemperature QQ=31: -0.7`

c-huetter, 2023-12-11:
Hi everyone,

I have analyzed the issue reported by @mivola that negative temperatures are not sent correctly to KNX. The functions `floatToInt16` and `int16ToFloat` that translate 2-byte floats into KNX data type 9 (and vice versa) seem to be buggy. 

I have [fixed the bugs in the translation functions](https://github.com/john30/ebusd/commit/86f0e46a6569f8eb2fc61d03bf44ad8a41955e5c) and [added a unit test for knxhandler](https://github.com/john30/ebusd/commit/ded63f660572e1e0d078f64a1580caa35e2734ef) in my fork [ebusd-with-knxd](https://github.com/c-huetter/ebusd-with-knxd). Please verify the bugfix and consider merging my pull request #1051.

Best,
Christian

john30, 2023-12-17:
I can't reproduce this. the values are converted correctly, only in edge cases there might be a rounding issue, but this is not related to this.
please post the hex value, the field definition and what was sent to knx for a wrong example

c-huetter, 2023-12-17:
I've used the following test values from the repository [knx-datapoints](https://github.com/Rafelder/knx-datapoints/blob/a08c388616c9e9a8007d0016b0eac08c5b0b47cc/tests/dpt009.spec.js#L6):

| Hex | Float |
|--------|--------|
| 0x0000 | 0.0 | 
| 0x07FF | 20.47 | 
| 0x6464 | 46039.04 | 
| 0x7FFF | 670760.96 | 
| 0x87FF | -0.01 | 
| 0x8000 | -20.48 | 
| 0x8A24 | -30.0 | 
| 0xAC00 | -327.68 | 
| 0xC8C8 | -9461.76 | 
| 0xF800 | -671088.64 |  

Please check out the test cases in [test_knxhandler.cpp](https://github.com/c-huetter/ebusd-with-knxd/blob/master/src/lib/knx/test/test_knxhandler.cpp) to reproduce the bug. It would be easier to test the functions floatToInt16 and int16ToFloat if they were static.

john30, 2023-12-23:
what are the values that fail from this list with the current implementation?

c-huetter, 2023-12-23:
The test case [test_knxhandler.cpp](https://github.com/c-huetter/ebusd-with-knxd/blob/master/src/lib/knx/test/test_knxhandler.cpp) returns the following values when using the original (probably buggy) functions [floatToInt16](https://github.com/john30/ebusd/blob/53fcc90f6e1ff1e419da7603a22e27a6c68414c8/src/ebusd/knxhandler.cpp#L320) and [int16ToFloat](https://github.com/john30/ebusd/blob/53fcc90f6e1ff1e419da7603a22e27a6c68414c8/src/ebusd/knxhandler.cpp#L343C13-L343C13):
```
Correct: 0x0 == 0.00
Correct: 0x7ff == 20.47
Correct: 0x6464 == 46039.04
Incorrect: 0x7fff != 4294967296.00, should be 670760.94
Incorrect: 0x87ff != -20.47, should be -0.01
Incorrect: 0x8000 != -0.00, should be -20.48
Incorrect: 0x8a24 != -10.96, should be -30.00
Correct: 0xac00 == -327.68
Incorrect: 0xc8c8 != -1024.00, should be -9461.76
Incorrect: 0xf800 != -0.00, should be -671088.62
Correct: 0.00 == 0x0
Incorrect: 20.47 != 0x7fe, should be 0x7ff
Correct: 46039.04 == 0x6464
Incorrect: 670760.94 != 0x7ffe, should be 0x7fff
Incorrect: -0.01 != 0x8001, should be 0x87ff
Incorrect: -20.48 != 0x8c00, should be 0x8000
Incorrect: -30.00 != 0x8ddc, should be 0x8a24
Incorrect: -9461.76 != 0xcf38, should be 0xc8c8
Incorrect: -671088.62 != 0x7fff, should be 0xf800
```

commit 2b5bd1b6 (2023-12-24): fix dpt 9 with negative values (closes #1074) and add rounding
closed 2023-12-24 by john30