john30/ebusd #1078: Adapter v5: Loose of Wifi connection after few hours

Wolla57, 2023-12-03:
### Description

Adapter v5, Firmware build: 20231015, WiFi connection 76% (-62dBm), host_task_tcp: accepted, ebusd 21.2.v21.2 running on Raspberry Pi 3 with Raspbian 8 (Jessie): 
- At the beginning the connection is running with temporary "ERR: SYN received" or "ERR: wrong symbol received" or "device status: eBUS comm error: overrun". Values are read by FHEM-get via ECMD/telnet over ebusd mostly successful
- More than 4 hours later "signal lost", "device invalid", "unable to open 192.168.178.2:9999: ERR: generic I/O error", "send to 15: ERR: no signal, give up", "send message part 0: ERR: no signal". From this time the adapter can't be reached by ping. The green LED is on permanently, the blue LED is on only for a short moment. 
- After pressing button at adapter v5: no success
- After Power off/on of adapter: no success
- After reboot of Raspberry: no success
- After reboot of Router/Wifi: no success
- The only way found to solve this hangup: re-flash of the firmware
- Test with firmware 20230917: same problem

### Actual behavior

see description

Hint: Might it be a problem/overrun of an internal counter of the firmware ?

### Expected behavior

mostly errorfree connection for longtime runs. In case of error/hangup automatical restart

### ebusd version

21.2

### ebusd arguments

EBUSD_OPTS="--scanconfig  -c /etc/ebusd  -d enh:192.168.178.2:9999  --latency=200 -p 8888  -l /var/log/ebusd.log"

### Operating system

other

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

adapter 5 via USB

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2023-12-03 09:27:01.402 [main notice] ebusd 21.2.v21.2 started with auto scan on enhanced device 192.168.178.2:9999
2023-12-03 09:27:01.484 [bus notice] bus started with own address 31/36
2023-12-03 09:27:01.508 [bus notice] signal acquired
2023-12-03 09:27:01.945 [bus notice] device status: reset
2023-12-03 09:27:06.387 [bus notice] new master 10, master count 2
2023-12-03 09:27:06.462 [bus notice] new master 03, master count 3
2023-12-03 09:27:06.462 [update notice] received unknown MS cmd: 1008b5110101 / 0966ff900000740100ff
2023-12-03 09:27:06.752 [update notice] received unknown MS cmd: 1008b5100900005f7affff000000 / 0101
2023-12-03 09:27:11.585 [main error] scan config 08: ERR: SYN received
2023-12-03 09:27:11.949 [bus error] device status: eBUS comm error: overrun
2023-12-03 09:27:13.683 [main error] scan config 15: ERR: SYN received
2023-12-03 09:27:16.528 [update notice] received unknown MS cmd: 1008b5110101 / 0966ff900000740100ff
2023-12-03 09:27:16.810 [update notice] received unknown MS cmd: 1008b5100900005f7affff000000 / 0101
2023-12-03 09:27:17.092 [update notice] received unknown MS cmd: 1008b5040100 / 0a03582909031207239000
2023-12-03 09:27:17.351 [update notice] received unknown MS cmd: 1008b5110102 / 05033c6e507a
2023-12-03 09:27:17.591 [update notice] received update-read broadcast vdatetime QQ=10: 09:29:56;03.12.2023
2023-12-03 09:27:17.865 [update notice] received unknown MS cmd: 1008b5110100 / 0838031002040f0081
2023-12-03 09:27:18.066 [update notice] received update-read broadcast outsidetemp QQ=10: 0.562
[... 499 lines cut ...]
2023-12-03 15:13:55.633 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:13:55.633 [bus notice] device invalid
2023-12-03 15:13:55.633 [bus error] send to 08: ERR: no signal, give up
2023-12-03 15:13:55.633 [bus error] send message part 0: ERR: no signal
2023-12-03 15:16:07.953 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:16:07.953 [bus notice] device invalid
2023-12-03 15:16:07.953 [bus error] send to 15: ERR: no signal, give up
2023-12-03 15:16:07.953 [bus error] send message part 0: ERR: no signal
2023-12-03 15:18:20.273 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:18:20.273 [bus notice] device invalid
2023-12-03 15:18:20.273 [bus error] send to 08: ERR: no signal, give up
2023-12-03 15:18:20.273 [bus error] send message part 0: ERR: no signal
2023-12-03 15:20:32.593 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:20:32.593 [bus notice] device invalid
2023-12-03 15:20:32.593 [bus error] send to 15: ERR: no signal, give up
2023-12-03 15:20:32.593 [bus error] send message part 0: ERR: no signal
2023-12-03 15:22:44.913 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:22:44.913 [bus notice] device invalid
2023-12-03 15:22:44.913 [bus error] send to 08: ERR: no signal, give up
2023-12-03 15:22:44.913 [bus error] send message part 0: ERR: no signal
2023-12-03 15:24:57.233 [bus error] unable to open 192.168.178.2:9999: ERR: generic I/O error
2023-12-03 15:24:57.233 [bus notice] device invalid
2023-12-03 15:24:57.233 [bus error] send to 15: ERR: no signal, give up
2023-12-03 15:24:57.233 [bus error] send message part 0: ERR: no signal
2023-12-03 15:26:07.734 [main notice] SIGTERM received

Wolla57, 2023-12-06:
Same problem after few days run. But the adapter is still reachable by ping. So log was found as follow (Note: No buttom was pressed at the adapter) :

Build: 20231015
ebusd device string: ens:192.168.178.2:9999
WiFi station/client: WLAN_2, 192.168.178.2, 76% (-62dBm)
WiFi access point: inactive
15:10:58	Info	execute: ebus
15:10:58	Info	init_ebus: switching to enhanced eBUS mode on TCP port (with bitrate deviation 0 and enhanced arbitration delay 200 us)
15:10:58	Info	setup_uart: eBUS UART 1 on TX=10(inv) RX=7(inv) with 2400 Bd
15:10:58	Info	uart: ALREADY NULL
15:10:58	Info	host_task_tcp: listening
15:10:58	Info	init_ebus: timer started
15:10:58	Info	execute: led 0 50
15:10:58	Info	execute: exit
15:10:58	Info	execute: command exit
15:10:58	Info	execute: command exit
15:10:58	Info	replExecuteFromFile: successfully executed 41/41 commands from '/data/main.eas'
15:10:58		Welcome to eBUS Adapter Shield Interface! Type 'help' to get the list of commands.
15:10:58		Your terminal application does not support escape sequences.
15:10:58	Info	main_task: Returned from app_main()
15:10:58	Info	statusLedTimerFn: duty set to 50
15:11:00	Info	cmd_wifi: sta disconnect, reconnect...
15:11:00	Info	repl_event_task: wifi event: 0
15:11:00	Info	execute: led 0 0
15:11:00	Info	statusLedTimerFn: duty set to 0
15:11:02	Info	cmd_wifi: sta connect, set fix ip
15:11:02	Info	repl_event_task: wifi event: 1
15:11:02	Info	execute: led 0 50
15:11:02	Info	execute: start
15:11:02	Info	execute: server
15:11:02	Info	esp-rest: Starting HTTP Server
15:11:02		server started
15:11:02	Info	execute: led 0 10
15:11:02	Info	statusLedTimerFn: duty set to 10
08:13:35	Info	host_task_tcp: accepted: 192.168.178.13
07:39:09	Info	statusLedTimerFn: btn short pressed 0
07:39:09	Info	execute: led 2 50 0
07:39:09		2 duty set to 50
07:39:09		2 fade started
07:39:14	Info	statusLedTimerFn: btn short pressed 1
07:39:16	Info	statusLedTimerFn: btn long pressed 1
11:59:42	Info	statusLedTimerFn: btn short pressed 0
11:59:42	Info	execute: led 2 50 0
11:59:42		2 duty set to 50
11:59:42		2 fade started
11:59:47	Info	statusLedTimerFn: btn short pressed 1
11:59:49	Info	statusLedTimerFn: btn long pressed 1
12:15:38	Info	statusLedTimerFn: btn short pressed 0
12:15:38	Info	execute: led 2 50 0
12:15:38		2 duty set to 50
12:15:38		2 fade started
12:15:43	Info	statusLedTimerFn: btn short pressed 1
12:15:45	Info	statusLedTimerFn: btn long pressed 1
17:00:43	Info	statusLedTimerFn: btn short pressed 0
17:00:43	Info	execute: led 2 50 0
17:00:43		2 duty set to 50
17:00:43		2 fade started
17:00:43	Info	statusLedTimerFn: btn short pressed 1
17:00:45	Info	statusLedTimerFn: btn long pressed 1
07:10:46	Info	statusLedTimerFn: btn short pressed 0
07:10:46	Info	execute: led 2 50 0
07:10:46		2 duty set to 50
07:10:46		2 fade started
07:10:51	Info	statusLedTimerFn: btn short pressed 1
07:10:53	Info	statusLedTimerFn: btn long pressed 1
13:11:56	Info	statusLedTimerFn: btn short pressed 0
13:11:56	Info	execute: led 2 50 0
13:11:56		2 duty set to 50
13:11:56		2 fade started
13:12:01	Info	statusLedTimerFn: btn short pressed 1
13:12:03	Info	statusLedTimerFn: btn long pressed 1
20:17:06	Error	ebus_tcp_server_loop: recv: errno 104
20:17:06	Error	ebus_tcp_server_loop: recv: errno 128
20:17:06	Info	host_task_tcp: closed: 192.168.178.13
20:17:06	Info	host_task_tcp: listening
20:17:06	Info	host_task_tcp: accepted: 192.168.178.13
20:26:20	Info	statusLedTimerFn: btn short pressed 0
20:26:20	Info	execute: led 2 50 0
20:26:20		2 duty set to 50
20:26:20		2 fade started
20:26:39	Info	statusLedTimerFn: btn short pressed 1
20:26:46	Info	statusLedTimerFn: btn long pressed 1
20:27:02	Info	statusLedTimerFn: btn short pressed 0
20:27:02	Info	execute: led 2 50 0
20:27:02		2 duty set to 50
20:27:02		2 fade started
20:27:08	Info	statusLedTimerFn: btn short pressed 1
20:27:10	Info	statusLedTimerFn: btn long pressed 1
20:27:12	Info	statusLedTimerFn: btn short pressed 0
20:27:12	Info	execute: led 2 50 0
20:27:12		2 duty set to 50
20:27:12		2 fade started
20:27:30	Info	statusLedTimerFn: btn short pressed 1
20:27:32	Info	statusLedTimerFn: btn long pressed 1
20:29:05	Info	statusLedTimerFn: btn short pressed 0
20:29:05	Info	execute: led 2 50 0
20:29:05		2 duty set to 50
20:29:05		2 fade started
20:29:27	Info	statusLedTimerFn: btn short pressed 1
20:29:29	Info	statusLedTimerFn: btn long pressed 1
20:33:26	Info	statusLedTimerFn: btn short pressed 0
20:33:26	Info	execute: led 2 50 0
20:33:26		2 duty set to 50
20:33:26		2 fade started
20:33:32	Info	statusLedTimerFn: btn short pressed 1
20:34:57	Info	statusLedTimerFn: btn long pressed 1
20:50:40	Info	statusLedTimerFn: btn short pressed 0
20:50:40	Info	execute: led 2 50 0
20:50:40		2 duty set to 50
20:50:40		2 fade started
20:50:46	Info	statusLedTimerFn: btn short pressed 1
20:51:04	Info	statusLedTimerFn: btn long pressed 1
20:54:14	Info	statusLedTimerFn: btn short pressed 0
20:54:14	Info	execute: led 2 50 0
20:54:14		2 duty set to 50
20:54:14		2 fade started
20:54:25	Info	statusLedTimerFn: btn short pressed 1
20:54:27	Info	statusLedTimerFn: btn long pressed 1
21:08:22	Info	statusLedTimerFn: btn short pressed 0
21:08:22	Info	execute: led 2 50 0
21:08:22		2 duty set to 50
21:08:22		2 fade started
21:08:28	Info	statusLedTimerFn: btn short pressed 1
21:08:30	Info	statusLedTimerFn: btn long pressed 1
21:48:13	Info	statusLedTimerFn: btn short pressed 0
21:48:13	Info	execute: led 2 50 0
21:48:13		2 duty set to 50
21:48:13		2 fade started
21:48:26	Info	statusLedTimerFn: btn short pressed 1
21:48:28	Info	statusLedTimerFn: btn long pressed 1
22:06:00	Warning	ebus_tcp_server_loop: closed
22:06:00	Info	host_task_tcp: closed: 192.168.178.13
22:06:00	Info	host_task_tcp: listening
22:06:00	Info	host_task_tcp: accepted: 192.168.178.13

john30, 2023-12-17:
not an ebusd issue, so closing this one. please use the appropriate repo as communicated: [ebusd-esp32](https://github.com/john30/ebusd-esp32)

closed 2023-12-17 by john30