john30/ebusd #1081: Adapter v5: W5500 Ethernet adapter not recognized

olli-r, 2023-12-06:
### Description

The proposed ethernet adapter W5500 is not detected by Ebus Adapter V5.

### Actual behavior

Plugged in the module flipped over as described. Ethernet port shows link and other LED is flashing when network traffic is recieved/transmitted.
Ethernet module not loaded during startup. See attached logfile.

### Expected behavior

Ethernet should be loaded.

### ebusd version

23.2

### ebusd arguments

(not revelant)

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 5 via Ethernet

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

17:34:49Info	execute: isio -l 9
17:34:49		IO 9 level 1 is unmet
17:34:49	Info	execute: io -iu 6
17:34:49		IO 6 direction set to 01
17:34:49		IO 6 pull set to 00
17:34:49		IO 6 = 1
17:34:49	Info	execute: var rpi 0
17:34:49	Info	execute: if "isio -l 6" "var rpi 1"
17:34:49	Info	execute: isio -l 6
17:34:49		IO 6 level 1 is unmet
17:34:49	Info	execute: var isnew 1
17:34:49	Info	execute: if "§isnvs ebus.host" "var isnew 0"
17:34:49	Info	execute: isnvs ebus.host
17:34:49	Info	execute: var isnew 0
17:34:49	Info	execute: if "§isnvs sta.on" "var isnew 0"
17:34:49	Info	execute: isnvs sta.on
17:34:49	Info	execute: var isnew 0
17:34:49	Info	execute: if "§isnvs eth.on" "var isnew 0"
17:34:49	Info	execute: isnvs eth.on
17:34:49	Warning	execute: command failed: isnvs eth.on = 1
17:34:49	Info	execute: on -lo iol 9 "led 0 255" "led 1 255" "led 2 255" "sleep 500" restart
17:34:49		io trigger installed
17:34:49	Info	execute: on iol 9 "led 2 50 0"
17:34:49		io trigger installed
17:34:49	Info	execute: if "var skipstart" "led 0 255 0 5000" "led 2 50" "ap EBUS" server exit
17:34:49	Info	execute: var skipstart
17:34:49		variable skipstart = 0
17:34:49	Info	execute: function start server "led 0 10"
17:34:49		function installed
17:34:49	Info	execute: on wifi 1 start
17:34:49		network trigger installed
17:34:49	Info	execute: on eth 1 start
17:34:49		network trigger installed
17:34:49	Info	execute: if "isnvs sta.on" "led 0 10 255 5000" "sleep 100" sta
17:34:49	Info	execute: isnvs sta.on
17:34:49	Info	execute: led 0 10 255 5000
17:34:49	Info	execute: sleep 100
17:34:49	Info	statusLedTimerFn: duty set to 10
17:34:49	Info	execute: sta
17:34:49	Info	pp: pp rom version: 9387209
17:34:49	Info	net80211: net80211 rom version: 9387209
17:34:49	Info	wifi_init: tcpip mbox: 32
17:34:49	Info	wifi_init: udp mbox: 6
17:34:49	Info	wifi_init: tcp mbox: 6
17:34:49	Info	wifi_init: tcp tx win: 5744
17:34:49	Info	wifi_init: tcp rx win: 5744
17:34:49	Info	wifi_init: tcp mss: 1440
17:34:49	Info	wifi_init: WiFi IRAM OP enabled
17:34:49	Info	wifi_init: WiFi RX IRAM OP enabled
17:34:49	Info	wifi_init: LWIP IRAM OP enabled
17:34:49	Info	phy_init: phy_version 1130,b4e4b80,Sep 5 2023,11:09:30
17:34:49		sta connecting to 'IoT'
17:34:52	Info	cmd_wifi: sta connect
17:34:53	Info	esp_netif_handlers: sta ip: 192.168.42.143, mask: 255.255.255.0, gw: 192.168.42.1
17:34:53	Info	cmd_wifi: sta got ip
17:34:53	Info	repl_event_task: wifi event: 1
17:34:53	Info	execute: led 0 50
17:34:53	Info	execute: start
17:34:53	Info	execute: server
17:34:53	Info	esp-rest: Starting HTTP Server
17:34:53		server started
17:34:53	Info	execute: led 0 10
17:34:53	Info	execute: if "var isnew" "ap EBUS" else "if \"isnvs ap.on\" ap"
17:34:53	Info	execute: var isnew
17:34:53		variable isnew = 0
17:34:53	Info	execute: if "isnvs ap.on" ap
17:34:53	Info	execute: isnvs ap.on
17:34:53	Info	execute: var eth 0
17:34:53	Info	execute: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""
17:34:53	Info	execute: eth -t
17:34:53	Error	w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
17:34:53	Error	w5500.mac: emac_w5500_init(795): vefiry chip ID failed
17:34:53	Error	esp_eth: esp_eth_driver_install(228): init mac failed
17:34:53	Error	eth: initialise_ethernet(250): install driver
17:34:53		Ethernet on USR-ES1 not available
17:34:53	Warning	execute: command failed: eth -t = 1
17:34:53	Info	execute: if "eth -wt" "var eth 2"
17:34:53	Info	execute: eth -wt
17:34:53	Error	w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
17:34:53	Error	w5500.mac: emac_w5500_init(795): vefiry chip ID failed
17:34:53	Error	esp_eth: esp_eth_driver_install(228): init mac failed
17:34:53	Error	eth: initialise_ethernet(250): install driver
17:34:53		Ethernet on W5500 module not available
17:34:53	Warning	execute: command failed: eth -wt = 1

olli-r, 2023-12-06:
Maybe this post leads to a hint:

[https://github.com/espressif/esp-idf/issues/11331](https://github.com/espressif/esp-idf/issues/11331)

john30, 2023-12-17:
not an ebusd issue, so closing this one. please use the appropriate repo as communicated: [ebusd-esp32](https://github.com/john30/ebusd-esp32)

closed 2023-12-17 by john30