john30/ebusd #1082: mqtt-integration filter-non-field ignored

hoedlmoser, 2023-12-09:
### Description

starting ebusd installed from debian repo on debian 12 bookworm with following parameters:
`--foreground --localhost --scanconfig=full --mqtthost=mqtt --mqttport=1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --log=all:info`

the mqtt-integration file /etc/ebusd/mqtt-hassio.cfg is untouched, no changes applied.

in mqtt-integration
`filter-non-field = ^sensor$`
is defined but seems to be ignored, temperature output in mqtt still also has this sensor field.

### Actual behavior

field sensor is in mqtt output
```
ebusd/ehp/SourceTempOutput/get (null)
ebusd/ehp/SourceTempOutput {
     "temp": {"value": 14.00},
     "sensor": {"value": "ok"}}
```

### Expected behavior

field sensor should be filtered from mqtt output
```
ebusd/ehp/SourceTempOutput/get (null)
ebusd/ehp/SourceTempOutput {
     "temp": {"value": 14.00},
```

### ebusd version

23.2

### ebusd arguments

--foreground --localhost --scanconfig=full --mqtthost=mqtt --mqttport=1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --log=all:info

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT generic

### Logs

```
2023-12-09 10:29:11.408 [mqtt info] received get topic for ehp SourceTempOutput
2023-12-09 10:29:11.408 [bus info] send message: 3108b509030d0800
2023-12-09 10:29:11.549 [update info] sent MS cmd: 3108b509030d0800 / 03e50000
2023-12-09 10:29:11.550 [update notice] sent poll-read ehp SourceTempOutput QQ=31: 14.31;ok
2023-12-09 10:29:11.550 [mqtt notice] read ehp SourceTempOutput: 

```

john30, 2023-12-10:
the filter is only for what is sent to HA discovery, not what is sent to MQTT as data in the first place

closed 2023-12-10 by john30

hoedlmoser, 2023-12-10:
sorry for the noise, I totally misunderstood "_Such a configuration file allows flexible **construction** of MQTT topics and payloads depending on message and field definitions as well as global status topics._" in https://github.com/john30/ebusd/wiki/MQTT-integration.