john30/ebusd #1087: Easy way to confirm what full path dir config is being loaded with`ebusctl info`

tcwilkinson, 2023-12-12:
### Description

Think this would be a useful added info when testing configurations; apologies if it already exists somewhere and I just haven't found it!

Background:- 

Am trying to run ebusd via a docker container, and testing different configs. So far it seems difficult to be sure whether ebusd is continuing to use the default config files, or actually loading the manually pointed to volume + config directories. OK, a workaround would be I could manually play with an example config and introduce some obvious falsely named parameter, but that seems a little risky.

Proposal:- 

Simple solution would be if `ebusctl info` would output the full path/web address of the current config dir, e.g. on a line after update: and before the addresses of masters and slaves.

For example:-
```
version: ebusd 23.2.23.2
device: /dev/ttyAMA0, enhanced high speed, firmware 1.1[3806].1[3806]
signal: acquired
symbol rate: 80
max symbol rate: 160
...
update: 10
configdir: /etc/ebusd/
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0901;HW=5103", loaded "vaillant/08.hmu.csv"
...
```

john30, 2023-12-17:
it is part of the start parameters, so whats the benefit exactly? don't you know them?

Tomasonx, 2024-01-04:
@tcwilkinson    
you mean: 
  **loaded "vaillant/08.hmu.csv"** 

you want to see something like this, the full path:
  **loaded "/etc/ebusd/config/ebusd-configuration/latest/en/vaillant/08.hmu.csv"**

Even if this parameter is set manually, it is important to see if it was taken into account as you expected.

tcwilkinson, 2024-01-22:
@Tomasonx yes, something like could work well.   Or to save on repetition, just show the base folder in the output:
`config loaded from "/etc/ebusd/config/ebusd-configuration/latest/en/"` 

@john30  It's very useful to be sure the location of the file that is actually being pulled is the one you intended.

- For example:- a specific if you are running ebusd in docker and have made a typo in passing the name of the config variable (EBUSD_CONFIG). In this case, ebusd will silently adopt the default config, and you'd be none the wiser why certain custom configs were not being properly loaded.

(Speaking here from experience of being unable to track down the source of my problem for days...! :-) )

commit 2ea20ed8 (2024-01-27): add config path to verbose info command (closes #1087)
closed 2024-01-27 by john30