john30/ebusd #1091: Device '/dev/ttyAMA0' does not exist in eBUSd (2ad9b828_ebusd).

1achy, 2023-12-17:
### Description

After update I have this error 👍 

Device '/dev/ttyAMA0' does not exist in eBUSd (2ad9b828_ebusd).

### Actual behavior

don't start up 

### Expected behavior

Device '/dev/ttyAMA0' does not exist in eBUSd (2ad9b828_ebusd).

### ebusd version

current source from git

### ebusd arguments

192.168.0.54:3333

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

adapter 2

### Logs

Device '/dev/ttyAMA0' does not exist in eBUSd (2ad9b828_ebusd).

1achy, 2023-12-17:
in other way if I change config I can't save . I have this error:

Impossibile salvare la configurazione del componente aggiuntivo, Device '/dev/ttyAMA0' does not exist in eBUSd (2ad9b828_ebusd). Got {'scanconfig': True, 'loglevel_all': 'notice', 'mqtttopic': 'ebusd', 'mqttint': '/etc/ebusd/mqtt-hassio.cfg', 'mqttjson': True, 'network_device': '192.168.0.54:3333', 'configpath': '/config/www/ebusd_mod_ctlv2/', 'device': '/dev/ttyAMA0', 'latency': 10, 'mode': 'enh'}

1achy, 2023-12-17:
config: 

scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
network_device: 192.168.0.54:3333
configpath: /config/www/ebusd_mod_ctlv2/
device: /dev/ttyAMA0
latency: 10

it is work 2 years

john30, 2023-12-17:
no idea what this has to do with ebusd in the first place. if the device is missing, then certainly something goes quite wrong on the host system. so: check the syslog, check the devices, check dmesg

closed 2023-12-17 by john30

1achy, 2023-12-23:
hi, the device is connect network with ping it response 👍 

PING 192.168.0.54 (192.168.0.54): 56 data bytes
64 bytes from 192.168.0.54: icmp_seq=0 ttl=255 time=106.311 ms
64 bytes from 192.168.0.54: icmp_seq=1 ttl=255 time=46.641 ms
64 bytes from 192.168.0.54: icmp_seq=2 ttl=255 time=65.268 ms
64 bytes from 192.168.0.54: icmp_seq=3 ttl=255 time=166.363 ms
64 bytes from 192.168.0.54: icmp_seq=4 ttl=255 time=9.892 ms
64 bytes from 192.168.0.54: icmp_seq=5 ttl=255 time=26.092 ms
64 bytes from 192.168.0.54: icmp_seq=6 ttl=255 time=8.618 ms
^C
--- 192.168.0.54 ping statistics ---
7 packets transmitted, 7 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 8.618/61.312/166.363/53.389 ms
achy@Achys-MacBook-Pro ~ % ~~~~
achy@Achys-MacBook-Pro ~ % 

and it he work before update

1achy, 2023-12-23:
Error:

[image]

[image]

Config 👍

1achy, 2023-12-23:
SOLVE:

disinstalla and reinstall OK