john30/ebusd #1092: ebusd container crashes after start because EBUSD_FOREGROUND is invalid/unknown

c-huetter, 2023-12-17:
### Description

Container built from development master crashes immediately after start.

Container environment settings are as follows:
```
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
EBUSD_ARCH=amd64
EBUSD_VERSION=23.2
EBUSD_FOREGROUND=
```

### Actual behavior

The container log shows the following error message below.

### Expected behavior

The exact same setup and configuration works fine with a container from the latest release 23.2.

### ebusd version

current source from git

### ebusd arguments

-c /etc/ebusd -d ens:/dev/ttyUSB0 --knxurl=ip:localhost --knxint=/etc/ebusd/knx.cfg

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

devel

### Hardware interface

adapter 3.0 USB

### Related integration

KNX

### Logs

```
2023-12-17 12:54:07.503 [main error] invalid/unknown argument in env (ignored): EBUSD_FOREGROUND=
cannot combine readonly with answer/generatesyn/initsend/scanconfig
```

john30, 2023-12-17:
its not crashing but reporting an error due to invalid arguments.
what is the Dockerfile being used?

john30, 2023-12-17:
guess I found it

closed 2023-12-17 by john30

c-huetter, 2023-12-17:
I've built a container from commit [53fcc90](https://github.com/john30/ebusd/commit/53fcc90f6e1ff1e419da7603a22e27a6c68414c8), which works like a charm. Thanks bro!

commit 53fcc90f (2023-12-17): missed change in 3d9dd9b1c3f9cc7e9f59c177883dd906c5294e5d (fixes #1092)