john30/ebusd #1095: device status: missing enhanced byte 2

hryamzik, 2023-12-17:
### Description

I've just updated firmware to [Version 20221231](https://adapter.ebusd.eu/v31/firmware/20221231-offset.hex) and it stopped working.

### Actual behavior

`2023-12-17 22:12:11.162 [bus error] device status: missing enhanced byte 2`

### Expected behavior

no errors

### ebusd version

23.2

### ebusd arguments

 --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=*** --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:/dev/ttyUSB0 --log=all:notice

### Operating system

other

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

adapter 3.0 USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2023-12-17 22:12:08.127 [bus notice] device status: resetting
2023-12-17 22:12:08.127 [bus notice] re-opened /dev/ttyUSB0
2023-12-17 22:12:11.162 [bus error] device status: missing enhanced byte 2

john30, 2023-12-23:
did you check for the correct config via ebus PIC loader as stated in the [changelog](https://adapter.ebusd.eu/v31/picfirmware.en.html#versions)?

hryamzik, 2023-12-23:
I did. I didn't get any issues after restarting the ebusd but I'm not confident with this setup now. Hopefully I'll have some time to adjust if needed once I get a chance to connect it to a heating system.

john30, 2023-12-24:
ok, closing as unrelated for now

closed 2023-12-24 by john30