john30/ebusd #1099: local csv files with erros

mediendiver, 2023-12-20:
### Description

Hi,
ich habe in meinem lokalen Ordner die Dateien 08.hmu00.csv 15.ctlv3.csv 26.vr_71.csv erstellt, diese sind leer. Wenn ich jetzt sudo systemctl start ebusd ausführe, findet er alle Dateien und zeigt diese über ebusctl info an. 
`version: ebusd 23.2.23.2
device: 192.168.178.122:9999, enhanced, firmware 1.1[3c11].1[3c11]
access: *
signal: acquired
symbol rate: 97
max symbol rate: 144
min arbitration micros: 10
max arbitration micros: 18
min symbol latency: 12
max symbol latency: 20
scan: finished
reconnects: 0
masters: 4
messages: 5
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu00.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704", loaded "vaillant/15.ctlv3.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103"`

Wenn ich jetzt die originalen Dateien nehme, den Inhalt mittels Sublime Text kopieren und in die csv Dateien einfüge, werden die csv Dateien nicht mehr geladen. Daher habe ich mal alle wieder gelöscht und nur die 26.vr_71.csv editiert mit folgenden Inhalt:
`# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comme>
#,VR_71,VR 71,104 104,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
w,,SetActorState,,,,,05,R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=of>`
und wieder ebusd gestartet, er lädt jetzt wieder alles, wenn ich dann die die Datei editiere und die Zeile:
`   w,,Mc1FlowTempDesired,,,,,0200,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,>
`
einfüge und wieder ebus starte, dann lädt er die csv Datei nicht, meine Log Datei sieht so aus:
`
ebusd.log                                                        
2023-12-20 22:29:15.165 [main notice] ebusd 23.2.23.2 started with auto scan on device: 192.168.178.122:9999, enhanced
2023-12-20 22:29:15.193 [bus notice] device status: resetting
2023-12-20 22:29:15.194 [bus notice] bus started with own address 31/36
2023-12-20 22:29:15.213 [bus notice] signal acquired
2023-12-20 22:29:15.503 [bus notice] new master 71, master count 2
2023-12-20 22:29:15.563 [bus notice] new master 03, master count 3
2023-12-20 22:29:15.564 [update notice] received unknown MS cmd: 7108b5110107 / 0a008d01c1082a00000000
2023-12-20 22:29:15.903 [bus notice] new master 10, master count 4
2023-12-20 22:29:15.923 [bus notice] device status: reset, supports info
2023-12-20 22:29:15.963 [update notice] received unknown MS cmd: 1008b5110101 / 094c4b0080ffff0000ff
2023-12-20 22:29:16.015 [bus notice] device status: extra info: firmware 1.1[3c11].1[3c11], jumpers 0x0f
2023-12-20 22:29:16.223 [update notice] received unknown MS cmd: 1076b5110101 / 094cff3d08ff530000ff
2023-12-20 22:29:16.493 [update notice] received unknown MS cmd: 1076b512030f0001 / 079d020066020eff
2023-12-20 22:29:16.701 [update notice] received unknown BC cmd: 10feb505025c00
2023-12-20 22:29:16.993 [update notice] received unknown MS cmd: 1026b5230106 / 103b02b8016b020080008000800080403d
2023-12-20 22:29:17.283 [update notice] received unknown MS cmd: 1026b5230107 / 0f008000800080008000008000805705
2023-12-20 22:29:17.546 [update notice] received unknown MS cmd: 1008b51009000050ffffff060000 / 0101
2023-12-20 22:29:17.784 [update notice] received unknown MS cmd: 1008b507020950 / 02b401
2023-12-20 22:29:18.053 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-12-20 22:29:18.323 [update notice] received unknown MS cmd: 1076b5040100 / 0a03202922201203233d08
2023-12-20 22:29:18.563 [update notice] received unknown BC cmd: 10feb516080017292220120323
2023-12-20 22:29:18.823 [update notice] received unknown MS cmd: 1008b5110100 / 0960020e000008000000
2023-12-20 22:29:19.001 [update notice] received unknown MS cmd: 7108b5110107 / 0a008d01c1082a00000000
2023-12-20 22:29:19.013 [bus notice] max. symbols per second: 105
2023-12-20 22:29:19.159 [update notice] received unknown MS cmd: 1026b5230f05ffffff000000ffffffffffff0000 / 0101
2023-12-20 22:29:19.413 [update notice] received unknown MS cmd: 1026b5230402000138 / 020100
2023-12-20 22:29:19.653 [update notice] received unknown MS cmd: 1026b5230402010150 / 02010a
[... 156 lines cut ...]
2023-12-20 22:29:28.753 [update notice] received unknown MS cmd: 1026b5230402010150 / 02010a
2023-12-20 22:29:28.965 [update notice] received unknown MS cmd: 0376b5120613000e750400 / 0202ff
2023-12-20 22:29:29.003 [bus notice] max. symbols per second: 108
2023-12-20 22:29:29.043 [update notice] received unknown MS cmd: 1026b5230402020139 / 02019c
2023-12-20 22:29:29.353 [update notice] received unknown MS cmd: 7108b5110107 / 0a008d01c1082a00000000
2023-12-20 22:29:30.931 [bus notice] scan 76: ;Vaillant;VWZIO;0902;5103
2023-12-20 22:29:30.931 [update notice] store 76 ident: done
2023-12-20 22:29:30.931 [update notice] sent scan-read scan.76  QQ=31: Vaillant;VWZIO;0902;5103
2023-12-20 22:29:30.931 [bus notice] scan 76: ;Vaillant;VWZIO;0902;5103
2023-12-20 22:29:30.932 [main error] unable to load scan config 76: no file from vaillant with prefix 76 found
2023-12-20 22:29:30.932 [main error] scan config 76: ERR: element not found
2023-12-20 22:29:32.833 [update notice] received unknown MS cmd: 7108b5110107 / 0a008d01c1082a00000000
2023-12-20 22:29:35.993 [update notice] received unknown MS cmd: 1008b5110101 / 094b4b0080ffff0000ff
2023-12-20 22:29:36.253 [update notice] received unknown MS cmd: 1076b5110101 / 094cff3d08ff530000ff
2023-12-20 22:29:36.393 [update notice] received unknown MS cmd: 7108b5110107 / 0a008d01c1082a00000000
2023-12-20 22:29:36.567 [update notice] received unknown MS cmd: 1076b512030f0001 / 079d020064020eff
2023-12-20 22:29:36.645 [update notice] received unknown MS cmd: 0376b5120613000e6c0400 / 0202ff
2023-12-20 22:29:36.895 [update notice] received unknown MS cmd: 1026b5230106 / 103902b90169020080008000800080403d
2023-12-20 22:29:37.033 [bus notice] max. symbols per second: 123
2023-12-20 22:29:37.193 [update notice] received unknown MS cmd: 1026b5230107 / 0f008000800080008000008000805705
2023-12-20 22:29:37.453 [update notice] received unknown MS cmd: 1008b51009000050ffffff060000 / 0101
2023-12-20 22:29:37.713 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2023-12-20 22:29:38.000 [update notice] received write vr_71 SetActorState QQ=10: -;-;-;off;off;off;-;-;-;-;-;-;00 00
2023-12-20 22:29:38.249 [update notice] received unknown MS cmd: 1026b5230402000138 / 020100
2023-12-20 22:29:38.503 [update notice] received unknown MS cmd: 1026b5230402010150 / 02010a

mediendiver, 2023-12-21:
d

john30, 2023-12-23:
closed due to missing description

closed 2023-12-23 by john30