john30/ebusd #1108: Xanthophyllum 23.3 not present in alpine linux repository

tjorim, 2023-12-26:
### Description

Similar issue to https://github.com/john30/ebusd/issues/958. Latest release notes (https://github.com/john30/ebusd/releases/tag/23.3)
suggest to run `apk --repository=http://dl-cdn.alpinelinux.org/alpine/edge/community add ebusd` for Alpine Linux. However, build is not present. Alpine repo still shows 'Tim' as developer but he said earlier he does not use or build it on alpine anymore.

### Actual behavior

Does not install ebusd 23.3, getting 23.2 instead.

### Expected behavior

Installs ebusd version 23.3.

### ebusd version

23.3

### ebusd arguments

n/a

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via USB

### Logs

n/a

john30, 2023-12-26:
I'll drop the line in the release notes until it is fixed

piotrtobolski, 2024-06-06:
@john30 Users using Docker are waiting for ebusd to be available in Alpine repository. Do you think next version will be available soon?

I see there wasn't a lot of changes since 23.3 and it is already about half of a year since 23.3 was released. Maybe 23.3.1 can be released?

john30, 2024-06-15:
why do docker user not simply use the docker devel image then?
`docker pull john30/ebusd:devel`

apbarratt, 2024-10-10:
Please reopen and repair this.
Docker users can indeed simply use the devel image, but users of the HomeAssistant Add-On cannot rely on this as releasing a version that points to a development build would be irresponsible for the devs maintaining that work there to do.

apbarratt, 2024-10-10:
I feel there must be some story I'm missing here for why there is push back on releasing for Alpine when it's been included before.  What is it I'm not understanding for why this decision has been made this time but not on previous occasions? :/

apbarratt, 2024-10-10:
For clarity, I've no idea why someone might choose to use Alpine over any other distro.  It just seems unusual that Alpine has been included in the past and LukasGrebe's project was well recommended and used frequently (i see yours is a fork of it, I'll give it a go, thank you!) And when it was noticed that this one had been missed, that sounds like it was just overlooked and not a deliberate error, it's then had quite a lot of push back on correcting that mistake.  It's that last bit I'm confused by.  What am I missing? :/

apbarratt, 2024-10-10:
That addon of yours seems to be working like a charm, thank you muchly :)

piotrtobolski, 2024-10-10:
I'm not expert at using Docker but if I understand it correctly, Alpine is often used because it is very small (in terms of MB) distribution. Here the problem was that the released version was for some reason incompatible with Alpine and it is already fixed but releasing it would require additional effort and @john30 decided it just isn't worth it. 23.4 (or whatever the next version will be called) should be working on Alpine as previously.

john30, 2024-10-27:
merge request for 24.1 submitted: https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/74199

john30, 2024-11-11:
merged in https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/74289

closed 2024-11-11 by john30