john30/ebusd #1113: Error select entities from csv vr_71

Desmoyo, 2023-12-29:
### Description

When I installed the new version 23.3 the day the new version 23.3 came out, the Select of the csv vr_71 disappeared from HA, they start with Mc1 to Mc3 and some more depending on the type of circuit.  I can search for that device and I have a response, I can see movement in mqttExplorer but they are not added in HA, my mqtt-hassio is always the same, I have not touched anything, if I go back to version 23.2 the devices appear again, What can happen?

### Actual behavior

All vr_71 select devices disappear

### Expected behavior

That those select from vr_71.csv do not disappear or can be included in HA

### ebusd version

current source from git

### ebusd arguments

version: '3' services:     ebusd:         container_name: ebusd         restart: unless-stopped         image: john30/ebusd:latest         privileged: true         environment:           - EBUSD_SCANCONFIG=full           - EBUSD_CONFIGPATH=/ebusd/ebusd-configuration/ebusd-2.1.x/en           #ebusd/ebusd-configuration/devel/en           - EBUSD_ACCESSLEVEL=*           - EBUSD_DEVICE=ens:192.168.88.252:9999           - EBUSD_MQTTHOST=192.168.88.85        #   - EBUSD_MQTTVAR=filter-direction=           - EBUSD_MQTTPORT=1883           - EBUSD_MQTTUSER=xxxxxxxx          - EBUSD_MQTTPASS=xxxxxx          - EBUSD_MQTTJSON=           - EBUSD_LATENCY=0           - EBUSD_MQTTINT=/ebusd/mqtt-hassio.cfg           - EBUSD_UPDATECHECK=on           - TZ=Europe/Madrid      # - EBUSD_LOGAREAS: all            - EBUSD_POLLINTERVAL=5     # - EBUSD_MQTTHOST: mqtt         volumes:          # - /docker/ebusd/:/ebusd/ebusd-configuration/ebusd-2.1.x/en           - /docker/ebusd/:/ebusd           - /etc/localtime:/etc/localtime:ro     

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

adapter 5 via Ethernet

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

Ninguno

john30, 2023-12-30:
what do you mean with the "Select of the csv"?
please provide an ebusd logging of around 1 minute since start for both versions

Desmoyo, 2024-01-02:
> ¿A qué te refieres con la "Selección del csv"? proporcione un registro de ebusd de aproximadamente 1 minuto desde el inicio para ambas versiones

Hello @john30 , I don't want to say the csv section, I'm referring to the entities with writable option selection from the csv vr_71, I don't know if I explain myself, I'll give you a photo, all of those do not appear with the latest version 23.3, there's no way that HA catches them and they are lost and there are readings and they can be polled 

[image]
[image]

john30, 2024-02-04:
still dont get your point. please pick one concrete example (instead of that many) in order to track it down. then check the mqtt messages for the particular one in the definition and value part so that it will be comparable

Desmoyo, 2024-02-08:
> Todavía no entiendo tu punto. Elija un ejemplo concreto (en lugar de tantos) para poder localizarlo. luego verifique los mensajes mqtt para el mensaje en particular en la parte de definición y valor para que sea comparable

I just want to say that with the latest version of ebusd 23.3 all the entities of the vr_71 device disappear from HA and mqtt Explorer continues reading them without problem, but they do not update their state and remain unknown in HA.  I have installed version 23.2 and they already work but with version 23.3 they disappear from HA.

john30, 2024-02-11:
if thats the case then please check the HA logging and also check the discovery messages for potential differences. you didn't forget to adjust the HA integration settings as needed, right? also there were some adjustments to the integration file: https://github.com/john30/ebusd/compare/23.2...23.3#diff-4952dade4072db5f6f477603bb2c18fa861198bc099ac7f746e00fb9a853c297

jan666, 2024-10-28:
I have the same problem, see https://github.com/john30/ebusd/issues/1357

I started with 23.3 and was unable to to the VR 70 values working. Found this ticket, downgraded to 23.2 and they are there.

john30, 2024-10-28:
see https://github.com/john30/ebusd/issues/1357#issuecomment-2442483651
anyway, such messages could be sent as passive read instead to discovery. I'll check if that's feasible

jan666, 2024-10-29:
An option to send em read only would be nice, too. I just want to log it

kromosomX, 2024-11-08:
It would be great if all multi field write messages would be treated as read-only. VR71 and VR70 units get updated very often (every couple of seconds) and are therefore very useful as read only for reading important information (pump on/off, desired flow temperature, mixer status). 

Also as a quick and dirty workaround you can:
1. set all writable fields as readable in configuration file 26.vr_71.csv (w->r)
2. restart ebusd
3. wait until the sensors appear in homeassistant/senos/ or just wait couple of minutes
4. set all fileds changed in step 1. back to readable
5. restart ebusd

But if you restart HA you have to do it all over again. 

This could all be simplified by simply automating the generation of discovery messages for homeassistant but I don't know how to do it... yet.

jan666, 2024-11-08:
> It would be great if all multi field write messages would be treated as read-only. VR71 and VR70 units get updated very often (every couple of seconds) and are therefore very useful as read only for [...]

I just tried it and it works. Too bad we have to change it back to `w` to really get the data. Would be a nice workaround if it just worked with `r` in the csv

john30, 2024-11-15:
should work in passive mode I think, so try with "u" instead of "w" or "r"

jan666, 2024-11-18:
> should work in passive mode I think, so try with "u" instead of "w" or "r"

Maybe I miss something. I did this:

-w,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
-w,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
-w,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
+u,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
+u,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
+u,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,

so replacing the "w"s with "u"s. But then ebusd won't start:

`2024-11-18 07:00:17.638 [main error] error reading scan config file vaillant/52.vr_70.csv for ID "vr_70", SW0109, HW2903: ERR: missing argument, vaillant/52.vr_70.csv:5: ERR: missing argument, circuit`

john30, 2024-11-24:
also the default line with "*w" needs to be adjusted to "*u" or duplicated with "*u"

jan666, 2024-11-25:
52.vr_70.csv

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment,field6,part (m/s),datatypes/templates,divider/values,unit,comment,field7,part (m/s),datatypes/templates,divider/values,unit,comment
#,VR_70,VR 70,109 109,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*w,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*u,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
u,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
u,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
u,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

```

ebusd starts but

```
2024-11-25 07:51:55.223 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:00.098 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:00.098 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
2024-11-25 07:52:04.917 [update error] unable to parse update-read vr_70 SetActorState from 1052b523080100ff0014ffff00 / 0101: ERR: invalid position
2024-11-25 07:52:05.165 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:10.112 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:10.112 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
2024-11-25 07:52:14.923 [update error] unable to parse update-read vr_70 SetActorState from 1052b523080100ff0014ffff00 / 0101: ERR: invalid position
2024-11-25 07:52:15.171 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:20.123 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:20.123 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
2024-11-25 07:52:25.714 [update error] unable to parse update-read vr_70 SetActorState from 1052b523080100ff0014ffff00 / 0101: ERR: invalid position
2024-11-25 07:52:26.227 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:30.135 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:30.135 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
2024-11-25 07:52:34.959 [update error] unable to parse update-read vr_70 SetActorState from 1052b523080100ff0014ffff00 / 0101: ERR: invalid position
2024-11-25 07:52:35.244 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:39.230 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:39.230 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
2024-11-25 07:52:44.934 [update error] unable to parse update-read vr_70 SetActorState from 1052b523080100ff0014ffff00 / 0101: ERR: invalid position
2024-11-25 07:52:45.182 [update error] unable to parse update-read vr_70 Mc2FlowTempDesired from 1052b5230402010150 / 020105: ERR: invalid position
2024-11-25 07:52:49.242 [mqtt error] decode vr_70 SetActorState: ERR: invalid position
2024-11-25 07:52:49.242 [mqtt error] decode vr_70 Mc2FlowTempDesired: ERR: invalid position
```

and no more updates in HASS. So I guess this won't work?

john30, 2024-11-26:
my bad, "*uw" it needs to be of course

jan666, 2024-11-27:
`52.vr_70.csv`

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment,field6,part (m/s),datatypes/templates,divider/values,unit,comment,field7,part (m/s),datatypes/templates,divider/values,unit,comment
#,VR_70,VR 70,109 109,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
u,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
u,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
u,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

```

`2024-11-27 06:51:40.156 [main error] error reading scan config file vaillant/52.vr_70.csv for ID "vr_70", SW0109, HW2903: ERR: missing argument, vaillant/52.vr_70.csv:5: ERR: missing argument, circuit
`

john30, 2024-12-01:
you need to change both, the default and the line with the message

jan666, 2024-12-02:
Ok, I think I finally got it :)

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment,field6,part (m/s),datatypes/templates,divider/values,unit,comment,field7,part (m/s),datatypes/templates,divider/values,unit,comment
#,VR_70,VR 70,109 109,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*r,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,"B523",,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,SetActorState,,,,,"01",R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,S7,,UCH,0=off;20=on,,
uw,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
uw,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
r,,SensorData,,,,,"03",S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,ignore,,HEX:3,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

thanks for your patience.

commit b7a65fc8 (2025-04-13): treat write message as read for mqtt when excluded by config (fixes #1113)
closed 2025-04-13 by john30