john30/ebusd #1117: Adapter v5 no signal

Solarer, 2023-12-30:
### Description

I connected 2 ebus adapter v5 via USB to a raspberry pi 3B running raspbian. One ebus adapter is connected to a Vaillant recovair 260/4 (ventilation) and the other one to a Vaillant Auto compact (heating). I am unable to get the connection initialized on both adapters. 

ebusd is running in a docker container:
```
sudo docker run --rm -d --name=ebusd --device=/dev/ebus-lueftung:/dev/ttyUSB0 -p 8888 john30/ebusd --scanconfig=full -d ens:/dev/ttyUSB0
```
I mapped `/dev/ttyACM0 to /dev/ebus-lueftung` using a udev rule. I also tried using `enh:/dev/...` and plain `-d /dev/...` instead of `ens` but get similar results.

After connecting to the docker container, this is the `info` that I get:
```
root@b5ee6777f622:/# ebusctl
localhost: info
version: ebusd 23.2.23.2
update check: version 23.3 available
device: /dev/ttyUSB0, enhanced
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

I get the same 'no signal' for both Vaillant devices - so I am most likely doing something wrong.
Green LED is always on. Blue LED blinking once every few seconds.

Log output:

```
pi@PI:~ $ sudo docker run --rm -d --name=ebusd --device=/dev/ebus-lueftung:/dev/ttyUSB0 -p 8888 john30/ebusd --scanconfig=full -d ens:/dev/ttyUSB0 --loglevel=debug --logareas=bus,main
e83ffc47a60a5d88f5c273b03c7c6a27e3937dcc41b1898a1334580b6ef25c71
pi@PI:~ $ sudo docker logs ebusd --follow --tail 500                                
2023-12-30 12:17:54.309 [main info] configPath URL is valid                               
2023-12-30 12:17:54.310 [main notice] ebusd 23.2.23.2 started with full scan on device: /dev/ttyUSB0, enhanced high speed
2023-12-30 12:17:54.310 [main info] loading configuration files from https://cfg.ebusd.eu/
2023-12-30 12:17:54.489 [main info] reading templates /
2023-12-30 12:17:54.667 [main info] read templates in /
2023-12-30 12:17:54.667 [main info] reading file broadcast.csv
2023-12-30 12:17:54.847 [main info] successfully read file broadcast.csv
2023-12-30 12:17:54.847 [main info] reading file memory.csv
2023-12-30 12:17:55.027 [main info] successfully read file memory.csv
2023-12-30 12:17:55.027 [main info] read config files, got 11 messages
2023-12-30 12:17:55.028 [bus notice] device status: resetting
2023-12-30 12:17:55.028 [bus notice] bus started with own address 31/36
2023-12-30 12:17:55.029 [bus notice] device status: reset, supports info
2023-12-30 12:17:55.029 [main info] registering data handlers
2023-12-30 12:17:55.029 [main info] registered data handlers
2023-12-30 12:18:05.030 [main debug] performing regular tasks
2023-12-30 12:18:15.030 [main debug] performing regular tasks
...
```

Any help is appreciated 👍

### Actual behavior

signal: no signal

### Expected behavior

signal: aquired

### ebusd version

23.2

### ebusd arguments

--scanconfig=full -d ens:/dev/ttyUSB0

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

armv7l

### Dockerized

same as ebusd version

### Hardware interface

adapter 5 via USB

### Logs

see above

Solarer, 2023-12-30:
Pulled the latest docker image 23.3.23.3 - same result.

john30, 2023-12-31:
I guess the device will not appear as USB in docker but as ACM, just check by using 
`sudo docker run -it --rm --device=/dev/ebus-lueftung:/dev/ttyUSB0 john30/ebusd ls -la /dev/ttyACM0 /dev/ttyUSB0`

Solarer, 2023-12-31:
My understanding is that the syntax `/dev/A:/dev/B`
passes A to the docker container and mounts it there as B. Confirmed by the output of your command:

```
pi@MightyPI:~ $ sudo docker run -it --rm --device=/dev/ebus-lueftung:/dev/ttyUSB0 john30/ebusd ls -la /dev/ttyACM0 /dev/ttyUSB0

ls: cannot access '/dev/ttyACM0': No such file or directory
crw-rw---- 1 root dialout 166, 0 Dec 31 10:15 /dev/ttyUSB0
```

What exactly does it mean for ebusd to have `no signal`? Is it a wiring issue? Should I be able to detect a voltage on the BUS line with or without the adapter?

john30, 2023-12-31:
ah sorry, didn't see that you renamed it.
the log suggests that connection to the adapter is fine ("device status: reset, supports info").
but the "signal acquired" message is missing, which does not fit to the green LED blinking as this is only the case when data really arrives. maybe check with raw logging to see whats coming in

Solarer, 2023-12-31:
I tried to enable raw logging like this, but not sure if it is working as expected:

```
sudo docker run -d --rm --name=ebusd --device=/dev/ebus-lueftung:/dev/ttyUSB0 -p 8888 john30/ebusd --scanconfig=full -d /dev/ttyUSB0 -l /var/log/ebusd.log --logareas=all --lograwdata --lograwdatafile=/var/log/raw
```

/var/log/raw is created but empty
/var/log/ebus.log does not exist (I assume logs get absorbed into docker standard log)

ebusd logs now contain a lot of network debug messages even though the adapter is connected via USB. Is this an issue?
```
sudo docker logs ebusd

2023-12-31 12:49:59.463 [network debug] SSL state 0=before SSL initialization: type 0x10=start, val 1=?
2023-12-31 12:49:59.464 [network debug] SSL state 0=before SSL initialization: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.465 [network debug] SSL BIO state 2 res -1
2023-12-31 12:49:59.469 [network debug] SSL BIO state 3 res 1
2023-12-31 12:49:59.469 [network debug] SSL BIO state 4 res 3
2023-12-31 12:49:59.470 [network debug] SSL BIO state 6 res 0
2023-12-31 12:49:59.470 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-12-31 12:49:59.490 [network debug] SSL BIO state 5 res -1
2023-12-31 12:49:59.490 [network debug] SSL BIO state 5 res 1
2023-12-31 12:49:59.490 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.490 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-12-31 12:49:59.511 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.513 [network debug] SSL state 3=SSLv3/TLS read server hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.513 [network debug] SSL state 38=TLSv1.3 read encrypted extensions: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.518 [network debug] SSL state 4=SSLv3/TLS read server certificate: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.521 [network debug] SSL state 39=TLSv1.3 read server certificate verify: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.521 [network debug] SSL state 11=SSLv3/TLS read finished: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.521 [network debug] SSL state 16=SSLv3/TLS write change cipher spec: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.522 [network debug] SSL state 18=SSLv3/TLS write finished: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.522 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x20=done, val 1=?
2023-12-31 12:49:59.522 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2023-12-31 12:49:59.522 [main info] configPath URL is valid
2023-12-31 12:49:59.523 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x4008=write,alert, val 256=close notify
2023-12-31 12:49:59.525 [bus notice] device status: transport opened
2023-12-31 12:49:59.525 [main info] registering data handlers
2023-12-31 12:49:59.525 [main info] registered data handlers
2023-12-31 12:49:59.525 [main notice] ebusd 23.3.23.3 started with full scan on device: /dev/ttyUSB0, serial
2023-12-31 12:49:59.525 [main info] loading configuration files from https://cfg.ebusd.eu/2023-12-31 12:49:59.525 [network debug] SSL state 0=before SSL initialization: type 0x10=start, val 1=?
2023-12-31 12:49:59.525 [network debug] SSL state 0=before SSL initialization: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.527 [network debug] SSL BIO state 2 res -1
2023-12-31 12:49:59.528 [network debug] SSL BIO state 3 res 1
2023-12-31 12:49:59.528 [network debug] SSL BIO state 4 res 5
2023-12-31 12:49:59.529 [network debug] SSL BIO state 6 res 0
2023-12-31 12:49:59.529 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-12-31 12:49:59.549 [network debug] SSL BIO state 5 res -1
2023-12-31 12:49:59.549 [network debug] SSL BIO state 5 res 1
2023-12-31 12:49:59.549 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.549 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2023-12-31 12:49:59.569 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.572 [network debug] SSL state 3=SSLv3/TLS read server hello: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.572 [network debug] SSL state 38=TLSv1.3 read encrypted extensions: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.575 [network debug] SSL state 4=SSLv3/TLS read server certificate: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.577 [network debug] SSL state 39=TLSv1.3 read server certificate verify: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.578 [network debug] SSL state 11=SSLv3/TLS read finished: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.578 [network debug] SSL state 16=SSLv3/TLS write change cipher spec: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.579 [network debug] SSL state 18=SSLv3/TLS write finished: type 0x1001=loop,connect, val 1=?
2023-12-31 12:49:59.579 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x20=done, val 1=?
2023-12-31 12:49:59.579 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
[... cut ...]
2023-12-31 12:53:10.138 [main debug] performing regular tasks
2023-12-31 12:53:20.138 [main debug] performing regular tasks
2023-12-31 12:53:30.139 [main debug] performing regular tasks
2023-12-31 12:53:40.139 [main debug] performing regular tasks
2023-12-31 12:53:50.139 [main debug] performing regular tasks
2023-12-31 12:54:00.139 [main debug] performing regular tasks
2023-12-31 12:54:10.140 [main debug] performing regular tasks
2023-12-31 12:54:20.140 [main debug] performing regular tasks
```

john30, 2024-01-05:
no need to change log level to debug.
you can get inline raw logging by using --lograwdata=bytes without lograwdatafile.
anyway: can you test the same when using transfer via wifi? just to exclude that something in the hardware is broken

john30, 2024-01-06:
I think I got it, seems to be related to https://github.com/espressif/esp-idf/commit/b8e8042

john30, 2024-01-06:
please check with new firmware [20240106](https://adapter.ebusd.eu/v5/ChangeLog#20240106)

john30, 2024-01-06:
closing this one, as it is no ebusd issue in the first place

closed 2024-01-06 by john30

Solarer, 2024-01-07:
@john30 
Updating the firmware did not change anything but with WIFI connection the scan completed successfully.

I would prefer to get this running via USB since I have almost no signal in the basement. Where should I open a new ticket?

```
localhost: info
version: ebusd 23.3.23.3
device: 192.168.0.99:9999, TCP, enhanced, firmware 1.1[4106].1[4106]
signal: acquired
symbol rate: 24
max symbol rate: 122
min arbitration micros: 6
max arbitration micros: 372
min symbol latency: 13
max symbol latency: 69
scan: running
reconnects: 0
masters: 4
messages: 325
conditional: 2
poll: 1
update: 10
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=VMS01;SW=0116;HW=0303", loaded "vaillant/06.vms.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0205;HW=9602", loaded "vaillant/bai.0010015600.inc" ([PROD='0010015611']), "vaillant/08.bai.csv"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0614;HW=6903", loaded "vaillant/ec.sol.sc.csv"
address ed: slave, scanned "MF=Vaillant;ID=VMS01;SW=0116;HW=0303"
```

john30, 2024-01-11:
https://github.com/john30/ebusd-esp32