john30/ebusd #1120: MQTT Homeassistant Support for "datetime" and "date" Fields

Xyaren, 2023-12-30:
### Description

Similar to https://github.com/john30/ebusd/issues/1006
I would like to request Homeassistant MQTT Discovery support for date(times)s aswell.

guillaumezin, 2024-01-03:
Hello, and happy new year.
Same here, for Domoticz, I can't figure out how to get date working, I'm having a really hard time understanding how parsing is supposed to work...

john30, 2024-01-07:
someone wants to find out how date+datetime are represented in HA when exchanged via MQTT?

Xyaren, 2024-01-08:
Currently I suggest using text as there is no dedicated date / datetime / time entity supported for mqtt discovery:

https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery:~:text=Entity%20integrations%20supported%20by%20MQTT%20discovery

https://community.home-assistant.io/t/mqtt-discovery-input-datetime/444720

https://community.home-assistant.io/t/mqtt-discovery-time-datetime/663870 <- Please Vote for the feature request to make it visible.

guillaumezin, 2024-01-10:
Text would be enough for Domoticz. But to be honest, I don't need this anymore, I'm using Hc1SFMode and Hc2SFMode to set holiday mode and it seems to work as expected

Koky05, 2024-01-11:
I add date and time in file "mqtt-hassio.cfg" on row 108 `filter-name = ....|date|time|...`

Change on rows from 148:
```
# HA integration: skip string/date/time types completely
type_map-string = string
type_map-date = date
type_map-time = time
```

Change on rows from 230:
```
type_switch-w-date =
    text,,,date = date,
    text,, = date,
    text,, =

type_switch-w-datetime =
    text,, = btime,
    text,, = dcfstate,
    text,, = datetime,
    text,, = bdate,
    text,, = 

type_switch-w-string =
    text,, = date,
    text,, = datetime,
    text,, = dcfstate,
    text,, = bdate,
    text,, = btime,
    text,, =

type_switch-date =
    sensor,, = date,
    sensor,, =

type_switch-datetime =
    sensor,, =

type_switch-string =
    sensor,, =
```

Change on rows from 330:
```
type_part-textdate = ,
  "command_topic":"%topic/set",
  "pattern": "^[0-3][0-9].[01][0-9].20[0-3][0-9]$"%state_class

type_part-textdatetime = ,
  "command_topic":"%topic/set"%state_class

type_part-text = ,
  "command_topic":"%topic/set"%state_class
```

[image]

Dinth, 2024-01-11:
Thanks for all the work on this, i still fail to understand how to declare anything in mqtt config file, but setting date is one of the most critical applications of ebusd im using.

Koky05, 2024-01-11:
I use local config files for eBUS Deamon so I also lower priority of that values. In other case you also can skip recorder for that values at all:

> exclude map (optional)
> Configure which integrations should be excluded from recordings. ([Configure Filter](https://www.home-assistant.io/integrations/recorder/#configure-filter))
> 
> domains list (optional)
> The list of domains to be excluded from recordings.
> 
> entity_globs list (optional)
> Exclude all entities matching a listed pattern from recordings (e.g., sensor.weather_*).
> 
> entities list (optional)
> The list of entity ids to be excluded from recordings.
> 
> event_types list (optional)
> The list of event types to be excluded from recordings.

```
# Example filter to include specified domains and exclude specified entities
recorder:
  include:
    domains:
      - alarm_control_panel
      - light
    entity_globs:
      - binary_sensor.*_occupancy
  exclude:
    entities:
      - light.kitchen_light
```

Xyaren, 2024-01-11:
> I use local config files for eBUS Deamon so I also lower priority of that values. In other case you also can skip recorder for that values at all:
>
>
> ```
> # Example filter to include specified domains [...]

I am aware of that, it's just something that you need to do explicitly.
Therefore exposing the property by default would be a bad idea.

john30, 2024-02-11:
> I am aware of that, it's just something that you need to do explicitly. Therefore exposing the property by default would be a bad idea.

thats what I think as well. would be better to just prepare it in the integration file and having to enable it explicitly imho (also valid for #1134)

Xyaren, 2024-02-11:
That is exactly what #1134 does. It does not automatically expose the attributes, just makes them editable if they are exposed.

chrizzzp, 2024-02-15:
I noticed even with the additions of https://github.com/john30/ebusd/pull/1134 in place, time entities with the datatype HTI are not exposed to HA. Does it require a HA template definition?

e.g. _templates.csv:
`time3,HTI,,,time`

Xyaren, 2024-05-13:
Any explanation on why the PR is not merged ?

commit 6d0511a1 (2024-10-02): add date+datetime mapping (closes #1120 and #1134)
closed 2024-10-02 by john30