john30/ebusd #1121: Default PID path not accessible

TeXiCiTy, 2023-12-31:
### Description

When using a self-compiled ebusd on a compute module 3, the default PID path makes it impossible to start:

Dec 31 13:21:37 systemd[1]: ebusd.service: Can't open PID file /run/ebusd.pid (yet?) after start: No such file or directory

Actions done on a fresh install: (bookworm & buster, both lite)
sudo apt-get update
sudo apt-get upgrade
sudo apt install git autoconf automake g++ make libssl-dev libmosquitto-dev cmake
git clone https://github.com/john30/ebusd.git
cd ebusd
sudo ./autogen.sh
cmake .
sudo make
sudo make install/strip
sudo systemctl enable ebusd
sudo nano /etc/default/ebusd --> set EBUSD_OPTS
sudo service ebusd start

The problem can be solved by adding --pidfile=/var/run/ebusd.pid to the EBUSD_OPTS but this used the be default.

### Actual behavior

Ebusd wants to write to /run/ebusd.pid but it's not permitted

### Expected behavior

Ebusd wants to write to /var/run/ebusd.pid and start normally

### ebusd version

current source from git

### ebusd arguments

--scanconfig --device=/dev/serial0 --latency=20 --configpath=/etc/ebusd/config --accesslevel=*

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat), MQTT generic

### Logs

From "journalctl -xeu ebusd.service":

The job identifier is 1318.
Dec 31 13:21:37 hostname systemd[1]: ebusd.service: Can't open PID file /run/ebusd.pid (yet?) after start: No such file or directory
Dec 31 13:23:07 hostname systemd[1]: ebusd.service: start operation timed out. Terminating.
Dec 31 13:23:07 hostname systemd[1]: ebusd.service: Failed with result 'timeout'.

john30, 2024-02-04:
what does `ebusd --help|grep ebusd.pid` give? if this is other than /var/run/ebusd.pid then your compilation is weird and you should check the default of the used autoconf localstatedir

TeXiCiTy, 2024-02-05:
I don't have access to this particular box for the coming weeks, so it'll take some time to check.
Maybe something with sudo which shouldn't be run as sudo?
The install/strip option instead of install-strip is a bit odd as well. The wiki says "install-strip" but I only get the "install/strip" option.

john30, 2024-02-11:
this is due to cmake doing things a bit different. try autoconf instead

TeXiCiTy, 2024-03-26:
Skipping the "cmake ." command and continuing with "make" solved the problem for me.

closed 2024-03-29 by john30