john30/ebusd #1122: HA discovery schema doesn't respect field1 keys

syssi, 2024-01-03:
### Description

I tried to set the `ReducedSetTemp` of my Weishaupt WTC-25 setup using this 3rd party configuration: https://github.com/J0EK3R/ebusd-configuration-weishaupt

Home Assistant complained about the structure of the MQTT payload:

```
2024-01-03 18:00:54.919 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'dict object' has no attribute '_16_Temp10' when rendering '{{value_json["_16_Temp10"].value}}'
2024-01-03 18:00:54.920 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute '_16_Temp10' (value: {
     "ReducedSetTemp": {"value": 14.0}}, template: {{value_json["_16_Temp10"].value}})
```

The schema description of the auto discovery looks like this:

```
homeassistant/number/ebusd_hc1_ReducedSetTemp__16_Temp10/config {
  "unique_id":"ebusd_hc1_ReducedSetTemp__16_Temp10",
  "name":"ReducedSetTemp _16_Temp10",
  "device":{
   "identifiers":"ebusd_hc1",
   "manufacturer":"ebusd.eu",
   "name":"ebusd hc1",
   "via_device":"ebusd",
   "sw_version":"23.3",
   "suggested_area":"Heating"
  },
  "value_template":"{{value_json[\"_16_Temp10\"].value}}",
  "state_topic":"ebusd/hc1/ReducedSetTemp",
  "command_topic":"ebusd/hc1/ReducedSetTemp/set",
  "min":-3276.7,
  "max":3276.7,
  "step":0.1,
  "unit_of_measurement":"°C",
  "device_class":"temperature"
 }
```

This doesn't match with the MQTT payload:

```
ebusd/hc1/ReducedSetTemp {
     "ReducedSetTemp": {"value": 14.0}
}
```

It looks like the MQTT payload uses the `field1` value as object key. The Home Assistant discovery schema applies the `template` value instead.

[image]

### Actual behavior

The MQTT payload uses the `field1` value as object key. HA tries to access the `template` value as object key. 

### Expected behavior

Both should be in sync. If I remove the speaking `field1` value from the configuration ebusd uses the `template` value instead. This is the new and HA compatible payload:

```
ebusd/hc1/ReducedSetTemp {
     "_16_Temp10": {"value": 14.0}
}
```

### ebusd version

23.3

### ebusd arguments

--jsonformat

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

adapter 5 via USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

```
2024-01-03 18:25:47.693 [update notice] sent read hc1 ReducedSetTemp QQ=31: 14.0
```

john30, 2024-02-11:
the issue is the message definition here which differs between the read and write. the read specifies the field name but the write does not and as a consequence, the template name is used instead. this then does not match with the expectation from discovery.
so either remove the field1 name from the read message or add it to the write message. it would be better anyway, to combine such definitions into a single line using "rw" in the first column to avoid exactly such.

syssi, 2024-02-11:
Thanks for your analysis! Let's close this issue because it's resolved.

closed 2024-02-11 by syssi