john30/ebusd #1132: Ebusd compilation fails: src/lib/utils/libutils.so: undefined reference to `SSL_alert_desc_string_long' (and more)

ConiKost, 2024-01-08:
### Description

Since I finally got an Ebusd v5 Adapter, I wanted to install Ebusd on Gentoo Linux. As no packages exists, I started to write my own ebuild. Unfortunately, the compilation itself seems to have a problem with SSL.

Note: While I am using version 23.3, I also applied https://github.com/john30/ebusd/commit/3384f3780087bd6b94d46bf18cdad18201ad516c, as it would fail otherwise due gcc13.

### Actual behavior

```
[34/40] : && /usr/lib/ccache/bin/x86_64-pc-linux-gnu-g++ -fPIC -march=native -mfpmath=sse,387 -mtune=intel -O3 -frecord-gcc-switches -fomit-frame-pointer -malign-data=abi -mtls-dialect=gnu2 -pipe -fvisibility-inlines-hidden -std=c++11  -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -Wl,-O1 -Wl,--sort-common -Wl,-z,now -shared -Wl,-soname,libknx.so -o src/lib/knx/libknx.so src/lib/knx/CMakeFiles/knx.dir/knx.cpp.o   && :
[35/40] : && /usr/lib/ccache/bin/x86_64-pc-linux-gnu-g++ -march=native -mfpmath=sse,387 -mtune=intel -O3 -frecord-gcc-switches -fomit-frame-pointer -malign-data=abi -mtls-dialect=gnu2 -pipe -fvisibility-inlines-hidden -std=c++11 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -Wl,-O1 -Wl,--sort-common -Wl,-z,now src/ebusd/CMakeFiles/ebusd.dir/bushandler.cpp.o src/ebusd/CMakeFiles/ebusd.dir/datahandler.cpp.o src/ebusd/CMakeFiles/ebusd.dir/request.cpp.o src/ebusd/CMakeFiles/ebusd.dir/network.cpp.o src/ebusd/CMakeFiles/ebusd.dir/mainloop.cpp.o src/ebusd/CMakeFiles/ebusd.dir/scan.cpp.o src/ebusd/CMakeFiles/ebusd.dir/main.cpp.o src/ebusd/CMakeFiles/ebusd.dir/main_args.cpp.o src/ebusd/CMakeFiles/ebusd.dir/knxhandler.cpp.o -o src/ebusd/ebusd  -Wl,-rpath,/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/ebus:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/knx:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/ebus/contrib:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/utils:  src/lib/ebus/libebus.so  -lpthread  -lrt  src/lib/knx/libknx.so  -lssl  -lcrypto  src/lib/ebus/contrib/libebuscontrib.so  src/lib/utils/libutils.so && :
FAILED: src/ebusd/ebusd
: && /usr/lib/ccache/bin/x86_64-pc-linux-gnu-g++ -march=native -mfpmath=sse,387 -mtune=intel -O3 -frecord-gcc-switches -fomit-frame-pointer -malign-data=abi -mtls-dialect=gnu2 -pipe -fvisibility-inlines-hidden -std=c++11 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -Wl,-O1 -Wl,--sort-common -Wl,-z,now src/ebusd/CMakeFiles/ebusd.dir/bushandler.cpp.o src/ebusd/CMakeFiles/ebusd.dir/datahandler.cpp.o src/ebusd/CMakeFiles/ebusd.dir/request.cpp.o src/ebusd/CMakeFiles/ebusd.dir/network.cpp.o src/ebusd/CMakeFiles/ebusd.dir/mainloop.cpp.o src/ebusd/CMakeFiles/ebusd.dir/scan.cpp.o src/ebusd/CMakeFiles/ebusd.dir/main.cpp.o src/ebusd/CMakeFiles/ebusd.dir/main_args.cpp.o src/ebusd/CMakeFiles/ebusd.dir/knxhandler.cpp.o -o src/ebusd/ebusd  -Wl,-rpath,/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/ebus:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/knx:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/ebus/contrib:/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build/src/lib/utils:  src/lib/ebus/libebus.so  -lpthread  -lrt  src/lib/knx/libknx.so  -lssl  -lcrypto  src/lib/ebus/contrib/libebuscontrib.so  src/lib/utils/libutils.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_alert_desc_string_long'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_free'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_ctrl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `X509_NAME_get_text_by_NID'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_set_info_callback'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_state_string_long'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_new_connect'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_read_ex'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_callback_ctrl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_load_verify_locations'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `X509_get_subject_name'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `X509_free'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_get_verify_result'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_write_ex'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_ctrl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_free_all'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_test_flags'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `OPENSSL_init_ssl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_set_verify'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_new'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `TLS_method'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_new_ssl_connect'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `ERR_reason_error_string'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_set_default_verify_paths'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_CTX_set_options'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_get1_peer_certificate'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_int_ctrl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_get_state'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `ERR_get_error'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### Expected behavior

Compilation should succeed :-)

### ebusd version

23.3

### ebusd arguments

N/A

### Operating system

other

### CPU architecture

x64

### Dockerized

None

[... 189 lines cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_get1_peer_certificate'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `BIO_int_ctrl'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `SSL_get_state'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: src/lib/utils/libutils.so: undefined reference to `ERR_get_error'
collect2: error: ld returned 1 exit status
[... cut ...]
 * If you need support, post the output of `emerge --info '=app-misc/ebusd-23.3::galactica'`,
 * the complete build log and the output of `emerge -pqv '=app-misc/ebusd-23.3::galactica'`.
 * The complete build log is located at '/var/tmp/portage/app-misc/ebusd-23.3/temp/build.log'.
 * The ebuild environment file is located at '/var/tmp/portage/app-misc/ebusd-23.3/temp/environment'.
 * Working directory: '/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3_build'
 * S: '/var/tmp/portage/app-misc/ebusd-23.3/work/ebusd-23.3'
```

ConiKost, 2024-01-08:
This seems to be an issue with the CMake build system. Using Autotools does not fail here.

john30, 2024-01-13:
looks like the ssl lib does not get detected thoroughly by cmake. somebody would need to fix that

closed 2024-01-20 by john30