john30/ebusd #1164: Unable to find any device on the bus when signal is acquired

aspyk, 2024-01-22:
### Description

I run for the first time the docker version of ebusd on a rapsberry pi with a v5 adapter. The adapter is connected to the heater directly. Everything seems to run fine except it looks like no other device is found on the ebus. Here is the output:

```
docker run --rm -it --device=/dev/ttyAMA0 --name=ebusd john30/ebusd /bin/bash
root@969d3973bfbe:/# ebusd --scanconfig -d ens:/dev/ttyAMA0 --latency=50
2024-01-22 08:13:41.988 [bus notice] device status: transport opened
2024-01-22 08:13:41.989 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: /dev/ttyAMA0, serial high speed, enhanced
2024-01-22 08:13:43.331 [bus notice] bus started with own address 31/36
2024-01-22 08:13:43.331 [bus notice] signal acquired
2024-01-22 08:13:43.331 [bus notice] device status: reset, supports info
2024-01-22 08:13:43.367 [bus notice] device status: extra info: firmware 1.1[3501].1[3501], jumpers 0x03
2024-01-22 08:13:53.333 [main notice] starting initial broadcast scan
2024-01-22 08:13:53.426 [update notice] sent write broadcast queryexistence QQ=31: 
2024-01-22 08:15:03.428 [main notice] scan completed 1 time(s), check again
2024-01-22 08:15:53.708 [main notice] update check: OK, device firmware 1[4106] available
2024-01-22 08:16:13.709 [main notice] scan completed 2 time(s), check again
2024-01-22 08:17:23.711 [main notice] scan completed 3 time(s), check again
2024-01-22 08:18:33.714 [main notice] scan completed 4 time(s), check again
2024-01-22 08:19:43.716 [main notice] scan completed 5 time(s), check again
2024-01-22 08:20:53.719 [main notice] scan completed 6 time(s), end
```

And here the output of `ebusctl info` in another terminal connected to the container:
```
docker exec -it ebusd /bin/bash
root@969d3973bfbe:/# ebusctl info
version: ebusd 23.3.23.3
update check: OK, device firmware 1[4106] available
device: /dev/ttyAMA0, serial high speed, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 23
max symbol rate: 30
min arbitration micros: 22
max arbitration micros: 22
min symbol latency: 4
max symbol latency: 4
scan: finished
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

Do you know what's going wrong ? Thank you !

### Actual behavior

The scan seems to be ok but no other device is found on the bus (except itself) in `ebusctl info`.

### Expected behavior

another address should be detected in the output of `ebusctl info`.

### ebusd version

current source from git

### ebusd arguments

--scanconfig -d ens:/dev/ttyAMA0 --latency=50

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)
[... 272 lines cut ...]
2024-01-22 13:42:59.411 [main debug] performing regular tasks
2024-01-22 13:43:09.411 [main debug] performing regular tasks
2024-01-22 13:43:09.411 [main notice] scan completed 2 time(s), check again
2024-01-22 13:43:19.412 [main debug] performing regular tasks
2024-01-22 13:43:29.412 [main debug] performing regular tasks
2024-01-22 13:43:39.413 [main debug] performing regular tasks
2024-01-22 13:43:49.413 [main debug] performing regular tasks
2024-01-22 13:43:59.414 [main debug] performing regular tasks
2024-01-22 13:44:09.414 [main debug] performing regular tasks
2024-01-22 13:44:19.415 [main debug] performing regular tasks
2024-01-22 13:44:19.415 [main notice] scan completed 3 time(s), check again
2024-01-22 13:44:29.416 [main debug] performing regular tasks
2024-01-22 13:44:39.416 [main debug] performing regular tasks
2024-01-22 13:44:49.417 [main debug] performing regular tasks
2024-01-22 13:44:59.417 [main debug] performing regular tasks
2024-01-22 13:45:09.418 [main debug] performing regular tasks
2024-01-22 13:45:19.418 [main debug] performing regular tasks
2024-01-22 13:45:29.419 [main debug] performing regular tasks
2024-01-22 13:45:29.419 [main notice] scan completed 4 time(s), check again
2024-01-22 13:45:39.419 [main debug] performing regular tasks
2024-01-22 13:45:49.419 [main debug] performing regular tasks
2024-01-22 13:45:59.420 [main debug] performing regular tasks
2024-01-22 13:46:09.420 [main debug] performing regular tasks
2024-01-22 13:46:19.421 [main debug] performing regular tasks
```

john30, 2024-01-27:
well if no unit is willing to respond to the broadcast scan, then just try with a full scan instead.
anyway, if there is no traffic at all on the bus, it'll probably be hard to find useful data.
either way, this does not look like an ebusd issue in the first place, so closing

closed 2024-01-27 by john30

aspyk, 2024-01-28:
Ok indeed the full scan make the device pop...
I'm going to start the exploration now, thank you very much!