john30/ebusd #1178: Auto populating default connections

kjoglum, 2024-02-02:
### Description

Running ebusd on RPi4, against ESEERA, everything's  been working fine until mid December according to logs. Frequently checking my connection status, and  all of a sudden I  struggle with my manual initiation progress towards ebusd. It appears like changes  have been made to the  default config  file,  so  even when running manual ebusd initiation, the system crashes because  of  invalid device (which would be a result (apparently) due to changes made to the sudo config file. Checking  sudo naono /etc/ebusd/defaulr, I just get an  empty  file.

BR,
Stian

### Actual behavior

Not as expected

### Expected behavior

Manual override

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

No supported logs

john30, 2024-02-04:
sounds like misconfigured OS to me. hard to tell without any logs in the first place

closed 2024-02-04 by john30

kjoglum, 2024-02-04:
A bit hasty conclusion..?

Even after a fresh install (Bookworm install on RPi4), following guidelines for ebusd installation, a root instance for /usr/bin/ebusd is initiated. The instance is automatically re-populated even if running _kill_, even if `sudo nano /etc/default/ebusd` is deactivated, i.e. `"#EBUSD_OPTS="--scanconfig"`.

So, when running `sudo ebusd -d /dev/ttyUSB0 --scanconfig --configpath=/etc/ebusd/config --enablehex --lograwdata --logareas=all --logfile=/var/log/ebusd.log --loglevel=debug -f --mqtthost=localhost --mqttport=1883 --accesslevel="*"`, signal is initially acquired, and `[bus noice]` is starting to come in, when all of a sudden there seems to occur an issue.

As mentioned, the ebusd setup has been workling flawlessly until last December (using a modified 08.hmu file).

```
024-02-04 21:36:02.424 [bus notice] <710835110107b400053267a0a0280d987d
2024-02-04 21:36:03.768 [main debug] performing regular tasks
2024-02-04 21:36:03.768 [bus info] scan 08 cmd: 3108070400
2024-02-04 21:36:03.796 [bus debug] start request 31
2024-02-04 21:36:03.796 [bus debug] arbitration start with 31
2024-02-04 21:36:03.881 [bus notice] >31
2024-02-04 21:36:03.881 [bus debug] arbitration lost
2024-02-04 21:36:03.881 [bus debug] ERR: arbitration lost during ready, retry
2024-02-04 21:36:03.881 [bus debug] start request 31
2024-02-04 21:36:03.881 [bus debug] arbitration start with 31
2024-02-04 21:36:03.967 [bus notice] >31
2024-02-04 21:36:03.967 [bus debug] arbitration lost
2024-02-04 21:36:03.967 [bus debug] ERR: arbitration lost during ready, retry
2024-02-04 21:36:03.967 [bus debug] start request 31
2024-02-04 21:36:03.967 [bus debug] arbitration start with 31
2024-02-04 21:36:04.053 [bus notice] >31
2024-02-04 21:36:04.053 [bus debug] arbitration lost
2024-02-04 21:36:04.053 [bus debug] ERR: arbitration lost during ready, retry
2024-02-04 21:36:04.053 [bus debug] start request 31
2024-02-04 21:36:04.053 [bus debug] arbitration start with 31
```