john30/ebusd #1184: Default root /usr/bin/ebusd intance causing conflicts

kjoglum, 2024-02-05:
### Description

On a fresh install (after having had a fully functional ebusd setup over several years), running Debian 12 (Bookworm) on RPi4, ebusd v23.3.23.3, ESERA eBus coupler USB, a default /usr/bin/ebusd instance is causing conflicts.

The default /usr/bin/ebusd instance is restarted automatically after kill.

`sudo nano /etc/default/ebusd ` is defined as `#EBUSD_OPTS="--scanconfig`.

Similar issues of Arbitration lost are reported over the last couple of weeks:
https://github.com/john30/ebusd/discussions

### Actual behavior

**1. Manual start of ebusd without killing default /usr/bin/ebusd instance**

`sudo ebusd -d /dev/ttyUSB0 --scanconfig --configpath=/etc/ebusd/config --enablehex --lograwdata --logareas=all --logfile=/var/log/ebusd.log --loglevel=debug -f --mqtthost=localhost --mqttport=1883 --accesslevel="*" `

Log indicate _invalid device_, which would point in the direction of already having an active instance (i.e. the /usr/bin/ebusd instance):

`2024-02-05 12:28:05.550 [bus error] device status: transport closed
2024-02-05 12:28:05.550 [bus error] device status: transport closed
2024-02-05 12:28:05.550 [bus error] device /dev/ttyUSB0 not available
2024-02-05 12:28:05.550 [main info] registering data handlers
2024-02-05 12:28:05.550 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2024-02-05 12:28:05.552 [mqtt debug] connection requested
2024-02-05 12:28:05.552 [main info] registered data handlers
2024-02-05 12:28:05.552 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: /dev/ttyUSB0, serial
2024-02-05 12:28:05.552 [main info] loading configuration files from /etc/ebusd/config/
2024-02-05 12:28:05.552 [main debug] reading directory /etc/ebusd/config/
2024-02-05 12:28:05.554 [main debug] file type of /etc/ebusd/config/._broadcast.csv is file
2024-02-05 12:28:05.554 [main debug] file type of /etc/ebusd/config/._.DS_Store is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/._vaillant is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/memory.csv is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/._memory.csv is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/vaillant is dir
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/.DS_Store is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/broadcast.csv is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/.__templates.csv is file
2024-02-05 12:28:05.555 [main debug] file type of /etc/ebusd/config/_templates.csv is file
2024-02-05 12:28:05.556 [main info] reading templates /
2024-02-05 12:28:05.558 [main info] read templates in /
2024-02-05 12:28:05.558 [main info] reading file ._broadcast.csv
2024-02-05 12:28:05.559 [main error] error reading config files from /etc/ebusd/config/: ERR: end of input reached, last error: ._broadcast.csv:1: ERR: end of input reached, missing message type/name/pbsb
2024-02-05 12:28:05.560 [bus notice] bus started with own address 31/36
2024-02-05 12:28:05.560 [bus notice] device invalid `

`2024-02-05 12:28:10.562 [bus error] device status: transport closed
2024-02-05 12:28:10.562 [bus error] device status: transport closed
2024-02-05 12:28:10.562 [bus notice] re-opened /dev/ttyUSB0
2024-02-05 12:28:10.562 [bus notice] device invalid `

**2. Manual start of ebusd after killing default /usr/bin/ebusd instance**

`sudo ebusd -d /dev/ttyUSB0 --scanconfig --configpath=/etc/ebusd/config --enablehex --lograwdata --logareas=all --logfile=/var/log/ebusd.log --loglevel=debug -f --mqtthost=localhost --mqttport=1883 --accesslevel="*" `

Initiation starts as expected, bus traffic recognised:

`2024-02-05 11:30:24.566 [bus notice] device status: transport opened
2024-02-05 11:30:24.566 [main info] registering data handlers
2024-02-05 11:30:24.566 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2024-02-05 11:30:24.568 [mqtt debug] connection requested
2024-02-05 11:30:24.568 [main info] registered data handlers
2024-02-05 11:30:24.568 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: /dev/ttyUSB0, serial
2024-02-05 11:30:24.568 [main info] loading configuration files from /etc/ebusd/config/
2024-02-05 11:30:24.568 [main debug] reading directory /etc/ebusd/config/
2024-02-05 11:30:24.570 [main debug] file type of /etc/ebusd/config/._broadcast.csv is file
2024-02-05 11:30:24.570 [main debug] file type of /etc/ebusd/config/._.DS_Store is file
2024-02-05 11:30:24.570 [main debug] file type of /etc/ebusd/config/._vaillant is file
2024-02-05 11:30:24.571 [main debug] file type of /etc/ebusd/config/memory.csv is file
2024-02-05 11:30:24.571 [main debug] file type of /etc/ebusd/config/._memory.csv is file
[... 74 lines cut ...]
23.3

### ebusd arguments

#EBUSD_OPTS="--scanconfig

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter v3 USB

### Logs

See above

john30, 2024-02-11:
duplicate of #1178, so closing.

anyway, repeating https://github.com/john30/ebusd/issues/1178#issuecomment-1937702280 :

if ebusd is installed, it registers in the OS as daemon and as such you have to stop that if you want to run it explicitly in foreground.

closed 2024-02-11 by john30

kjoglum, 2024-02-11:
But how/why is this feature introduced between version 23.2 and 23.3? Even if killing the root instance, it restarts automatically. And if one uses  custom config files, one would not be able to use ebusd v23.3.

john30, 2024-02-11:
killing the "root instance" as you call it (which is the one started from systemd or init.d) won't solve it as it will be restarted intentionally (as with any daemon set up to do so e.g. in the systemd config file https://github.com/john30/ebusd/blob/7456bd6a54248507f9a6ae4c2bbb3116b3963aa2/contrib/debian/systemd/ebusd.service#L8)