john30/ebusd #1188: homeassistant log error:  Invalid value for number.ebusd_700_continuosheating_tempv: -25.0 (range 0.0 - 100.0)

tikismoke, 2024-02-06:
### Description

HI john, i don't know how to avoid those sort of log in HA.

### Actual behavior

value of -25.0 in the sensor i think?

### Expected behavior

log:
 Invalid value for number.ebusd_700_continuosheating_tempv: -25.0 (range 0.0 - 100.0) 

### ebusd version

current source from git

### ebusd arguments

EBUSD_OPTS="-d X.X.X.X:3333 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson  --loglevel=error --mqtthost=mosquito --mqttport=1883 --scanconfig --pollinterval=3 --accesslevel=*"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

not relevant in this case

tikismoke, 2024-02-06:
pretty sure it's a simple parameter in mqtthassio config file but can't understand wh needs to be changed or where :(

john30, 2024-02-11:
not so easy, as it is a valid value in the first place. plus it looks like -26 (and maybe even -25) is a special value according to the comment in the definition. so, not possible to get rid of that one for now