john30/ebusd #1198: HA Addon ebusd / tado / no MQTT data refresh

Dominik-1980, 2024-02-19:
### Description

I use tado as room thermostat. 
When I start ebusd HA Addon, the values I specified with priority 1 in the bai file (Flame / WaterPressure) where synced every 2 seconds like I configured. For testing I set this to 2 sec instead of 5...
So far so good. But at the moment the communication with tado starts to show up in the logs these values are not synced anymore. 

I marked the position in the log file where these values are not synced any more...

### Actual behavior

No MQTT values are synced while communicating with tado

### Expected behavior

Monitor specified values with ebusd and MQTT in HA besides tado controlling the heater

### ebusd version

23.2

### ebusd arguments

none

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Logs

2024-02-19 16:43:02.360 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;1003;7603
2024-02-19 16:43:02.360 [bus notice] scan 08: ;Vaillant;BAI00;1003;7603
2024-02-19 16:43:02.377 [main notice] read common config file vaillant/broadcast.csv
2024-02-19 16:43:02.378 [main notice] read common config file vaillant/scan.csv
2024-02-19 16:43:02.380 [main notice] read common config file vaillant/general.csv
2024-02-19 16:43:02.382 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW1003, HW7603
2024-02-19 16:43:02.635 [update notice] sent scan-read scan.08 id QQ=31: 
2024-02-19 16:43:02.917 [update notice] sent scan-read scan.08 id QQ=31: 
2024-02-19 16:43:03.735 [update notice] sent scan-read scan.08 id QQ=31: 
2024-02-19 16:43:03.995 [update notice] sent scan-read scan.08 id QQ=31: 21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:04.020 [main notice] found messages: 224 (3 conditional on 28 conditions, 2 poll, 10 update)
2024-02-19 16:43:04.283 [update notice] sent poll-read bai WaterPressure QQ=31: 1.532;ok
2024-02-19 16:43:04.556 [update notice] sent scan-read scan.08 id QQ=31: 21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:04.815 [update notice] sent scan-read scan.08 id QQ=31: 21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:04.995 [update notice] received unknown MS cmd: 1008b5110100 / 0917040f4b040f00813d
2024-02-19 16:43:05.225 [update notice] sent scan-read scan.08 id QQ=31: 21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:05.495 [update notice] sent scan-read scan.08 id QQ=31: 21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:05.495 [bus notice] scan 08: ;21;23;26;0010025210;1610;007720;N0
2024-02-19 16:43:05.836 [update notice] received read bai Status01 QQ=10: 65.0;55.5;-;40.0;41.0;on
2024-02-19 16:43:06.847 [update notice] received read bai Status02 QQ=10: auto;30;75.0;90;65.0
2024-02-19 16:43:07.286 [update notice] sent poll-read bai Flame QQ=31: on
2024-02-19 16:43:07.735 [bus notice] scan 15: ;Vaillant;37000;0129;6002
2024-02-19 16:43:07.735 [update notice] store 15 ident: done
2024-02-19 16:43:07.736 [update notice] sent scan-read scan.15  QQ=31: Vaillant;37000;0129;6002
2024-02-19 16:43:07.736 [bus notice] scan 15: ;Vaillant;37000;0129;6002
2024-02-19 16:43:07.865 [update notice] received poll-read bai currenterror QQ=10: -;-;-;-;-
2024-02-19 16:43:08.104 [main notice] read scan config file vaillant/15.370.csv for ID "37000", SW0129, HW6002
2024-02-19 16:43:08.112 [main notice] found messages: 368 (3 conditional on 28 conditions, 162 poll, 10 update)
[... 50 lines cut ...]
2024-02-19 16:43:49.136 [update notice] received unknown MS cmd: 1008b5110100 / 0917040f4a040f00813c
2024-02-19 16:43:49.336 [update notice] sent poll-read bai averageIgnitiontime QQ=31: 2.5
2024-02-19 16:43:50.158 [update notice] received read bai Status01 QQ=10: 65.0;56.0;-;40.0;41.0;on
2024-02-19 16:43:51.126 [update notice] received read bai Status02 QQ=10: auto;30;75.0;90;65.0
2024-02-19 16:43:52.171 [update notice] received poll-read bai currenterror QQ=10: -;-;-;-;-
2024-02-19 16:43:52.455 [update notice] sent poll-read bai BlockTimeHcMax QQ=31: 1
2024-02-19 16:43:52.475 [bus notice] arbitration won in invalid state ready
2024-02-19 16:43:53.112 [update notice] received unknown MS cmd: 1008b509030d4a00 / 00
2024-02-19 16:43:54.114 [update notice] received poll-read bai HwcWaterflow QQ=10: 0.00
2024-02-19 16:43:55.116 [update notice] received poll-read bai HcHours QQ=10: 1414
2024-02-19 16:43:55.376 [update notice] sent poll-read bai BoilerType QQ=31: 2
2024-02-19 16:43:56.127 [update notice] received poll-read bai HwcHours QQ=10: 39
2024-02-19 16:43:57.128 [update notice] received poll-read bai HwcStarts QQ=10: 700
2024-02-19 16:43:58.135 [update notice] received poll-read bai HcStarts QQ=10: 3000
2024-02-19 16:43:58.365 [update notice] sent poll-read bai ChangesDSN QQ=31: 0
2024-02-19 16:43:59.156 [update notice] received poll-read bai CirPump QQ=10: on
2024-02-19 16:44:00.155 [update notice] received unknown MS cmd: 1008b509030d7300 / 014c
2024-02-19 16:44:01.156 [update notice] received poll-read bai Gasvalve QQ=10: on
2024-02-19 16:44:02.158 [update notice] received poll-read bai FlowTempDesired QQ=10: 65.50
2024-02-19 16:44:03.167 [update notice] received poll-read bai IonisationVoltageLevel QQ=10: 0.0
2024-02-19 16:44:04.276 [update notice] sent poll-read bai CounterStartattempts1 QQ=31: 0
2024-02-19 16:44:06.276 [update notice] received update-write bai SetMode QQ=10: auto;65.5;45.0;-;0;0;0;0;0;0
2024-02-19 16:44:06.395 [update notice] received unknown MS cmd: 1008b51003055a00 / 0101
2024-02-19 16:44:07.392 [update notice] sent poll-read bai CounterStartattempts2 QQ=31: 0
2024-02-19 16:44:08.256 [update notice] received update-write bai StatusCirPump QQ=10: on

john30, 2024-02-29:
no idea. is this tado thing connected to ebus or how would it influence ebusd in the first place?

wrongisthenewright, 2024-03-28:
Don't own a tado bu i was looking at it before settling all my needs with Ebusd :)

Taso thermostat is compatible with Ebus and it writes on the bus to control the HVAC system.
This page states compatibility:

https://support.tado.com/de/articles/3387245-ist-tado-mit-meiner-therme-kompatibel

Thanks John!

john30, 2024-03-29:
maybe it is just an address conflict, so check the result of "ebusctl info" for that and post the list of found addresses

championc, 2024-09-25:
Personally, I have my TADO Extension Kit connected to the EVU connector, and with the Entension Kit configured with a Potential Free wired for NO

john30, 2024-10-26:
closing due to inactivity

closed 2024-10-26 by john30